espressif/esp32-camera #304: View non-JPEG image in Example

LakyG, 2021-07-27:
I am trying to send non-JPEG images using one of the following (currently trying with Greyscale):
```
config.pixel_format = PIXFORMAT_GRAYSCALE;
//config.pixel_format = PIXFORMAT_YUV422;
//config.pixel_format = PIXFORMAT_RGB565;
```
However, when I try to view the image in the browser, I can't view the image if it isn't JPEG. In the browser, it shows up as a small black box, but images are being sent since I am printing whenever the image is read from the frame buffer on the ESP32.

I am using the ESP32 `WebServer` library with the HTTP protocol.

I tried changing the `Content-Type: image/bmp\r\n\r\n`, but this still let me see the image.

Any suggestions for what I am missing?

I appreciate the help!

TDA-2030, 2021-07-28:
Can you try to run the [test code](https://github.com/espressif/esp32-camera/blob/6a9497bbe909165663d958986e621e98dabcf994/test/test_camera.c#L318)?  you can copy the base64 code from console to [decoder](https://base64.guru/converter/decode/file) to show the image.
This can verify whether the data you read out is correct

me-no-dev, 2021-07-28:
You can see how you can display the images in browser in this example: https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server

closed 2021-10-03 by github-actions[bot]