espressif/esp32-camera #305: Reading current exposure time values from AEC registers

frka1204, 2021-08-01:
I have a ESP32-CAM that basically is switched on when a PIR sensor is triggered. I would like to minimize the time from boot to photo capture. I use auto exposure control (AEC) . My idea was to read the AEC values from 0x45, 0x10 and 0x04 and see when they are stabilized, then the proper exposure should have been calculated and a photo can be captured. However, the registers don't seem to be updating or my code is wrong.
Here's the relevant code snippet I'm using:

```
sensor_t * s = esp_camera_sensor_get();
s->set_reg(s, 0xFF, 0x01, 0xFF);
uint16_t aec1 = s->get_reg(s, 0x45, 0x1F);
uint16_t aec2 = s->get_reg(s, 0x10, 0xFF);
uint16_t aec3 = s->get_reg(s, 0x04, 0x03);
uint16_t aec = aec3 + (aec2 << 2) + (aec1 << 10); // Value: 671
while(1) {
        s->set_reg(s, 0x13, 0xFF, 0xC0);
        printf("AEC value: %lu \n", (unsigned long) aec);  // Always the same value: 671
        aec1 = s->get_reg(s, 0x45, 0x1F);
        aec2 = s->get_reg(s, 0x10, 0xFF);
        aec3 = s->get_reg(s, 0x04, 0x03);
        aec = aec3 + (aec2 << 2) + (aec1 << 10); // Always the same value: 671
        vTaskDelay(100 / portTICK_RATE_MS);
} 
```
My question is if there's a way to update the registers or a way to determine when AEC, AGC, AWB have been properly set up so that a photo can be captured in minimum time after boot?

TDA-2030, 2021-08-06:
@frka1204 I don't know much about sensors, but it seems that you should go to the sensor datasheet for information.

achao1104, 2022-06-06:
d

> I have a ESP32-CAM that basically is switched on when a PIR sensor is triggered. I would like to minimize the time from boot to photo capture. I use auto exposure control (AEC) . My idea was to read the AEC values from 0x45, 0x10 and 0x04 and see when they are stabilized, then the proper exposure should have been calculated and a photo can be captured. However, the registers don't seem to be updating or my code is wrong. Here's the relevant code snippet I'm using:
> 
> ```
> sensor_t * s = esp_camera_sensor_get();
> s->set_reg(s, 0xFF, 0x01, 0xFF);
> uint16_t aec1 = s->get_reg(s, 0x45, 0x1F);
> uint16_t aec2 = s->get_reg(s, 0x10, 0xFF);
> uint16_t aec3 = s->get_reg(s, 0x04, 0x03);
> uint16_t aec = aec3 + (aec2 << 2) + (aec1 << 10); // Value: 671
> while(1) {
>         printf("AEC value: %lu \n", (unsigned long) aec);  // Always the same value: 671
>         aec1 = s->get_reg(s, 0x45, 0x3F);
>         aec2 = s->get_reg(s, 0x10, 0xFF);
>         aec3 = s->get_reg(s, 0x04, 0x03);
>         aec = aec3 + (aec2 << 2) + (aec1 << 10); // Always the same value: 671
>         vTaskDelay(100 / portTICK_RATE_MS);
> } 
> ```
> 
> My question is if there's a way to update the registers or a way to determine when AEC, AGC, AWB have been properly set up so that a photo can be captured in minimum time after boot?

I have the same issue, and do you know how to read the exposure time in AEC auto mode?