espressif/esp32-camera #306: Understanding I2S / Camera Data Protocol

LakyG, 2021-08-04:
Hi,

I am new to using the ESP32 Camera Library and I wanted to understand how the I2S peripheral is used to receive, construct and store images (both RGB [uncompressed] and JPEG [compressed]).

I understand the basics of the VSYNC, HSYNC and PCLK signals and how they are the main signals that are used for data transmission. However, what I don't understand is how the ESP32 then reconstructs the image data received from the Camera (OV2640 in my case).

Lets say one **_uncompressed_** frame is being transmitted from the camera (UXGA - 1600x1200) to the ESP32. That means one long VSYNC signal with many HSYNCs (for each line) and even more PCLKs (for each pixel/column). The data is being sent in the 'BG/GR' order for each line of HSYNC.

[image]

**1. Firstly, in the above image, P_0, P_1, P_2, etc. refer to the BG/GR values is that correct?**

---

So far, that is all on the camera side. Now onto the ESP32 side which is where I am mostly confused.

As PCLK oscillates and the pixel values (BG/GR values) are sent over the parallel data lines, I2S is receiving them and the DMA configuration is continuously storing this data.

**2. Is the DMA storing that data directly into the `cam_obj->frames` array, and then in `cam_obj->frames[i]->fb`?**
**3. Is the raw data being stored there, or is it first processed in some way?**
**4. Is there a complete image in each of the `cam_obj->frames[i]`, or just part of the image? If it's just a part, how does the rest of it get appended?** 
**5. Does that frame array store the entire video stream? If so, how many frames in the past?**
 
---

**6. Uncompressed images still make sense, but for JPEG (compressed images), how does the decoding happen on the ESP32 side since the data is much more sporadic/irregular?**

I think the above few questions are where I am most lost.

This is probably a very long post that is asking a lot of conceptual questions so I appreciate the time anyone takes to provide answers. Thank you so much!

TDA-2030, 2021-08-06:
@LakyG 
I'm not very clear, but I can briefly explain my understanding.  For the esp32 chip, the data received by I2S is first DMA transferred to a buffer, and then copied to the frame buffer. Reciprocating multiple times to complete a frame of image. If it is JPEG, it will check the header and footer marks of JPEG and remove invalid data after completing the transmission of one frame.

me-no-dev, 2021-08-09:
1. Px are 8bit values from the image stream. Their meaning depends on the format. In your case, they are BG/GR
2. DMA on the ESP32 stores into a special chunk of DMA accessible memory that is on the internal RAM. That memory is then further split into DMA items of equal size (less than 4KB and dependent on the resolution/mode). When I2S fills one such item, an interrupt is triggered and that item is sent to a queue which is accessed by a task which then copies the data to the currently used `cam_obj->frames[i]->fb`.
3. Data is stored into the frame buffers as-is and no modification is done to it.
4. Yes. each frame buffer holds one complete image/frame. Of course exception is the buffer that is currently being filled by the DMA task.
5. Depending on the resolution, one frame can take a lot of space, so how many frames you can store depends on how much memory you have available (max 4MB PSRAM on ESP32) and is a configuration value when you init the camera
6. Think of the bus as you are an SPI slave device, but 8bit. HREF is your SlaveSelect, PCLK is your SCLK, you get 8bits each SCLK and you switch the buffer when you get VSYNC (IRQ). It does not care how often you send the data, it shifts only when HREF is LOW and PCLK is going HIGH

LakyG, 2021-08-09:
@me-no-dev That makes a lot of sense. Your both of these answers explain that he frame buffer is just filled up while VSYNC is asserted (positive polarity) AND while HREF is asserted. Both those signals being asserted is essentially the enable to continue filling up the frame buffer. At the end of VSYNC, we go to the next frame. And this is true for Uncompressed and JPEG, correct?

One thing to clarify, HREF and HSYNC in this case are interchangeable? The OV2640 connector is labeled as "HREF" so is it actually HREF, or HSYNC on that line, and what is the difference between those two?

Once again, I very much appreciate the activeness and prompt replies in this community!

me-no-dev, 2021-08-10:
HREF == HSYNC :)

VSYNC signals new frame on all formats

closed 2021-10-16 by github-actions[bot]

dumarjo, 2022-11-04:
On this datasheet https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf you have a timing for the LCD, but not for the CAM. It's not clear when the data is sampled from the ESP32S3 perpective. 

Let's say for example I need to use this TOF camera with the ESP32S3. 
[image]

What are the settings needed to be sets ? 

For my specific project, the DCLK (PCLK) is generated internally from the camera and it's between 10Mhz and 20Mhz

From my reading, the HSYNC and PCLK looks like inverted from the default initialization of the driver ?

Any helps appreciated

me-no-dev, 2022-11-06:
@dumarjo for S3 you can find the controls for signal polarity [here](https://github.com/espressif/esp32-camera/blob/master/target/esp32s3/ll_cam.c#L263-L268). There are controls for hsync, vsync and clk inverting. Give them a try :)