espressif/esp32-camera #307: OV5640 with the ESP32-CAM: "cam_hal: NO-SOI" and "camera: failed to capture frame on time"

ava-droid, 2021-08-04:
I'm trying to use the OV5640 sensor with the ESP32-CAM and have run into some trouble getting the sensor to work. Right now, I'm just trying to capture a frame and parse the characteristics of the image and not even display the picture yet but am running into an error that I can seem to find any information on. I have used the OV2640 sensor with this board with no issue but it leaves a lot to be desired in terms of image quality which is why I want to try the OV5640

Here is a link to the [code](https://gist.github.com/ava-droid/5d187b3baec712b16f6633e832ad4e70)

Here is a link of the [logs](https://gist.github.com/ava-droid/35f04a787778045566fb03bb0773505f)

I have looked up the nature of the "Failed to get the frame on time!" error message and have tried solutions that have seemed to work for other users but nothing has worked for me. Specifically looking into the `cam_hal: NO-SOI` warning message turns up nothing in my searching. I have tried all sorts of different settings in the camera configuration with no success such as changing the `.pixel_format`, `xclk_freq_hz`, `.frame_size`, `.jpeg_quality`, and `.fb_count`

Here is a link to the [sdkconfig](https://gist.github.com/ava-droid/e313d2baa48eebfd2b41fa45ae1b081b)
I have enabled the SPI-connected RAM as well as the rtc_gpio_desc support array in the menuconfig.

TDA-2030, 2021-08-06:
@ava-droid Can you test with decrease the `frame_size`? The `FRAMESIZE_XGA` is really a big size. Some strange problems sometimes occur when the resolution is too large :(

ava-droid, 2021-08-09:
I tried reducing the `frame_size` and I still see the same result of `cam_hal: NO-SOI` and `camera: failed to capture frame on time`. Do you think this issue is due to the lack of memory available on the ESP32-CAM (I think it has 4 MB of PSRAM) and if I were to go to a larger module, such as the ESP-EYE with 8MB of PSRAM the issue would disappear?

ava-droid, 2021-08-10:
I got an ESP-EYE module and have set it up with the same configuration as the ESP32-CAM. I am still getting the error of `cam_hal: NO-SOI` and `camera: failed to capture frame on time`. I'm wondering if it could be a faulty camera sensor. Looking further into `NO-SOI` seems to indicate that the device can't find the start of the JPEG image. 

I tried capturing using `PIXFORMAT_YUV422` and could only capture with `.frame_size = FRAMESIZE_HVGA`, any larger sizes and I would run into the error: `cam_hal: FB-SIZE: 610560 != 614400` or `cam_hal: FB-SIZE: 1568768 != 1572864` depending on what `.frame_size` I used (VGA and XGA, respectively for the two mentioned).

lancefi, 2021-08-10:
Hey I'm getting the same "failed to  capture frame on time" issue with ov5640 and esp32-cam with 4mb psram.

Works okay with ov2640.

TDA-2030, 2021-08-10:
@ava-droid The error `cam_hal: NO-SOI` has nothing to do with insufficient memory. It means that driver can't find the JPEG SOI(Start of Image) mark. If there is insufficient memory, initialization cannot be completed. In addition to reducing the resolution, you can also reduce frequency of xclk. 

The error `cam_hal: FB-SIZE: 610560 != 614400` means that the received data length is not equal to the data length corresponding to `.framesize`.

In addition, on the current driver, if you want to change framesize or pixelformat, you must first call `esp_camera_deinit()` and then call `esp_camera_init` with new parameter.

ava-droid, 2021-08-10:
> @ava-droid The error `cam_hal: NO-SOI` has nothing to do with insufficient memory. It means that driver can't find the JPEG SOI(Start of Image) mark. If there is insufficient memory, initialization [...]

Thanks for the information. I have tried reducing the XCLK frequency and have had no change with the results. In regards to your last point, do you mean even after flashing new code I should be calling `esp_camera_denit()` before calling `esp_camera_init()`?

ava-droid, 2021-08-12:
I received two new OV5640 sensors and have used them with both the ESP32-CAM as well as the ESP-EYE. With both modules the issue disappears so it does seem like I have a faulty sensor with the original one I was using. One issue with the ESP32-CAM is that when I test without a webserver the camera is detected and can capture a frame with no issues, but when I try to test with a webserver the module is unable to detect what camera is connected. Weird issue but unrelated to this one.

closed 2021-08-12 by ava-droid

lancefi, 2021-08-13:
Hey @ava-droid what were the models printed on the ribbon cable of the cameras that didn't work and the ones that did work?

ava-droid, 2021-08-16:
> 
> 
> Hey @ava-droid what were the models printed on the ribbon cable of the cameras that didn't work and the ones that did work?

The model printed on the ones that are working is CV5021-G2. The one that wasn't working is FD5640 500W-V1 1