espressif/esp32-camera #308: ESP32-S2-DevKitC with camera ov2640

PureHing, 2021-08-25:
@me-no-dev 
Hi
When I used esp32s2 to initialize the camera, it failed. Demo code modified from  [here](https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-kaluga-1/examples/camera)
| CAMERA PIN | XCLK | PCLK | VSYNC | HSYNC | D2   | D3   | D4   | D5   | D6   | D7   | D8   | D9   |      |
| ---------- | ---- | ---- | ----- | ----- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- |
| **GPIO**   | 1    | 33   | 2     | 3     | 46   | 45   | 41   | 42   | 39   | 40   | 21   | 38   |      |

```
I (1081) cpu_start: Project name:     ov2640_camera
I (1087) cpu_start: App version:      1
I (1091) cpu_start: Compile time:     Aug 25 2021 15:15:36
I (1097) cpu_start: ELF file SHA256:  4d22d5b75af39dc3...
I (1103) cpu_start: ESP-IDF:          v4.4-dev-2184-g166c30e7b2-dirty
I (1110) heap_init: Initializing. RAM available for dynamic allocation:
I (1118) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (1124) heap_init: At 3FFC2AD8 len 00039528 (229 KiB): DRAM
I (1130) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (1137) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (1145) spi_flash: detected chip: generic
I (1149) spi_flash: flash io: qio
I (1154) cpu_start: Starting scheduler on PRO CPU.
I (1159) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1159) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1169) gpio: GPIO[11]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1179) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1189) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1199) lcd: lcd_buffer_size: 2048, lcd_dma_size: 1024, lcd_dma_node_cnt: 2

I (1509) lcd: ILI19341 init...

I (1709) lcd: lcd init ok

I (1709) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (1709) cam_hal: cam init ok
I (1709) sccb: pin_sda 8 pin_scl 7
I (1739) camera: Detected camera at address=0x30
I (1839) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (3439) camera: Detected OV2640 camera
I (3439) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 15

I (3439) cam_hal: cam config ok
I (5359) main: Camera Init done
I (5359) main: Camera jpeg mode
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x400864d0  PS      : 0x00060730  A0      : 0x80085aec  A1      : 0x3ffc5320  
0x400864d0: cam_verify_jpeg_eoi at /home/librazxc/SoftwareDevelop/esp/ov2640_camera/build/../components/esp32-camera/driver/cam_hal.c:43
 (inlined by) cam_take at /home/librazxc/SoftwareDevelop/esp/ov2640_camera/build/../components/esp32-camera/driver/cam_hal.c:424

A2      : 0x00000190  A3      : 0x000001a4  A4      : 0x3f780974  A5      : 0x3ffc1c04  
A6      : 0x3ffc5330  A7      : 0x0000000c  A8      : 0x3f781dc5  A9      : 0x0000d9ff  
A10     : 0x00009ac8  A11     : 0x3f7809c8  A12     : 0xffffd9ff  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000002  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3f781dc5  LBEG    : 0xffffd9ff  LEND    : 0x00000001  LCOUNT  : 0x400265e9  
0x400265e9: _xt_user_exc at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:627

Backtrace:0x400864cd:0x3ffc53200x40085ae9:0x3ffc5350 0x4008572c:0x3ffc5370 0x4009b864:0x3ffc5420 
0x400864cd: cam_verify_jpeg_eoi at /home/librazxc/SoftwareDevelop/esp/ov2640_camera/build/../components/esp32-camera/driver/cam_hal.c:43
 (inlined by) cam_take at /home/librazxc/SoftwareDevelop/esp/ov2640_camera/build/../components/esp32-camera/driver/cam_hal.c:424

0x40085ae9: esp_camera_fb_get at /home/librazxc/SoftwareDevelop/esp/ov2640_camera/build/../components/esp32-camera/driver/esp_camera.c:318

0x4008572c: app_main at /home/librazxc/SoftwareDevelop/esp/ov2640_camera/build/../main/main.c:102

0x4009b864: main_task at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/freertos/port/port_common.c:142
```

The problem seems to be [here](https://github.com/espressif/esp32-camera/blob/fa5020f5c75df4f4aeaaf7d7a7ddceb84969c604/driver/cam_hal.c#L43)

PureHing, 2021-09-02:
@TDA-2030 accidental

PureHing, 2021-09-06:
@TDA-2030 inevitable

closed 2021-09-06 by PureHing

SaurabhCoolR, 2022-01-28:
Hi @PureHing :

Was the issue resolved?. We are facing the same issue with following:
esp idf v4.4
esp camera v2.0.0