espressif/esp32-camera #312: Photo capture produces artefacts

remibert, 2021-08-28:
The latest versions produce artifacts, part of the image does not have the same contrast. I am working on a motion detection product, and if I put the git version 6b1efc1eddf03502296bce7aca18d7a18fb00bfe (add gcxxxx sensors (#293)
), I have this kind of problem, and if I put the version 722497cb19383cd4ee6b5d57bb73148b5af41b24(Merge pull request #265 from espressif/develop/update_structure) the problem disappears. I haven't been able to find a version between the two because the compilation failed. The very last one also has the problem. I work with ESP32CAM with ov2640.

remibert, 2021-08-28:
[image]
[image]
[image]
[image]
[image]

jameszah, 2021-08-28:
I think those are "broken jpegs".  I noticed it with the arrival of 1.05 esp32-arduino, but don't know how your version numbers of the esp32-camera fit in with that.

Try this -- my theory was the i2s of the  camera fighting with the i2s of the wifi ???

`  esp_err_t set_ps = esp_wifi_set_ps(WIFI_PS_NONE);`

https://github.com/espressif/esp32-camera/issues/244#issuecomment-831561336

remibert, 2021-08-28:
It is not a problem of wifi, the change of git version produces or not the problem. With an old version of esp32_camera no problem over a week of shooting, with 2 images seconds, with a recent version, every 10 seconds, we can see a problem with the images. The rest of the firmware is the same, I tried to go up to find the origin in the git, but I have problems compiling with the esp32-camera on certain revision. Looking at the photos I have attached, you can visually see the problem. Changing the contrast in the middle of a shot resembles a synchronization that is no longer done at the right time.

remibert, 2021-10-30:
I tested version 3022601f876cc158f98c168624a701978509ee43 the problem is still present. So I stay on the 722497cb19383cd4ee6b5d57bb73148b5af41b24 version which does not produce thi
[image]
[image]
[image]
s.

remibert, 2021-10-30:
The change in contrast seems to be in the middle of an image, it must be missing a synchronization.

closed 2022-01-06 by github-actions[bot]