espressif/esp32-camera #315: Getting Error while initializing OV7670 in  Example due to PSRAM. I am using esp32 flash-4MB

viveky1794, 2021-09-07:
ESP32
4MB Flash Size, NO PSRAM
ECLIPSE_ESP_IDF 

1-> I upload esp32_camera/example code successfully.
2-> When I run I got error related to PSRAM.

` 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6388
load:0x40078000,len:14236
load:0x40080400,len:3820
0x40080400: _init at ??:?

entry 0x40080698
I (27) boot: ESP-IDF v4.4-dev-2359-g58022f859-dirty 2nd stage bootloader
I (27) boot: compile time 11:10:14
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0c368h ( 50024)
 map
I (129) esp_image: segment 1: paddr=0001c390 vaddr=3ffb0000 size=034d8h ( 13528)
 load
I (135) esp_image: segment 2: paddr=0001f870 vaddr=40080000 size=007a8h (  1960)
 load
I (136) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=211ach (135596)
 map
I (193) esp_image: segment 4: paddr=000411d4 vaddr=400807a8 size=0bfc4h ( 49092)
 load
I (213) esp_image: segment 5: paddr=0004d1a0 vaddr=50000000 size=00010h (    16)
 load
I (220) boot: Loaded app from partition at offset 0x10000
I (220) boot: Disabling RNG early entropy source...
I (233) cpu_start: Pro cpu up.
I (233) cpu_start: Starting app cpu, entry point is 0x400810b8
0x400810b8: call_start_cpu1 at C:/Users/Vivek/Desktop/esp-idf/components/esp_sys
tem/port/cpu_start.c:155

I (0) cpu_start: App cpu up.
I (248) cpu_start: Pro cpu start user code
I (248) cpu_start: cpu freq: 160000000
I (248) cpu_start: Application information:
I (252) cpu_start: Project name:     app-template
I (257) cpu_start: App version:      1
I (262) cpu_start: Compile time:     Sep  7 2021 11:09:45
I (268) cpu_start: ELF file SHA256:  e13c8c15263adeb4...
I (274) cpu_start: ESP-IDF:          v4.4-dev-2359-g58022f859-dirty
I (281) heap_init: Initializing. RAM available for dynamic allocation:
I (288) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (294) heap_init: At 3FFB3E30 len 0002C1D0 (176 KiB): DRAM
I (300) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (307) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (313) heap_init: At 4008C76C len 00013894 (78 KiB): IRAM
I (320) spi_flash: detected chip: winbond
I (324) spi_flash: flash io: dio
W (328) spi_flash: Detected size(4096k) larger than the size in the binary image
 header(2048k). Using the size in the binary image header.
I (342) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (352) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:2
I (362) cam_hal: cam init ok
I (362) sccb: pin_sda 26 pin_scl 27
I (382) camera: Detected camera at address=0x21
I (382) ov7725: Mismatch PID=0x76
I (382) camera: Detected OV7670 camera
I (382) camera: Camera PID=0x76 VER=0x73 MIDL=0x7f MIDH=0xa2
I (442) esp32 ll_cam: node_size: 2560, nodes_per_line: 1, lines_per_node: 1, dma
_half_buffer_min:  2560, dma_half_buffer: 15360, lines_per_half_buffer:  6, dma_
buffer_size: 30720, image_size: 153600
I (442) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size:
2560, node_cnt: 12, total_cnt: 10
I (452) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
E (462) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (472) cam_hal: cam_config(364): cam_dma_config failed
E (472) camera: Camera config failed with error 0xffffffff
Camera Init Failed
To exit from IDF monitor please use "Ctrl+]"

`

me-no-dev, 2021-09-08:
1. You have selected (or have) 2MB flash instead of 4MB `I (48) boot.esp32: SPI Flash Size : 2MB` and `W (328) spi_flash: Detected size(4096k) larger than the size in the binary image
header(2048k). Using the size in the binary image header.`

2. You need PSRAM in order to capture such RGB images (you can not allocate 153600 bytes in internal RAM) `E (462) cam_hal: cam_dma_config(280): frame buffer malloc failed`. You can use internal RAM for some JPEG resolutions, but the sensor you are using doe not support JPEG.

closed 2021-09-08 by me-no-dev

viveky1794, 2021-09-09:
1-> I changed the flash size from 2MB to 4 MB

```ruby 

I (27) boot: ESP-IDF v4.4-dev-2359-g58022f859-dirty 2nd stage bootloader
I (27) boot: compile time 15:49:36
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0cd5ch ( 52572)
 map
I (130) esp_image: segment 1: paddr=0001cd84 vaddr=3ffb0000 size=03294h ( 12948)
 load
I (136) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=2388ch (145548)
 map
I (189) esp_image: segment 3: paddr=000438b4 vaddr=3ffb3294 size=00248h (   584)
 load
I (189) esp_image: segment 4: paddr=00043b04 vaddr=40080000 size=0ce64h ( 52836)
 load
I (216) esp_image: segment 5: paddr=00050970 vaddr=50000000 size=00010h (    16)
 load
I (223) boot: Loaded app from partition at offset 0x10000
I (223) boot: Disabling RNG early entropy source...
I (236) cpu_start: Pro cpu up.
I (237) cpu_start: Starting app cpu, entry point is 0x400810e0
0x400810e0: call_start_cpu1 at C:/Users/Vivek/Desktop/esp-idf/components/esp_sys
tem/port/cpu_start.c:155

I (0) cpu_start: App cpu up.
I (251) cpu_start: Pro cpu start user code
I (251) cpu_start: cpu freq: 160000000
I (251) cpu_start: Application information:
I (255) cpu_start: Project name:     app-template
I (261) cpu_start: App version:      1
I (265) cpu_start: Compile time:     Sep  7 2021 15:49:16
I (271) cpu_start: ELF file SHA256:  59a282c659a11336...
I (277) cpu_start: ESP-IDF:          v4.4-dev-2359-g58022f859-dirty
I (284) heap_init: Initializing. RAM available for dynamic allocation:
I (291) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (297) heap_init: At 3FFB3E50 len 0002C1B0 (176 KiB): DRAM
I (304) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (310) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (316) heap_init: At 4008CE64 len 0001319C (76 KiB): IRAM
I (324) spi_flash: detected chip: winbond
I (327) spi_flash: flash io: dio
I (333) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (342) example:take_picture: Starting nvs_flash_init
I (392) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:2
I (392) cam_hal: cam init ok
STEP-1
I (392) sccb: pin_sda 26 pin_scl 27
I (402) camera: Detected camera at address=0x21
I (402) ov7725: Mismatch PID=0x76
PID : 76
I (412) camera: Detected OV7670 camera
I (412) camera: Camera PID=0x76 VER=0x73 MIDL=0x7f MIDH=0xa2
STEP-2
STEP-3
I (462) esp32 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 3, dma
_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 12, dma_
buffer_size: 30720, image_size: 38400
I (462) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size:
3840, node_cnt: 8, total_cnt: 2
I (472) cam_hal: Allocating 19200 Byte frame buffer in OnBoard RAM
I (482) cam_hal: cam config ok 
```

Now I am able to compile and run example program. But Now I don't know How can i see image after getting image data into buffer. My code is very simple. Which is below.

How can I convert this fd->buf  ```      camera_fb_t *fb = esp_camera_fb_get();  ``` data into Image ??

```ruby

//=========================================================
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_GRAYSCALE, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_QQVGA,    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 12, //0-63 lower number means higher quality
    .fb_count = 1,       //if more than one, i2s runs in continuous mode. Use only with JPEG
	.fb_location = CAMERA_FB_IN_DRAM,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
};

//===========================================================

void app_main()
{
    ESP_LOGI(TAG,"Starting nvs_flash_init");
    nvs_flash_init();

	  if(ESP_OK != init_camera()) {
	        return;
	   }

    while (1)
    {
      
         ESP_LOGI(TAG, "Taking picture...");

        
        camera_fb_t *fb = esp_camera_fb_get();

        // use pic->buf to access the image
        ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", fb->len);

        esp_camera_fb_return(fb);

        vTaskDelay(5000 / portTICK_RATE_MS);
    }
}

```

me-no-dev, 2021-09-09:
that depends on your application. Based on the config above, you are getting 8bit per pixel grayscale data with QQVGA resolution. Those bytes are in `fb->buf`. You can then display them on LCD, convert them to jpeg and send over web, etc... there are examples in https://github.com/espressif/esp-who

viveky1794, 2021-09-10:
I am now successfully able to get image but I am getting this using python via converting byte array into image.

@python file

```ruby

arr = [ 255, 216, 255, 224, 0, 16, 74, 70, 73, 70, 0, 1, 1, 0, 0, 1, 0, 1, 0, 0, 255, 219, 0, 67, 0, 12, 8, 9, 10, 9, 7, 12, 10, 10, 10, 13, 13, 12, 14, 18, 30, 19, 18, 16, 16, 18, 36, 26, 27, 21, 30, 43, 38, 45, 44, 42, 38, 41, 41, 47, 53, 68, 58, 47, 50, 64, 51, 41, 41, 59, 81, 60, 64, 70, 73, 76, 77, 76, 46, 57, 84, 90, 83, 74, 89, 68, 75, 76, 73, 255, 219, 0, 67, 1, 13, 13, 13, 18, 16, 18, 35, 19, 19, 35, 73, 49, 41, 49, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 73, 255, 192, 0, 17, 8, 0, 120, 0, 160, 3, 1, 34, 0, 2, 17, 1, 3, 17, 1, 255, 196, 0, 31, 0, 0, 1, 5, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 255, 196, 0, 181, 16, 0, 2, 1, 3, 3, 2, 4, 3, 5, 5, 4, 4, 0, 0, 1, 125, 1, 2, 3, 0, 4, 17, 5, 18, 33, 49, 65, 6, 19, 81, 97, 7, 34, 113, 20, 50, 129, 145, 161, 8, 35, 66, 177, 193, 21, 82, 209, 240, 36, 51, 98, 114, 130, 9, 10, 22, 23, 24, 25, 26, 37, 38, 39, 40, 41, 42, 52, 53, 54, 55, 56, 57, 58, 67, 68, 69, 70, 71, 72, 73, 74, 83, 84, 85, 86, 87, 88, 89, 90, 99, 100, 101, 102, 103, 104, 105, 106, 115, 116, 117, 118, 119, 120, 121, 122, 131, 132, 133, 134, 135, 136, 137, 138, 146, 147, 148, 149, 150, 151, 152, 153, 154, 162, 163, 164, 165, 166, 167, 168, 169, 170, 178, 179, 180, 181, 182, 183, 184, 185, 186, 194, 195, 196, 197, 198, 199, 200, 201, 202, 210, 211, 212, 213, 214, 215, 216, 217, 218, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 255, 196, 0, 31, 1, 0, 3, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 255, 196, 0, 181, 17, 0, 2, 1, 2, 4, 4, 3, 4, 7, 5, 4, 4, 0, 1, 2, 119, 0, 1, 2, 3, 17, 4, 5, 33, 49, 6, 18, 65, 81, 7, 97, 113, 19, 34, 50, 129, 8, 20, 66, 145, 161, 177, 193, 9, 35, 51, 82, 240, 21, 98, 114, 209, 10, 22, 36, 52, 225, 37, 241, 23, 24, 25, 26, 38, 39, 40, 41, 42, 53, 54, 55, 56, 57, 58, 67, 68, 69, 70, 71, 72, 73, 74, 83, 84, 85, 86, 87, 88, 89, 90, 99, 100, 101, 102, 103, 104, 105, 106, 115, 116, 117, 118, 119, 120, 121, 122, 130, 131, 132, 133, 134, 135, 136, 137, 138, 146, 147, 148, 149, 150, 151, 152, 153, 154, 162, 163, 164, 165, 166, 167, 168, 169, 170, 178, 179, 180, 181, 182, 183, 184, 185, 186, 194, 195, 196, 197, 198, 199, 200, 201, 202, 210, 211, 212, 213, 214, 215, 216, 217, 218, 226, 227, 228, 229, 230, 231, 232, 233, 234, 242, 243, 244, 245, 246, 247, 248, 249, 250, 255, 218, 0, 12, 3, 1, 0, 2, 17, 3, 17, 0, 63, 0, 209, 248, 39, 255, 0, 34, 197, 239, 253, 125, 159, 253, 1, 107, 209, 43, 205, 190, 15, 93, 65, 103, 225, 11, 249, 238, 36, 9, 26, 93, 18, 73, 255, 0, 113, 106, 222, 161, 226, 203, 173, 78, 224, 218, 233, 234, 97, 136, 156, 110, 254, 38, 167, 82, 74, 42, 236, 218, 142, 30, 85, 93, 145, 221, 44, 209, 52, 134, 53, 145, 75, 142, 170, 15, 52, 250, 202, 210, 45, 83, 76, 211, 188, 235, 135, 253, 227, 13, 206, 204, 122, 123, 81, 109, 115, 121, 127, 112, 205, 11, 8, 173, 129, 224, 224, 18, 106, 83, 186, 184, 58, 73, 182, 226, 244, 93, 77, 83, 211, 140, 102, 176, 181, 109, 90, 235, 76, 195, 121, 59, 193, 236, 71, 21, 186, 6, 0, 25, 207, 189, 54, 88, 163, 154, 51, 28, 168, 174, 167, 168, 97, 154, 214, 148, 227, 25, 94, 74, 232, 228, 169, 77, 201, 166, 157, 172, 112, 183, 254, 39, 75, 228, 17, 186, 136, 199, 126, 107, 30, 73, 98, 114, 112, 194, 186, 173, 103, 194, 54, 146, 43, 75, 108, 222, 83, 31, 225, 110, 149, 193, 106, 186, 125, 238, 158, 196, 48, 99, 93, 146, 141, 26, 145, 247, 76, 212, 42, 71, 113, 215, 113, 161, 45, 211, 142, 248, 172, 139, 155, 124, 228, 246, 168, 39, 190, 185, 67, 130, 205, 199, 173, 82, 155, 84, 156, 130, 15, 53, 195, 82, 148, 205, 147, 190, 228, 210, 91, 175, 124, 10, 140, 196, 153, 235, 199, 161, 172, 247, 187, 158, 67, 212, 254, 21, 36, 17, 93, 204, 192, 40, 35, 61, 235, 53, 11, 124, 76, 181, 22, 222, 138, 230, 246, 147, 45, 189, 172, 162, 71, 219, 145, 207, 21, 218, 39, 143, 110, 165, 129, 45, 172, 237, 151, 120, 27, 67, 99, 53, 196, 233, 154, 35, 72, 20, 207, 41, 0, 250, 26, 237, 116, 91, 107, 45, 61, 151, 108, 96, 183, 169, 173, 161, 138, 161, 75, 75, 115, 29, 52, 178, 202, 181, 61, 233, 104, 191, 19, 162, 240, 164, 90, 171, 206, 247, 119, 100, 133, 151, 239, 6, 28, 159, 74, 234, 43, 50, 222, 253, 99, 137, 65, 0, 15, 74, 179, 111, 118, 37, 148, 167, 229, 85, 90, 167, 181, 151, 54, 134, 78, 131, 130, 178, 90, 34, 205, 45, 20, 86, 38, 97, 69, 20, 80, 7, 207, 30, 31, 185, 156, 233, 173, 104, 174, 222, 89, 125, 219, 1, 227, 62, 181, 233, 94, 20, 211, 96, 176, 182, 109, 70, 252, 133, 10, 50, 50, 107, 136, 240, 29, 148, 79, 4, 151, 119, 13, 182, 24, 142, 88, 158, 157, 235, 114, 247, 83, 184, 214, 110, 86, 222, 217, 74, 219, 33, 194, 175, 175, 185, 168, 173, 45, 121, 164, 246, 183, 228, 123, 116, 233, 183, 74, 48, 91, 181, 248, 127, 95, 129, 209, 62, 167, 63, 136, 47, 132, 81, 169, 91, 96, 120, 30, 190, 245, 215, 218, 192, 45, 237, 82, 36, 0, 96, 126, 181, 131, 225, 251, 40, 52, 219, 125, 243, 17, 191, 29, 42, 123, 237, 120, 70, 118, 198, 66, 254, 52, 160, 154, 92, 243, 234, 115, 226, 32, 234, 201, 82, 164, 189, 212, 109, 150, 88, 147, 50, 56, 250, 154, 195, 213, 252, 68, 150, 170, 86, 220, 7, 110, 153, 174, 115, 81, 215, 39, 148, 182, 198, 36, 158, 245, 140, 205, 52, 185, 103, 108, 145, 223, 52, 165, 89, 116, 46, 158, 3, 1 [... cut ...]
       printf("=======================\n\n");

     //   fmt2jpg_cb( fb->buf, fb->len, fb->width,fb->height, fb->format, camera_config.jpeg_quality , jpgCallBack, NULL);

        frame2jpg_cb(fb,  camera_config.jpeg_quality, jpgCallBack, NULL );

        printf("\n\n=======================\n\n");

        // use pic->buf to access the image
        ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", fb->len);

        esp_camera_fb_return(fb);

        vTaskDelay(5000 / portTICK_RATE_MS);
    }
}

```