espressif/esp32-camera #316: Can't receive a RAW Full HD frame in high freq

Arsanios, 2021-09-07:
Hi,
I am implementing a driver for GC2093 and i am facing some issues with ESP32-CAM
I need to receive a 1920x1080 RAW-8 from the sensor, which are 2,073,600 bytes.
The sensor's PLL is set to 1:1 which means that XCLK = PCLK and i cannot change it
If i go faster that 7.5 MHz i can't receive the full frame,  it always stuck in

```
void IRAM_ATTR ll_cam_send_event(cam_obj_t *cam, cam_event_t cam_event, BaseType_t * HPTaskAwoken)
{
    if (xQueueSendFromISR(cam->event_queue, (void *)&cam_event, HPTaskAwoken) != pdTRUE) {
        ll_cam_stop(cam);
        cam->state = CAM_STATE_IDLE;
        ESP_EARLY_LOGE(TAG, "EV-OVF %d",cam_event);
    }
}
```

I am trying to go on XCLK = 8 MHz, so PCLK = 8MHz.

I want to know if it is a limit regarding the DMA or it is something that can be adjusted?

Thanks

Arsanios, 2021-09-07:
Here is my camera configuration

```
static camera_config_t camera_config = {
    .pin_pwdn = PWDN_GPIO_NUM,
    .pin_reset = RESET_GPIO_NUM,
    .pin_xclk = XCLK_GPIO_NUM,
    .pin_sscb_sda = SIOD_GPIO_NUM,
    .pin_sscb_scl = SIOC_GPIO_NUM,

    .pin_d7 = Y9_GPIO_NUM,
    .pin_d6 = Y8_GPIO_NUM,
    .pin_d5 = Y7_GPIO_NUM,
    .pin_d4 = Y6_GPIO_NUM,
    .pin_d3 = Y5_GPIO_NUM,
    .pin_d2 = Y4_GPIO_NUM,
    .pin_d1 = Y3_GPIO_NUM,
    .pin_d0 = Y2_GPIO_NUM,
    .pin_vsync = VSYNC_GPIO_NUM,
    .pin_href = HREF_GPIO_NUM,
    .pin_pclk = PCLK_GPIO_NUM,

    .xclk_freq_hz = 8000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_RAW, 
    .frame_size = FRAMESIZE_FHD,   

    .fb_count = 1
};
```

me-no-dev, 2021-09-08:
I assume that you are using ESP32? 2MBytes is quite a lot. On ESP32 the image data can not be directly used, because the format that it comes in is a bit different (this has changed in ESP32-S2 and ESP32-S3), so when getting an image, the DMA buffers are sent to a task, which extracts the usable data from the samples and copies it to PSRAM (the frame buffer). Since you are getting EV-OFV, that means that the task does not have enough time to copy everything from DMA. Lowering XCLK in effect lowers the rate at which data comes (and frame rate) so that is why you see it succeed with lower frequencies.

things to check in `idf.py menuconfig`:
- Flash frequency is 80MHz
- PSRAM frequency is 80MHz
- CPU frequency is 240MHz
- FreeRTOS Tick Rate is 1000Hz

Arsanios, 2021-09-08:
Thank you, 
My flash and psram frequencies were set on 40MHz instead of 80MHz

Now it is working

closed 2021-09-08 by me-no-dev

Arsanios, 2021-09-14:
@me-no-dev I want to ask if is it possible that the i2s receive in a wrong way for high freq pclk like 8 Mhz ?

me-no-dev, 2021-09-24:
- I2S has it's limitations on what it can handle
- Newer chips are able to get the frames much easier since they do not require the task to extract each sample and copy it to PSRAM
- Not sure why you mention PLL_D2_CLK? It has nothing to do with XCLK and what PCLK the ESP32 can handle. I2S on ESP32 runs off of APB_CLK which is 80MHz