espressif/esp32-camera #321: JPG image exif data / meta data

npsantini, 2021-09-29:
Is there a way to set exif or meta data when the image is captured from an ESP32-cam? If so can some provide an example? The specific information I need is need is the time the image was taken.

At the moment it appears the only information available is the last modified date which isn't useful since the images are uploaded through ftp to a web server. So that time is when the image was uploaded and not when it was taken.

Any help I can get with this is extremely appreciated!

TDA-2030, 2021-10-09:
Every framebuffer from `esp_camera_get_fb` possess a timestamp, it may help you.
https://github.com/espressif/esp32-camera/blob/3022601f876cc158f98c168624a701978509ee43/driver/include/esp_camera.h#L138

closed 2021-12-17 by github-actions[bot]

step135, 2024-08-31:
Did you find any solution?

mvaranda, 2025-03-20:
I added libexif for getting metadata as a component for it. I was searching for a simpler solution but I guess there is none (other than using external lib).

mvaranda, 2025-03-20:
https://github.com/Varanda-Labs/ESP32-CAM-METADATA