espressif/esp32-camera #322: esp-32 Camera probe failed with error 0x20004

Themis1010, 2021-10-04:
E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
I have this board
[image]
[image]

with this code for control a rc car thourgh internet:
`//Viral Science www.youtube.com/c/viralscience  www.viralsciencecreativity.com
//ESP32 Camera Surveillance Car

#include "esp_camera.h"
#include <WiFi.h>

//
// WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
//            or another board which has PSRAM enabled
//
// Adafruit ESP32 Feather

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_M5STACK_PSRAM
#define CAMERA_MODEL_AI_THINKER

const char* ssid = "COSMOTE-3B2FDC";   //Enter SSID WIFI Name
const char* password = "6h4tVxhSfJj9KQEs";   //Enter WIFI Password`

and goes on the code
and with that features:
[image]
what is the fault and i get this error?

TDA-2030, 2021-10-09:
What model of sensor do you plug into the circuit board? It seems you are using an unsupported sensor

Themis1010, 2021-10-09:
> What model of sensor do you plug into the board? It seems you are using an unsupported sensor

I program it with FT232RL FTDI USB converter if you mean that. I don't have any other sensor on.

TDA-2030, 2021-10-09:
@Themis1010 I mean the camera sensor

TDA-2030, 2021-10-09:
oh, it looks like OV2640. Then the problem may be that the pin is not configured correctly or the sensor is broken

closed 2021-12-14 by Themis1010