espressif/esp32-camera #326: sc031gs with esp32s3

PureHing, 2021-10-20:
@me-no-dev 
Hi,
Can sc031gs work on esp32s3？If so, can the frame rate reach 120fps?

closed 2021-12-23 by PureHing

weilian1977, 2022-03-18:
> Driver should work on S3 yes. FPS performance would/should be the same on all chips. Unfortunately I do not have the sensor here, so I can not comment on how many FPS would be achievable :)

where is the Driver？

me-no-dev, 2022-03-22:
this repository is the driver.

weilian1977, 2022-03-24:
can provide the driver link？

me-no-dev, 2022-03-24:
link? you are literally commenting on an issue in the driver's repository. You are on the link :)

weilian1977, 2022-03-24:
emm，I'm not very familiar with githu. But exactly I can not find sc031gs file  in the camera/tree/master/sensors/private_include/
or esp32-camera/tree/feat/add_libjpeg/sensors/private_include or esp32-amera/tree/feature/lcd_cam_hw/sensors/private_include

weilian1977, 2022-03-28:
en，you said driver is esp32-camera/tree/master/driver？

WangYuxin-esp, 2022-09-26:
@weilian1977 @PureHing The SC031GS driver has been released.

weilian1977, 2022-09-26:
I had done it two months ago, But I will test it thanks a lot！

bedz15, 2025-12-31:
Did you guys make your own devboard with a level shifter to use with the SC031GS, if not what board are you using?

bedz15, 2026-01-02:
@WangYuxin-esp Do you know any good boards for the SC031GS?

bedz15, 2026-02-08:
@WangYuxin-esp Don't the SC035HGS and OV9281 drivers only support MIPI though? Also those drivers don't seem to be in the esp32-camera repo by default unless I missed them