espressif/esp32-camera #330: OV2640, JPEG -> cam_hal: NO-SOI

GrobiG, 2021-11-05:
Hi,

I am having some issues with a esp32-cam (ai-thinker clone) module and its OV2640 camera,
when I set the pixel format to PIXFORMAT_JPEG it always ends up with a lot of cam_hal: NO-SOI messages, a final cam_hal: Failed to get the frame on time! and than the device reboots : Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

It doesn't matter what resolution or xclk_freq I try it is always the same outcome with pixel format = JPEG.
E.g. GrayScale mode with HD resolution works absolutely fine!

Is it possible that the camera I got in my bundle ommits the jpeg header in the frame buffer?
The print on its ribbon cable says:
DC-26
40-V3
that is basically all the information I got about it.

on this run:
    config.xclk_freq_hz = 10000000;
    config.pixel_format = PIXFORMAT_JPEG; //YUV422,GRAYSCALE,RGB565,JPEG
    config.frame_size = FRAMESIZE_QVGA;    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG
    config.jpeg_quality = 12; //0-63 lower number means higher quality
    config.fb_count = 2;       //if more than one, i2s runs in continuous mode. Use only with JPEG
    config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
...

Monitor output:
I (27) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (27) boot: compile time 15:03:03
I (27) boot: chip revision: 1
I (30) boot.esp32: SPI Speed      : 80MHz
I (35) boot.esp32: SPI Mode       : DIO
I (39) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=087f8h ( 34808) map
I (106) esp_image: segment 1: paddr=00018820 vaddr=3ffb0000 size=026d4h (  9940) load
I (110) esp_image: segment 2: paddr=0001aefc vaddr=40080000 size=0511ch ( 20764) load
I (122) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1f16ch (127340) map
I (164) esp_image: segment 4: paddr=0003f194 vaddr=4008511c size=098f8h ( 39160) load
I (178) esp_image: segment 5: paddr=00048a94 vaddr=50000000 size=00010h (    16) load
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (198) psram: This chip is ESP32-D0WD
I (199) spiram: Found 64MBit SPI RAM device
I (199) spiram: SPI RAM mode: flash 80m sram 80m
I (202) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (209) cpu_start: Pro cpu up.
I (213) cpu_start: Starting app cpu, entry point is 0x40080ed0
0x40080ed0: call_start_cpu1 at C:/Users/grobi/esp/esp-idf/components/esp_system/port/cpu_start.c:141

I (0) cpu_start: App cpu up.
I (692) spiram: SPI SRAM memory test OK
I (700) cpu_start: Pro cpu start user code
I (700) cpu_start: cpu freq: 240000000
I (700) cpu_start: Application information:
I (703) cpu_start: Project name:     esp32-cam_test
I (708) cpu_start: App version:      1
I (712) cpu_start: Compile time:     Nov  5 2021 15:15:19
I (719) cpu_start: ELF file SHA256:  be7eb7124421e225...
I (725) cpu_start: ESP-IDF:          v4.3.1-dirty
I (730) heap_init: Initializing. RAM available for dynamic allocation:
I (737) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (743) heap_init: At 3FFB3260 len 0002CDA0 (179 KiB): DRAM
I (749) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (756) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (762) heap_init: At 4008EA14 len 000115EC (69 KiB): IRAM
I (768) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (777) spi_flash: detected chip: generic
I (781) spi_flash: flash io: dio
I (786) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (795) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (805) example:take_picture: Mounting SD card...
I (845) example:take_picture: SD card mount successfully!
I (845) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (855) cam_hal: cam init ok
I (855) sccb: pin_sda 26 pin_scl 27
I (855) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (905) camera: Detected camera at address=0x30
I (995) camera: Detected OV2640 camera
I (995) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (2605) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 3
I (2605) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (2605) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (2615) cam_hal: cam config ok
I (2615) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
W (4065) cam_hal: NO-SOI
W (4105) cam_hal: NO-SOI
W (4145) cam_hal: NO-SOI
I (4155) CAMERA:: Taking picture and storing in /sdcard/pic_0.dat ...
W (4185) cam_hal: NO-SOI
...
W (8545) cam_hal: NO-SOI
W (8555) cam_hal: Failed to get the frame on time!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

GrobiG, 2021-11-05:
ok, turns out the reboot was entirely my fault but the:
cam_hal: NO-SOI
..
cam_hal: Failed to get the frame on time!
issue remains.

me-no-dev, 2021-11-06:
have you tried other resolutions? this is a strange issue... it's not uncommon to get a board with bad image sensor. Do other image formats work?

GrobiG, 2021-11-06:
Yes, I have tried it with every resolution for this 2MP camera sensor, all the same with jpeg I always get NO-SOI as soon as the camera is initialized and running.
I also tried it with rgb565 and grayscale, it works in every resolution.
Noticed grayscale looks way better than rgb565.
In rgb565 the colors are bit wonky and kind of dithered but so far I did not try it with natural light.
Btw frame2jpg seems to be broken, with framesizes XGA, HD, SXGA & UXGA the images are corrupted (the framebuffer contains uncorrupted images), the smaller resolutions work fine.
I have ordered another one of these ai-thinker esp32-cam clones from a different source in hope I get a different camera module with it.

GrobiG, 2021-11-10:
So I did some more tests after two new modules have arrived I came to the conclusion the camera is the culprit and NOT the software!
One of the modules came with the same camera, same print on its cable as the one where JPEG does not work, switching it to JPEG works without a hitch !?!!...????!
I get one or two cam_hal: EV-EOF-OVF at the start but no further issues after that.
Also the lens on that one wasn't glued down so it was totally out of focus. During correcting this (had to turn it a lot to the left) I found there are areas where the colors get messed up just like I see on the module where JPEG isn't working (of course there the lens is glued down so no correcting that unless I can somehow get rid of the glue without damaging the thing).
The second module had a totally different camera (8225N V2.0 171026) it says on its cable.
That one seems to be the "real deal", JPEG works and the picture quality is soooooooo much better compared to the other ones!!!
All three cameras are detected as:
camera: Detected camera at address=0x30
camera: Detected OV2640 camera
camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
so it just comes down to luck with those cheap-o-modules, who could have guessed that? :)

closed 2021-11-10 by GrobiG

jomjol, 2021-11-10:
Thanks for you detailed analyis and especially for sharing your results. I'm quite often in discussion of not working systems and the acceptance, that it might be hardware related is hard to achieve :-).

me-no-dev, 2021-11-12:
this is mostly because OV2640 is EOL and not supported/manufactured from a loong time. What is left on the webs are some stale batches of sensors with questionable quality and are mostly hit and miss. Our future chips will support MIPI so we will have newer and better options for sensors

manishtomar-cpi, 2024-10-01:
> Hi,
>
> I am having some issues with a esp32-cam (ai-thinker clone) module and its OV2640 camera, when I set the pixel format to PIXFORMAT_JPEG it always ends up with a lot of cam_hal: NO-SOI messages, [...]