espressif/esp32-camera #332: Chase IDF change making periph enable/disable APIs private

abgandar, 2021-11-13:
Driver currently does not build with ESP-IDF from HEAD any more because periph enable API is now private:

> /esp-idf/components/esp32-camera/target/esp32/ll_cam.c: In function 'll_cam_config':
> /esp-idf/components/esp32-camera/target/esp32/ll_cam.c:278:5: error: implicit declaration of function 'periph_module_enable'; did you mean 'esp_intr_enable'? [-Werror=implicit-function-declaration]
>     periph_module_enable(PERIPH_I2S0_MODULE);
>     ^~~~~~~~~~~~~~~~~~~~
>     esp_intr_enable
> /esp-idf/components/esp32-camera/target/esp32/ll_cam.c:278:26: error: 'PERIPH_I2S0_MODULE' undeclared (first use in this function); did you mean 'PERIPHS_IO_MUX_MTDO_U'?
>     periph_module_enable(PERIPH_I2S0_MODULE);
>                          ^~~~~~~~~~~~~~~~~~
>                          PERIPHS_IO_MUX_MTDO_U

Fix should be easy according to [commit message](https://github.com/espressif/esp-idf/commit/16677b0d3cc4cf1c650fa3d08e39d89630f5af77):
> this is still possible by including the header in this way:
> "esp_private/peripheral_ctrl.h"

abgandar, 2021-11-14:
Note that the commit message referenced above uses the wrong header name. Build is fixed by adding `#include "esp_private/periph_ctrl.h"` in `target/private_include/ll_cam.h`.

me-no-dev, 2021-11-15:
Thanks for finding this as well as resolving it :) Should be now ok

closed 2021-11-15 by me-no-dev