espressif/esp32-camera #342: Pin Map for ESP32-S2 SOC

lukedukeus, 2022-01-05:
I plan on using this library with an ESP32-S2FN4R2 or  ESP32-S2. What pin mapping would I need for this?

me-no-dev, 2022-01-18:
it's up to you. You specify it when starting the driver.

closed 2022-01-18 by me-no-dev

lukedukeus, 2022-01-18:
Right, but Im sure some pins need to be used for specific things. On the ESP32-CAM boards, pins 20 and 22 are connected to I2C. Is this a requriement, or do all pins just need to be gpio pins?
[image]

me-no-dev, 2022-01-18:
The camera itself needs I2C, but you can pick which ESP32 GPIOs you would use for that. As opposed to other micros, on the ESP32 just about any peripheral can be connected to any GPIO of choice. Of course there are some bootstrap pins that you want to ensure are not pulled in the wrong direction on boot, but you can test all that very easy on a breadboard first.