espressif/esp32-camera #343: jpg2rgb888 (and esp_jpg_decode) return wrong colors

y3nd, 2022-01-07:
Hello,

We have an applications with esp32-cam where we need to have the same color in the JPEG and in the converted rgb888 file, it should be possible (approximately) because JPEG color space is YCrCb converted to RGB.
However, after testing we are getting a blue-ish wrong colored result from jpg2rgb888, the pixel colors are not the same even though we get a similar image to original JPEG.

Here are the result we have :
Converted from JPEG
[image]
Original JPEG
[image]

Is this a bug in the JPEG decoder ?

me-no-dev, 2022-01-18:
I'm not a color specialist, but I would not call this exactly a bug, but rather wrong color handling somewhere. The JPEG decoder is written with low memory in mind and in most cases resides in the chip's ROM to speed it up even more. It is possible that there is some wrong calculation done in this library as well. If you can find a pattern exactly what is the color difference, that could help narrow down the cause.

closed 2022-03-28 by github-actions[bot]

AIWintermuteAI, 2022-04-07:
I can confirm the issue. 
[image]
Looks like it returns BGR instead of RGB - the color of the robot legs in the picture is yellow.
I'm using the following function for conversion:
```
bool converted = fmt2rgb888(fb->buf, fb->len, PIXFORMAT_JPEG, image);
```