espressif/esp32-camera #354: CameraWebServer OV2460 bad image

rtek1000, 2022-01-19:
Hello,

I would like to know if the camera is defective, because the image is very bad, it seems to be very low resolution and very dark too.

I bought two ESP32-CAM from the same seller and both kits make these bad images:

[image]

The expectation was to get a more defined image like this from someone else:

[image]

Image source: http://blog.marxy.org/2019/11/esp32-cam-notes-to-self.html

Usage tutorial: https://www.elementzonline.com/blog/esp32-cam-video-streaming-face-recognition-tutorial

me-no-dev, 2022-01-20:
Did you give a bit of time for the camera to autobalance it's gains and all? Looks like you got the image too early after init?

rtek1000, 2022-01-20:
Yes I did, I waited some time, I tried it several times, and I also noticed that it seems to have a lot of dead pixels.

How long does it take for gain adjustment?

Is there any way to know if the camera is fake?

me-no-dev, 2022-01-20:
I would not call it fake, but rather defective. The issue is mainly the OV2640 has been EOL for quite some time, so companies get whatever they can find. Sometimes they get batches with some bad sensors, bad lenses, etc... Generally in 5 seconds or less you should have a good image.

closed 2022-03-29 by github-actions[bot]