espressif/esp32-camera #355: esp32-cam IDF v5.0-dev issue

vasilKubica, 2022-01-22:
Hello,
I'm trying to flash take_picture.c example with IDF v5.0-dev to esp32-cam board and it constantly reboots. With IDF v4.1 works ok.
Here is output:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6980
load:0x40078000,len:14432
ho 0 tail 12 room 4
load:0x40080400,len:4420
0x40080400: _init at ??:?

entry 0x400806ac
I (29) boot: ESP-IDF v5.0-dev-797-g279c8aeb8a 2nd stage bootloader
I (29) boot: compile time 21:30:56
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (46) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (55) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (83) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (91) boot:  2 factory          factory app      00 00 00020000 00100000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=10e48h ( 69192) map
I (137) esp_image: segment 1: paddr=00030e70 vaddr=3ffb0000 size=035d0h ( 13776) load
I (142) esp_image: segment 2: paddr=00034448 vaddr=40080000 size=0bbd0h ( 48080) load
I (158) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=27694h (161428) map
I (203) esp_image: segment 4: paddr=000676bc vaddr=4008bbd0 size=0540ch ( 21516) load
I (210) esp_image: segment 5: paddr=0006cad0 vaddr=50000000 size=00010h (    16) load
I (219) boot: Loaded app from partition at offset 0x20000
I (219) boot: Disabling RNG early entropy source...
I (232) psram: This chip is ESP32-D0WD
I (233) spiram: Found 64MBit SPI RAM device
I (233) spiram: SPI RAM mode: flash 80m sram 80m
I (235) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (242) cpu_start: Pro cpu up.
I (246) cpu_start: Starting app cpu, entry point is 0x40081334
0x40081334: call_start_cpu1 at /home/tata/esp/esp-idf/components/esp_system/port/cpu_start.c:164

I (239) cpu_start: App cpu up.
I (746) spiram: SPI SRAM memory test OK
I (754) cpu_start: Pro cpu start user code
I (754) cpu_start: cpu freq: 240000000 Hz
I (754) cpu_start: Application information:
I (757) cpu_start: Project name:     camera
I (762) cpu_start: App version:      1
I (766) cpu_start: Compile time:     Jan 21 2022 21:31:32
I (772) cpu_start: ELF file SHA256:  9b4758d683cab353...
I (778) cpu_start: ESP-IDF:          v5.0-dev-797-g279c8aeb8a
I (785) heap_init: Initializing. RAM available for dynamic allocation:
I (792) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (798) heap_init: At 3FFB3F68 len 0002C098 (176 KiB): DRAM
I (804) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (810) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (817) heap_init: At 40090FDC len 0000F024 (60 KiB): IRAM
I (823) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (831) spi_flash: detected chip: generic
I (836) spi_flash: flash io: qio
I (840) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (855) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (863) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e3559  PS      : 0x00060b30  A0      : 0x800e1cc9  A1      : 0x3ffb7350  
0x400e3559: gpio_ll_iomux_in at /home/tata/esp/esp-idf/components/hal/esp32/include/hal/gpio_ll.h:561
 (inlined by) gpio_iomux_in at /home/tata/esp/esp-idf/components/driver/gpio.c:718

A2      : 0x000000e0  A3      : 0x40100000  A4      : 0x3ffb8dc8  A5      : 0x3ff49010  
A6      : 0x00000000  A7      : 0x02000000  A8      : 0x80000000  A9      : 0x00000001  
A10     : 0x00013ffc  A11     : 0x00000000  A12     : 0x3ffb3888  A13     : 0x00060b23  
A14     : 0x00060b20  A15     : 0x00000001  SAR     : 0x00000003  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80000000  LBEG    : 0x40087f8e  LEND    : 0x40087f99  LCOUNT  : 0x00000000  
0x40087f8e: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x40087f99: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x400e3556:0x3ffb73500x400e1cc6:0x3ffb7370 0x400d9a88:0x3ffb73d0 0x400d9349:0x3ffb7400 0x400d90c2:0x3ffb7440 0x400d90e3:0x3ffb7460 0x400f74be:0x3ffb7480 0x4008d225:0x3ffb74a0 
0x400e3556: gpio_ll_iomux_in at /home/tata/esp/esp-idf/components/hal/esp32/include/hal/gpio_ll.h:561
 (inlined by) gpio_iomux_in at /home/tata/esp/esp-idf/components/driver/gpio.c:718

0x400e1cc6: ll_cam_set_pin at /home/tata/esp/esp-idf/components/esp32-camera/target/esp32/ll_cam.c:371

0x400d9a88: cam_init at /home/tata/esp/esp-idf/components/esp32-camera/driver/cam_hal.c:319 (discriminator 2)

0x400d9349: esp_camera_init at /home/tata/esp/esp-idf/components/esp32-camera/driver/esp_camera.c:224

0x400d90c2: init_camera at /home/tata/esp/camera/main/main.c:128

0x400d90e3: app_main at /home/tata/esp/camera/main/main.c:140

0x400f74be: main_task at /home/tata/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x4008d225: vPortTaskWrapper at /home/tata/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:132

ELF file SHA256: 9b4758d683cab353

Rebooting...
ets Jun  8 2016 00:22:57
```

me-no-dev, 2022-01-24:
We have not tested/verified with 5.0. It's a bit too early and seems that the issue might be in IDF itself. Please use 4.4 or earlier.

igrr, 2022-01-24:
@me-no-dev FWIW, this seems to be related to the recent commit: https://github.com/espressif/esp32-camera/commit/b8ed779427b31d883422c01641b4112cff974393#diff-733311667eee41d4da2ab2ef6b9ac6a603fcc699074d24a701b17bd48a31f6c3R27

I'm not sure why you have defined gpio_matrix_in to gpio_iomux_in? matrix != iomux

me-no-dev, 2022-01-24:
@igrr I will have a look

me-no-dev, 2022-05-31:
Everything is fixed in the latest release (v2.0.1). Closing :)