espressif/esp32-camera #358: Stack Overflow bzw Memory "CORRUPTED"

moefear85, 2022-02-04:
I have the following camera capture task running:

`void capture_task()
{
    mwifi_data_type_t data_type={};
    char* magic_string="magic start";

    while(true)
    {
        if (!stream)
        {
            printf("No Sources. Waiting...\n");
            xSemaphoreTake(camera_semaphore,portMAX_DELAY);
            printf("Starting Capture\n");
        }
        
        camera_fb_t * fb = esp_camera_fb_get();
        /*
        printf("Frame Size: %d\n",fb->len);

        int q= fb->len/MWIFI_PAYLOAD_LEN;
        int r=fb->len%MWIFI_PAYLOAD_LEN;

        //printf("Q: %d, R: %d\n",q,r);        
        uint8_t magic_buf[sizeof(magic_string)+sizeof(camera_fb_t)];
        strcpy((char*)magic_buf,magic_string);
        memcpy(magic_buf+sizeof(magic_string),fb,sizeof(camera_fb_t));
        mwifi_write(addr_display,&data_type,magic_buf,sizeof(magic_buf),true);
    
        for(int i=0;i<q;i++)
        {
            mwifi_write(addr_display,&data_type,fb->buf+MWIFI_PAYLOAD_LEN*i,MWIFI_PAYLOAD_LEN,true);
        }
        if (r) mwifi_write(addr_display,&data_type,fb->buf+MWIFI_PAYLOAD_LEN*q,r,true);
        */
        esp_camera_fb_return(fb);        
    }
}`

Note most of the logic is gutted out to find the source of the error. But even just obtaining & releasing the framebuffer causes the error. The task has tskIDLE_PRIORITY. If I further comment out the framebuffer lines, the error disappears, even if I use a tight printf loop.

The error is:

> ***ERROR*** A stack overflow in task capture_task has been detected.

>Backtrace:0x40081db9:0x3ffcf5200x4008cf6d:0x3ffcf540 0x40090dd5:0x3ffcf560 0x4008ede4:0x3ffcf5e0 0x4008d06c:0x3ffcf600 0x4008d01e:0xa5a5a5a5  |<-CORRUPTED
> 0x40081db9: panic_abort at /home/moefear85/Source/esp-idf-v4.4/components/esp_system/panic.c:402

> 0x4008cf6d: esp_system_abort at /home/moefear85/Source/esp-idf-v4.4/components/esp_system/esp_system.c:121
> 0x40090dd5: vApplicationStackOverflowHook at /home/moefear85/Source/esp-idf-v4.4/components/freertos/port/xtensa/port.c:394
> 0x4008ede4: vTaskSwitchContext at /home/moefear85/Source/esp-idf-v4.4/components/freertos/tasks.c:3506
> 0x4008d06c: _frxt_dispatch at /home/moefear85/Source/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:436
> 0x4008d01e: _frxt_int_exit at /home/moefear85/Source/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:231

> ELF file SHA256: 1fbfa20dcf136620
> CPU halted.

Sometimes I get a stack overflow error instead (on successive reboots) even without changing the code.

me-no-dev, 2022-02-05:
in xTaskCreate use larger stack size. 4K will do. Then do not `uint8_t magic_buf[sizeof(magic_string)+sizeof(camera_fb_t)];`! Instead:
```cpp
uint8_t * magic_buf = (uint8_t *)malloc(sizeof(magic_string)+sizeof(camera_fb_t));
if(magic_buf == NULL){
    esp_camera_fb_return(fb);
    continue;
}
// rest of code
free(magic_buf);
esp_camera_fb_return(fb);
```

closed 2022-02-05 by me-no-dev

moefear85, 2022-02-22:
OK I'll try it later... but your answer doesn't really shed any light on why/what is going wrong. Btw, I've tried stack sizes all the way up to 10240bytes. It didn't seem to help. I was hoping you could provide insight into decoding the error output. Is there anything there that is telling you it is definately a stack overflow problem? I've read the docs on decoding the errors, but it's either insufficient, or simply too complex for anybody to really fathom on-the-fly each time the chip panics. The decoding always seems partial and leaves alot to be desired. Is there any tool that can process it and always provide much more detailed info?

me-no-dev, 2022-02-23:
the line that I told you to change will try in your code to allocate it on the stack and you overflow. If you allocate it manually, it will go on the heap and will not cause the error. Error has nothing to do with the camera itself in any way. You are just trying to allocate more space on the stack than it's available for this task