espressif/esp32-camera #363: CameraWebServer example Get Still function serves old frames

maximemoreillon, 2022-02-16:
Hello,

Thank you very much for your work on the camera support for the ESP32.

I recently ran into a problem when dealing with frame capture. It happens to all sketches I have which capture frames, but also the CameraWebServer example sketch provided for Arduino, with default settings, which should be the easiest to replicate.

On the web interface of the CameraWebServer example, when I click "Get Still", the frame I get is not what is currently in front of the camera, but what used to be when I previously clicked the button. In other words, the ESP32 is serving old frames that are somehow still in memory and is thus always between 3 to 4 frames behind.

I currently run version 2.0.2 of the esp32 boards in the Arduino board manager and using an AI thinker board.

Any idea how to solve this?

me-no-dev, 2022-02-17:
In the camera config structure, change `grab_mode` to `CAMERA_GRAB_LATEST`: https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h#L126 and https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h#L83 :)

closed 2022-02-17 by me-no-dev

maximemoreillon, 2022-02-17:
That solved my problem! Thanks so much!