espressif/esp32-camera #364: error at "gpio_ll_iomux_in"

zhongtaifeng, 2022-02-17:
I (27) boot: ESP-IDF v5.0-dev-1592-g03fb3fd25b 2nd stage bootloader
I (27) boot: compile time 19:04:06
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 80MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00177000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=10da4h ( 69028) map
I (132) esp_image: segment 1: paddr=00020dcc vaddr=3ffb0000 size=03664h ( 13924) load
I (137) esp_image: segment 2: paddr=00024438 vaddr=40080000 size=0bbe0h ( 48096) load
I (154) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=27500h (161024) map
I (203) esp_image: segment 4: paddr=00057528 vaddr=4008bbe0 size=05378h ( 21368) load
I (211) esp_image: segment 5: paddr=0005c8a8 vaddr=50000000 size=00010h (    16) load
I (220) boot: Loaded app from partition at offset 0x10000
I (220) boot: Disabling RNG early entropy source...
I (232) psram: This chip is ESP32-D0WD
I (233) spiram: Found 64MBit SPI RAM device
I (233) spiram: SPI RAM mode: flash 80m sram 80m
I (238) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (245) cpu_start: Pro cpu up.
I (249) cpu_start: Starting app cpu, entry point is 0x40081348
I (242) cpu_start: App cpu up.
I (749) spiram: SPI SRAM memory test OK
I (756) cpu_start: Pro cpu start user code
I (757) cpu_start: cpu freq: 240000000 Hz
I (757) cpu_start: Application information:
I (760) cpu_start: Project name:     esp_ov2640
I (765) cpu_start: App version:      1
I (769) cpu_start: Compile time:     Feb 17 2022 19:03:59
I (775) cpu_start: ELF file SHA256:  c426b3924f79278e...
I (781) cpu_start: ESP-IDF:          v5.0-dev-1592-g03fb3fd25b
I (788) heap_init: Initializing. RAM available for dynamic allocation:
I (795) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (801) heap_init: At 3FFB4008 len 0002BFF8 (175 KiB): DRAM
I (807) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (814) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (820) heap_init: At 40090F58 len 0000F0A8 (60 KiB): IRAM
I (826) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (835) spi_flash: detected chip: generic
I (839) spi_flash: flash io: dio
I (844) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (854) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e34a5  PS      : 0x00060230  A0      : 0x800e1c8d  A1      : 0x3ffb6c00  
0x400e34a5: gpio_ll_iomux_in at /Users/.esp/esp-idf/components/hal/esp32/include/hal/gpio_ll.h:561
 (inlined by) gpio_iomux_in at /Users/.esp/esp-idf/components/driver/gpio.c:718

A2      : 0x000000e0  A3      : 0x40100000  A4      : 0x3ffb7df8  A5      : 0x3ff49010  
A6      : 0x00000000  A7      : 0x02000000  A8      : 0x80000000  A9      : 0x00000001  
A10     : 0x00013ffc  A11     : 0x00000000  A12     : 0x3ffb3920  A13     : 0x00060223  
A14     : 0x00060220  A15     : 0x00000001  SAR     : 0x00000003  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x80000000  LBEG    : 0x40087f42  LEND    : 0x40087f4d  LCOUNT  : 0x00000000  
0x40087f42: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x40087f4d: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x400e34a2:0x3ffb6c000x400e1c8a:0x3ffb6c20 0x400d9ad4:0x3ffb6c80 0x400d9411:0x3ffb6cb0 0x400d91c2:0x3ffb6cf0 0x400d91e3:0x3ffb6d10 0x400f7328:0x3ffb6d30 0x4008d199:0x3ffb6d50 
0x400e34a2: gpio_ll_iomux_in at /Users/.esp/esp-idf/components/hal/esp32/include/hal/gpio_ll.h:561
 (inlined by) gpio_iomux_in at /Users/.esp/esp-idf/components/driver/gpio.c:718

0x400e1c8a: ll_cam_set_pin at /Users/Desktop/esp32_ov2640/components/esp32-camera/target/esp32/ll_cam.c:371

0x400d9ad4: cam_init at /Users/Desktop/esp32_ov2640/components/esp32-camera/driver/cam_hal.c:319 (discriminator 2)

0x400d9411: esp_camera_init at /Users/Desktop/esp32_ov2640/components/esp32-camera/driver/esp_camera.c:224

0x400d91c2: camera_init() at /Users/Desktop/esp32_ov2640/main/main.cpp:81

me-no-dev, 2022-02-17:
yes, the driver is not yet compatible with IDF 5. Please use 4.4 or earlier

sxsx2yzyz, 2022-03-02:
esp32-camera/target/esp32/ll_cam.c 
Note that lines 37 to 41 solve the problem
`// #if (ESP_IDF_VERSION_MAJOR >= 5)
// #define GPIO_PIN_INTR_POSEDGE GPIO_INTR_POSEDGE
// #define GPIO_PIN_INTR_NEGEDGE GPIO_INTR_NEGEDGE
// #define gpio_matrix_in(a,b,c) gpio_iomux_in(a,b)
// #endif
`

me-no-dev, 2022-03-02:
@sxsx2yzyz are you saying that `gpio_matrix_in` was added again to IDF5?

sxsx2yzyz, 2022-03-03:
> @sxsx2yzyz are you saying that `gpio_matrix_in` was added again to IDF5?
I don't know, but it worked in IDF5

sxsx2yzyz, 2022-03-03:
> 

I don't know, but it worked in IDF5

nopnop2002, 2022-03-28:
For ESP32, we are operating 0x38 (= 56) here.

https://github.com/espressif/esp32-camera/blob/master/target/esp32/ll_cam.c#L371

ESP32 does not have GPIO56.
Panic occurs here.

igrr, 2022-05-27:
Still valid, to be fixed by https://github.com/espressif/esp32-camera/pull/382.

@me-no-dev the PR appears to be inactive, perhaps you could cherry-pick that commit and make the follow-up fix? It'd be nice to resolve the issue.

nopnop2002, 2022-05-29:
Since portTICK_RATE_MS is used, a compile error will occur in Ver5.

https://github.com/espressif/esp32-camera/blob/master/driver/sccb.c#L134

closed 2022-05-30 by me-no-dev