espressif/esp32-camera #367: Compilation error

cgrrty, 2022-02-25:
```
pi@ubuntu:~/workspace/esp/esp32-camera$ 
pi@ubuntu:~/workspace/esp/esp32-camera$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/pi/workspace/esp/esp32-camera/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/pi/workspace/esp/esp32-camera/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/pi/workspace/esp/esp32-camera"...
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:65 (register_component):
  Unknown CMake command "register_component".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.10)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/pi/workspace/esp/esp32-camera/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

```

cgrrty, 2022-02-25:
add code cmake_minimum_required(VERSION 3.10)
```
cmake_minimum_required(VERSION 3.10)
if(IDF_TARGET STREQUAL "esp32" OR IDF_TARGET STREQUAL "esp32s2" OR IDF_TARGET STREQUAL "esp32s3")
  set(COMPONENT_SRCS
    driver/esp_camera.c
    driver/cam_hal.c
    driver/sccb.c
    driver/sensor.c
    sensors/ov2640.c
    sensors/ov3660.c
    sensors/ov5640.c
    sensors/ov7725.c
    sensors/ov7670.c
    sensors/nt99141.c
    sensors/gc0308.c
    sensors/gc2145.c
    sensors/gc032a.c
    sensors/bf3005.c
    sensors/bf20a6.c
    conversions/yuv.c
    conversions/to_jpg.cpp
    conversions/to_bmp.c
    conversions/jpge.cpp
    conversions/esp_jpg_decode.c
    )

  set(COMPONENT_ADD_INCLUDEDIRS
    driver/include
    conversions/include
    )

  set(COMPONENT_PRIV_INCLUDEDIRS
    driver/private_include
    sensors/private_include
    conversions/private_include
    target/private_include
    )

  if(IDF_TARGET STREQUAL "esp32")
    list(APPEND COMPONENT_SRCS
      target/xclk.c
      target/esp32/ll_cam.c
      )
  endif()

  if(IDF_TARGET STREQUAL "esp32s2")
    list(APPEND COMPONENT_SRCS
      target/xclk.c
      target/esp32s2/ll_cam.c
      target/esp32s2/tjpgd.c
      )

    list(APPEND COMPONENT_PRIV_INCLUDEDIRS
      target/esp32s2/private_include
      )
  endif()

  if(IDF_TARGET STREQUAL "esp32s3")
    list(APPEND COMPONENT_SRCS
      target/esp32s3/ll_cam.c
      )
  endif()

  set(COMPONENT_REQUIRES driver)
  set(COMPONENT_PRIV_REQUIRES freertos nvs_flash)

  register_component()
endif()
```

```
pi@ubuntu:~/workspace/esp/esp32-camera$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/pi/workspace/esp/esp32-camera/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/pi/workspace/esp/esp32-camera/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/pi/workspace/esp/esp32-camera"...
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:66 (register_component):
  Unknown CMake command "register_component".

-- Configuring incomplete, errors occurred!
See also "/home/pi/workspace/esp/esp32-camera/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

me-no-dev, 2022-02-25:
you are trying to compile from the root of the driver. This driver is an IDF component, not a ready to go project. To test example, you can `cd` into the example folder and try from there (clean the component first)

closed 2022-05-05 by github-actions[bot]

erosenst, 2023-11-01:
~/esp/ESP8266_RTOS_SDK/examples/peripherals/hw_timer$ ../../../tools/idf.py reconfigure
Checking Python dependencies...
Python requirements from /home/eros/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Executing action: reconfigure
Running cmake in directory /home/eros/esp/ESP8266_RTOS_SDK/examples/peripherals/hw_timer/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/eros/esp/ESP8266_RTOS_SDK/examples/peripherals/hw_timer"...
CMake Error at CMakeLists.txt:7 (register_component):
  Unknown CMake command "register_component".

-- Configuring incomplete, errors occurred!
See also "/home/eros/esp/ESP8266_RTOS_SDK/examples/peripherals/hw_timer/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1