espressif/esp32-camera #368: OV2640, JPEG strange cam_hal: NO-SOI behaviour

Jona2892, 2022-03-03:
Hey there,

I got some warnings on one of my modules: cam_hal: NO-SOI
After my research I found these two issues #307 and #330 . The result of both of them is that this behaviour is based on a bad/old or fake ov2640. So far this seems a valid explanation. To confirm this I did tests with 3 ov2640 and two different esp32-wrover modules. Now the strange thing is, that on one esp32-module I do NOT get these warnings and on the other I get them. And this for all 3 ov2640, same Code on both modules.
Thus I think the cause for the "cam_hal: NO-SOI Warning" does not lie on a bad ov2640. Maybe it is the combination of the esp32-module and the ov2640? Or something in the code?

I have also testet other formats like greyscale and rgb565 different resolutions. They work fine, but this is right now just guessing, because I can not see the pictures.

My Setups:
Code:
- ESP-IDF
- Simple Webserver to show the jpeg images from [here](https://github.com/EcoBay/esp-ai-webcam)
- newest esp32-camera code from here

1. Setup (No warnings)
- ESP32-WROVER-IB (not really sure about the B): ESP32-D0WDQ6 (revision 1) [click](https://www.amazon.de/Freenove-ESP32-WROVER-Bluetooth-Compatible-Tutorials/dp/B09BC5CNHM?ref_=ast_sto_dp)
- one other device on the same i2c lines of the ov2640

2. Setup(cam_hal: NO-SOI warning)
- ESP32-WROVER-IE: ESP32-D0WD-V3 (revision 3) (own pcb, which is basically a copy of the above module, but with newer esp32-chip))
- one other device on the same i2c lines of the ov2640

Do you have any suggestion what can cause this behaviour or some advice how I can do more research/testing about this issue?

Best,
Jonathan

Jona2892, 2022-03-06:
Hello,
Found my problem, I have a mistake in Setup 2 with my own pcb in the pin mapping of the camera. If the data lines are not in the right order esp can not find the right SOI or EOI obviously. 
Everything works just fine now!
Best

closed 2022-03-06 by Jona2892

Nickobellic15X, 2022-08-28:
Hey can you tell me know what is SOI and EOI, also would be great if you can point me to a direction where I can learn more about how these cameras work.

krupis, 2023-07-10:
Same here. I am getting:
```
cam_hal: NO-SOI
```

And not sure how to solve it. Any information regarding SOI would be great!

Jona2892, 2023-07-11:
NO-SOI stands for "NO Start Of Image" and NO-EOI for "NO End Of Image".

The image data has specific bytes at the beginning and at the end of the image data. The driver can not find these Start and End bytes of the image.

In my case this was due to the fact, that I messed up the pin-mapping. So first thing I would check are the pin settings and also the power settings.

krupis, 2023-07-11:
> NO-SOI stands for "NO Start Of Image" and NO-EOI for "NO End Of Image".
>
> The image data has specific bytes at the beginning and at the end of the image data. The driver can not find these Start and [...]

Thanks for quick reply. In my case, I used official Espressif Kaluga example project so I doubt the pin mapping is the issue.
It is worth mentioning that NO-SOI error is only returned when I use:
```
.pixel_format = PIXFORMAT_JPEG,
```

It works without any issues with other pixel_formats

Jona2892, 2023-07-11:
I have never worked with the Kaluga example...
Did you set the `CONFIG_CAMERA_JPEG_MODE` macro or just changed the `.pixel_format` line?
You could also try to play around with the `xclk_freq_hz`.

krupis, 2023-07-11:
> I have never worked with the Kaluga example... Did you set the `CONFIG_CAMERA_JPEG_MODE` macro or just changed the `.pixel_format` line? You could also try to play around with the `xclk_freq_hz` and lower frame-sizes `frame_size.`

There is no such option in menuconfig:
```
CONFIG_CAMERA_JPEG_MODE
```

I have only changed the .pixel_format

I have created a ticket in esp-idf git. If I get this issue solved il post back here

Jona2892, 2023-07-11:
Usually in the menuconfig you should see all the options listed in this [file](https://github.com/espressif/esp-dev-kits/blob/master/esp32-s2-kaluga-1/examples/camera/main/Kconfig.projbuild) under the "Example Configuration" menu.

If you can not see them at all you may need to full clean and rebuild. 
Since there are a lot of `#ifdef CONFIG_CAMERA_JPEG_MODE` in the example I think this is your problem. I think you can not just switch the `.pixel_format` in the Kaluga example.

Jona2892, 2023-07-11:
Ah sorry, I was looking at the wrong project...

radzimir, 2024-02-17:
In my case it was probably broken cable. Other camera worked fine.

xieyanname, 2025-01-17:
I also encountered this problem because the IO settings were incorrect. Modifying the IO worked fine, but I also encountered "NO-EOI". Here are the parameters I set:
.pixel_format = PIXFORMAT_JPEG, 
.frame_size = FRAMESIZE_QSXGA,
.jpeg_quality = 2,
But changing jpeg_quality to 3 can output photos (925 KB) normally. Does anyone know why this is?
Compilation environment:
ESP-IDF v5.0.7
esp32-camera-v2.0.15