espressif/esp32-camera #369: Compilation Problem

simogaspa84, 2022-03-10:
Hi guys.. 

When I try to compile the project (platformio ide espressif framework) I got 

 error: 'CONFIG_SCCB_CLK_FREQ'

[image]

me-no-dev, 2022-03-14:
This is a value set in sdkconfig by `idf.py menuconfig`. You could probably define it yourself in your PIO project as well.

tennisparty, 2022-04-11:
I get the same error trying to build code on platformio (vscode ext) with the expressive framework. I have looked in the sdkconfig file but find no option here

In the kconfig file there is the following line

    config SCCB_CLK_FREQ
    int "SCCB clk frequency"
    default 100000
    range 100000 400000
    help
        Increasing this value can reduce the initialization time of the sensor.
        Please refer to the relevant

I still get the error,

Any help appreciated,
Sam

zakkie, 2022-04-12:
@tennisparty 
I got the same error using platformio. But I resolved it with the following:

```shell
cd some-project/
cd src
ln -s ../.pio/libdeps/(board name)/esp32-camera/Kconfig Kconfig.projbuild
(Run Menuconfig and enable camera configuration)
```

tennisparty, 2022-04-13:
Thankyou so much @zakkie, that has solved it for me. You are a legend.

closed 2022-05-31 by me-no-dev

RangeOfGlitching, 2022-07-05:
@tennisparty Can you give me a reference where can I enable camera configuration?
[image]

me-no-dev, 2022-07-06:
@RangeOfGlitching typo there? "camNera?

RangeOfGlitching, 2022-07-06:
@me-no-dev sorry about that
but unfortunately even I type camera still nothing appear

RangeOfGlitching, 2022-07-11:
I sove the problem by [this](https://esp32.com/viewtopic.php?t=27293)