espressif/esp32-camera #370: ESP32_cam error in serial monitor

ataris99, 2022-03-11:
It seems i am having an error i am using esp32_cam connected with ftdi rx with u0t and tx with u0r and gnd to gnd and 5v to vcc etc but this is the error i recieve 

i dont know what is it

E (897) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
CAMERA FAIL
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0

E (897) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
CAMERA FAIL

ataris99, 2022-03-13:
Anyone able to help?

me-no-dev, 2022-03-16:
you are using another library all together. `esp32cam` is where you should ask for help.

MaxSvjatoha, 2022-03-17:
I had the same error. This likely does not solve the root problem, but at least my code runs now.

Just before running esp_camera_init(), I added the following lines:
digitalWrite(PWDN_GPIO_NUM, LOW);
delay(10);
digitalWrite(PWDN_GPIO_NUM, HIGH);
delay(10);

and the problem disappeared for me

closed 2022-05-24 by github-actions[bot]