espressif/esp32-camera #372: SC031GS camera

weilian1977, 2022-03-18:
in the doument （https://github.com/espressif/esp-faq/blob/78813a2f062f684ae657fe6f529955883d8e23f1/docs/zh_CN/application-solution/camera-application.rst）
ESP32 系列芯片支持哪种类型的摄像头？
OV 系列：OV2640、OV3660、OV5640、OV7725
GC 系列：gc0308、gc032a、gc2145
SC 系列：sc031gs
but esp32-camera(https://github.com/espressif/esp32-camera) there do not have drivers，can provide sc031gs driver？

TDA-2030, 2022-03-22:
`sc031gs` doesn't have a good enough driver at present. It will be corrected in FAQ

weilian1977, 2022-03-22:
is there sc031gs driver code？I test usb sc031gs module，It seems acceptable.

TDA-2030, 2022-03-22:
This repository only supports DVP interface, USB camera is another driver.

weilian1977, 2022-03-22:
yes，I have tested usb sc031gs module  is a usb chip + sc031gs （DVP or mipi），It can directly get image  in windows ，so I can  estimate image quality. the image quality seems acceptable,so I need  esp32s3 to driver DVP sc031gs

TDA-2030, 2022-03-22:
Unfortunately, I don't think it's easy to get the same effect as a USB camera module, as it would require us to have enough time and data to tune the camera from scratch

weilian1977, 2022-03-22:
是不是可以直接中文沟通？

TDA-2030, 2022-03-22:
如果一定要用这个摄像头的话，需要在官网提交需求：https://www.espressif.com/zh-hans/contact-us/technical-inquiries

weilian1977, 2022-03-23:
原来用安森美的芯片，现在一直缺货，只能换国产，如果咱们已有sc031gs的驱动（哪怕不完善）还请麻烦提供一下，我们目前正在写驱动。

closed 2022-05-31 by me-no-dev