espressif/esp32-camera #380: cam_hal: Failed to get the frame on time!

heculess, 2022-04-08:
board：esp32-wrover-e

setting：

camera_config_t config;
config.ledc_channel = LEDC_CHANNEL_0;
config.ledc_timer = LEDC_TIMER_0;
config.pin_d0 = Y2_GPIO_NUM;
config.pin_d1 = Y3_GPIO_NUM;
config.pin_d2 = Y4_GPIO_NUM;
config.pin_d3 = Y5_GPIO_NUM;
config.pin_d4 = Y6_GPIO_NUM;
config.pin_d5 = Y7_GPIO_NUM;
config.pin_d6 = Y8_GPIO_NUM;
config.pin_d7 = Y9_GPIO_NUM;
config.pin_xclk = XCLK_GPIO_NUM;
config.pin_pclk = PCLK_GPIO_NUM;
config.pin_vsync = VSYNC_GPIO_NUM;
config.pin_href = HREF_GPIO_NUM;
config.pin_sscb_sda = SIOD_GPIO_NUM;
config.pin_sscb_scl = SIOC_GPIO_NUM;
config.pin_pwdn = PWDN_GPIO_NUM;
config.pin_reset = RESET_GPIO_NUM;
config.xclk_freq_hz = 10000000;
config.pixel_format = PIXFORMAT_JPEG;
config.fb_location = CAMERA_FB_IN_PSRAM;
config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;

config.frame_size = FRAMESIZE_HD; 
config.jpeg_quality = 6;      
config.fb_count = 1;

camera ：nt99141

log：

I (2572) cam_hal: cam init ok
D (2575) camera: Initializing SSCB
I (2578) sccb: pin_sda 4 pin_scl 27
D (2582) intr_alloc: Connected src 50 to int 12 (cpu 0)
D (2586) camera: Resetting camera
I (2589) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (2618) camera: Searching for camera address
I (2628) camera: Detected camera at address=0x2a
I (2631) camera: Detected NT99141 camera
I (2631) camera: Camera PID=0x1410 VER=0x00 MIDL=0x00 MIDH=0x00
D (2631) camera: Doing SW reset of sensor
D (2900) NT99141: Camera defaults loaded
D (2900) NT99141: set_ae_level can not be configured at present
I (3002) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 45
I (3002) cam_hal: Allocating 184320 Byte frame buffer in PSRAM
D (3006) intr_alloc: Connected src 32 to int 13 (cpu 0)
I (3011) cam_hal: cam config ok
D (3014) camera: Setting frame size to 1280x720
D (3173) NT99141: Set FRAMESIZE_HD
D (3302) NT99141: Set pixformat to: 3
D (3303) NT99141: Set quality to: 6
D (3315) NT99141: get_agc_gain can not be configured at present
D (3315) NT99141: get_aec_value can not be configured at present

W (41907) cam_hal: Failed to get the frame on time!
E (41907) Camera: Camera capture failed
W (41907) cam_hal: Failed to get the frame on time!
E (41907) Camera: Camera capture failed
W (41907) cam_hal: Failed to get the frame on time!
E (41907) Camera: Camera capture failed
W (41907) cam_hal: Failed to get the frame on time!
E (41907) Camera: Camera capture failed
W (41907) cam_hal: Failed to get the frame on time!
E (41907) Camera: Camera capture failed

weilian1977, 2022-04-08:
D (3315) NT99141: get_agc_gain can not be configured at present
D (3315) NT99141: get_aec_value can not be configured at present

MaxSvjatoha, 2022-04-08:
There is a hard timer in esp_camera.c (called FB_GET_TIMEOUT) for how much time there is to take a picture. The better quality and resolution, the more time you need to take a picture. 

- One simpler way of increasing this time is to set config.xclk_freq_hz to something lower. Your's is 10 MHz, I have mine at 2-2.5 MHz and it works fine. Below 2 MHz and other things start failing for me, e.g. cam init fails.
- Another quick way is to set your config.fb_count to 2 or higher (note than this takes up extra RAM). The reason why this works is listed in this very repo's README. Eseentially, it doubles the time you have to take an image. Quote: "When 2 or more frame bufers are used, I2S is running in continuous mode and each frame is pushed to a queue that the application can access. This approach puts more strain on the CPU/Memory, but allows for double the frame rate. Please use only with JPEG."
- I recommend also to setting your config.grab_mode to CAMERA_GRAB_LATEST, especially when using fb_count > 1.

Hope this helps!

heculess, 2022-04-08:
It seems that ll_cam_vsync_isr and ll_cam_dma_isr never to be triggered . Maybe the camera need to be opened at first?

WangYuxin-esp, 2022-04-20:
The current sensor is working in an abnormal state. I mean that the power supply or pin configuration of the sensor may not be normal. It is recommended to check the line configuration. Or use the official development board for testing. Hope this helps!

heculess, 2022-05-05:
Which board is the official development board for sensor NT99141？I also want it

WangYuxin-esp, 2022-05-06:
The official development boards with camera include ESP-EYE, ESP32-S3-EYE, ESP32-S2-Kaluga-1.   
I did some tests using the ESP32-S3-EYE board with the NT99141 sensor and the device works fine.  Hope this helps!

heculess, 2022-11-09:
I can't get any image from ESP32-S3-EYE with NT99141 sensor.
[image]

MattiaDallaCosta, 2022-12-15:
I have a similar problem but not exactly the same

themanyone, 2023-03-03:
I have the same problem with VSCode-compiled esp32cam. cam_hal: Failed to get the frame on time!
The same code works using Arudino IDE, so there is some configuration issue with VSCode. Even with xclk_freq_hz = 20000000 for high-speed captures.
Trying to narrow it down.

WangYuxin-esp, 2023-03-03:
@me-no-dev Please help confirm whether some changes in Arduino are not synchronized to the `esp32-camera`.

me-no-dev, 2023-03-13:
@WangYuxin-esp I saw similar issue as well. The crazy part is that I tried it with the same everything that is compiling Arduino's libs and also failed. Only worked reliably in Arduino. I can suggest to diff the `sdkconfig` files between Arduino and the project and look for differences that could influence the result. It's like something is messing with the peripheral/interrupts. Just a note, my issue was switching resolutions only for S3-EYE and I already plan a workaround just in case.

Abhay-2412, 2023-11-10:
so any solution right now im getting same problem!!

Meowchine1, 2024-06-04:
That solution helps me https://www.youtube.com/watch?v=6KKHgyapl-c

Meowchine1, 2024-06-04:
Also I format sd card