espressif/esp32-camera #381: ESP32-S3-EYE，OV2640、OV5640，JPEG，高分辨率、低压缩率  ： cam_hal: NO-EOI

cocowts, 2022-04-11:
请教大家一个问题，使用 **ESP32-S3-EYE** ，可以驱动 **ov5640/OV2640** 摄像头，但是当使用 **jpeg** 模式，设置**高像素，低压缩率**的时候就会拍照失败（低像素或者高压缩率拍照成功），提示 **cam_hal: NO-EOI** （测试似乎拍照图片太大就会失败），阅读了 cam_hal ， esp_camera 文件，已经启动了 psram 但还没有解决

> cam 初始化配置：FRAMESIZE_QSXGA （2560x1920）， 压缩率 2
```
camera_t *camera_init(void)
{
    if (init(20000000, PIXFORMAT_JPEG, FRAMESIZE_QSXGA, 1) == ESP_OK) { 
        return &g_camera;
    }

    return NULL;
}

// ......

static esp_err_t init(uint32_t xclk_freq_hz, pixformat_t pixel_format, framesize_t frame_size, uint8_t fb_count)
{
    camera_config_t camera_config = {
        .pin_pwdn = PWDN_GPIO_NUM,
        .pin_reset = RESET_GPIO_NUM,
        .pin_xclk = XCLK_GPIO_NUM,
        .pin_sscb_sda = SIOD_GPIO_NUM,
        .pin_sscb_scl = SIOC_GPIO_NUM,

        .pin_d7 = Y9_GPIO_NUM,
        .pin_d6 = Y8_GPIO_NUM,
        .pin_d5 = Y7_GPIO_NUM,
        .pin_d4 = Y6_GPIO_NUM,
        .pin_d3 = Y5_GPIO_NUM,
        .pin_d2 = Y4_GPIO_NUM,
        .pin_d1 = Y3_GPIO_NUM,
        .pin_d0 = Y2_GPIO_NUM,
        .pin_vsync = VSYNC_GPIO_NUM,
        .pin_href = HREF_GPIO_NUM,
        .pin_pclk = PCLK_GPIO_NUM,

        //EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
        .xclk_freq_hz = xclk_freq_hz,
        .ledc_timer = LEDC_TIMER_0,
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = pixel_format,      //YUV422,GRAYSCALE,RGB565,JPEG
        .frame_size = frame_size,            //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

        .jpeg_quality = 2,                       //0-63 lower number means higher quality
        .fb_count = fb_count,                //if more than one, i2s runs in continuous mode. Use only with JPEG
        .grab_mode = CAMERA_GRAB_WHEN_EMPTY
    };

    //initialize the camera
    return esp_camera_init(&camera_config);
}
```

> 系统启动时的一些信息：
```
I (24) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (25) boot: compile time 10:47:45
I (25) boot: chip revision: 0
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a5f0h ( 42480) map
I (100) esp_image: segment 1: paddr=0001a618 vaddr=3fc92df0 size=02ff0h ( 12272) load
I (103) esp_image: segment 2: paddr=0001d610 vaddr=40374000 size=02a08h ( 10760) load
I (111) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1e274h (123508) map
I (140) esp_image: segment 4: paddr=0003e29c vaddr=40376a08 size=0c3e8h ( 50152) load
I (151) esp_image: segment 5: paddr=0004a68c vaddr=50000000 size=00010h (    16) load
I (157) boot: Loaded app from partition at offset 0x10000
I (158) boot: Disabling RNG early entropy source...
I (171) opi psram: vendor id : 0x0d (AP)
I (171) opi psram: dev id    : 0x02 (generation 3)
I (172) opi psram: density   : 0x03 (64 Mbit)
I (175) opi psram: good-die  : 0x01 (Pass)
I (180) opi psram: Latency   : 0x01 (Fixed)
I (185) opi psram: VCC       : 0x01 (3V)
I (189) opi psram: SRF       : 0x01 (Fast Refresh)
I (195) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (200) opi psram: BurstLen  : 0x01 (32 Byte)
I (205) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (211) opi psram: DriveStrength: 0x00 (1/1)
W (216) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (227) spiram: Found 64MBit SPI RAM device
I (231) spiram: SPI RAM mode: sram 80m
I (236) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (243) cpu_start: Pro cpu up.
I (246) cpu_start: Starting app cpu, entry point is 0x4037531c
0x4037531c: call_start_cpu1 at D:/JDE/esp-idf/frameworks/esp-idf-v4.4/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (673) spiram: SPI SRAM memory test OK
I (681) cpu_start: Pro cpu start user code
I (681) cpu_start: cpu freq: 240000000
I (681) cpu_start: Application information:
I (684) cpu_start: Project name:     main
I (689) cpu_start: App version:      1
I (693) cpu_start: Compile time:     Apr  8 2022 16:45:52
I (699) cpu_start: ELF file SHA256:  d114563a88c1624e...
I (705) cpu_start: ESP-IDF:          v4.4-dirty
I (710) heap_init: Initializing. RAM available for dynamic allocation:
I (718) heap_init: At 3FC96800 len 00049800 (294 KiB): D/IRAM
I (724) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (731) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (737) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (743) spiram: Adding pool of 8192K of external SPI memory to heap allocator       //   注意这里！！！！！
I (751) spi_flash: detected chip: generic
I (756) spi_flash: flash io: dio
I (760) sleep: Configure to isolate all GPIO pins in sleep state
I (767) sleep: Enable automatic switching of GPIO sleep configuration
I (774) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (794) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (794) s3 ll_cam: DMA Channel=4
I (804) cam_hal: cam init ok
I (804) sccb: pin_sda 4 pin_scl 5
I (814) gpio: GPIO[43]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (834) gpio: GPIO[44]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (884) camera: Detected camera at address=0x3c
I (884) camera: Detected OV5640 camera
I (884) camera: Camera PID=0x5640 VER=0x00 MIDL=0x00 MIDH=0x00
I (1554) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 1600
I (1554) cam_hal: Allocating 1638400 Byte frame buffer in PSRAM                         //   注意这里！！！！！
I (1554) cam_hal: cam config ok
I (1574) ov5640: Set PLL: bypass: 0, multiplier: 200, sys_div: 4, pre_div: 2, root_2x: 0, pclk_root_div: 2, pclk_manual: 1, pclk_div: 4
I (1574) ov5640: Calculated XVCLK: 20000000 Hz, REFIN: 10000000 Hz, VCO: 2000000000 Hz, PLL_CLK: 200000000 Hz, SYSCLK: 50000000 Hz, PCLK: 12500000 Hz
```

> 出错时的日志：
```
I (4604) example:take_picture: picture: 2560 x 1920, size: 857833
W (5114) cam_hal: NO-EOI  //  注意这里！！！！当拍照背景复杂，导致图片大小变大时错误
W (5354) cam_hal: NO-EOI
I (6134) example:take_picture: picture: 2560 x 1920, size: 120370
```

WangYuxin-esp, 2022-04-20:
Hi，'NO-EOI' 说明数据传输结束后，没有检查到 JPEG 图像的终止标识。
产生这个问题的主要原因是图片过大，且设置的 jpeg_qualtiy 很小（图像质量很好）的情况下，在一个帧传输时间内要传输的数据将剧增，最终可能导致摄像头的 FPC 线上的数据之间产生干扰，导致数据出错。建议设置的 jpeg_aquality 不大于 4 或者降低 xclk.

cocowts, 2022-04-24:
@WangYuxin-esp 非常感谢您的耐心讲答，我仔细阅读了您的建议，做了如下测试：

1. 使用较大的压缩率将减小图片体积，会拍照成功，但**也会损失图像质量**；
2. 尝试在保证图片高质量的同时尽量降低 xclk， 但是测试下来似乎还是不行（最低已经降低到了 2M）。

---

按照您上面的讲解，是图片太大，导致一个帧传输时间内要传输的数据将剧增， FPC 线上数据更容易出现干扰导致的问题吗？

在初始化摄像头的时候，已经设置了摄像头的工作频率，那么数据线上每秒的速率没有变化，只是一帧的传输时间变长
而且测试发现，当图片体积接近 1M，基本是必出错，没有概率性出现大体积图像拍照成功
这让我十分困惑的，表现不像是干扰造成的

非常期待您的答复

WangYuxin-esp, 2022-04-24:
初始化时配置的摄像头的频率为 xclk，该频率将用于提供 sensor 的时钟，包括 sensor 输出到 esp32 的 vsync、pclk、href。  
当图片数据很大时，pclk 将变的很快（这是 sensor 决定的），这可能超出设备能接收的频率范围而出现异常。data[0]~data[7] 之间的数据由于并非差分信号，也会产生干扰。  这两个方面是造成 接收不到或接收错误的 EOI 的主要原因。当然，不排除是 sensor 的寄存器配置不正常，导致在一些情况下，sensor 本身输出的数据就是有问题的。  如果您正在进行产品的开发，请尝试与 sensor 的技术支持联系，获取更稳定的 sensor 寄存器配置参数。  
最后，如果您正在进行的是识别类的项目，请尝试使用 sensor 的 ISP 功能来获取更好的识别效果，而不要将希望全部寄托在更高质量的 JPEG 图片上。

xinhaojie, 2022-04-28:
@cocowts 你用的那个板子呀，我用S3官方那个开发板，JPEG的也是一直打印NO-SOI，怎么修改配置都不行。

cocowts, 2022-04-28:
开发板：ESP32-S3-EYE
这是我使用的几个参数：
0. 确定 IO 连接与程序对应
1. 使用大的压缩率，通常大于 4
2. 使用低的分辨率
3. 这是我现在使用的时钟频率：.xclk_freq_hz = 20000000
4. 开启 psram [没办法截图 menuconfig 的配置项，如果怀疑是这里的问题，可以加我微信 18805171729 ]

```
/**Include ---------------------------------------------------*/

#include "camera.h"

/** Private define -------------------------------------------*/

#define PWDN_GPIO_NUM     43
#define RESET_GPIO_NUM    44

#define VSYNC_GPIO_NUM    6
#define HREF_GPIO_NUM     7
#define PCLK_GPIO_NUM     13
#define XCLK_GPIO_NUM     15

#define SIOD_GPIO_NUM     4
#define SIOC_GPIO_NUM     5

#define Y9_GPIO_NUM       16
#define Y8_GPIO_NUM       17
#define Y7_GPIO_NUM       18
#define Y6_GPIO_NUM       12
#define Y5_GPIO_NUM       10
#define Y4_GPIO_NUM       8
#define Y3_GPIO_NUM       9
#define Y2_GPIO_NUM       11

/** Private function prototypes ------------------------------*/

static camera_fb_t* camera_get_fb(void);
static void camera_release_fb(camera_fb_t *fb);

/**Private variables ----------------------------------------*/

static camera_handler g_camera = {
    .get_fb     = camera_get_fb,
    .release_fb = camera_release_fb
};

/**Applaton ------------------------------------------------*/

camera_handler *camera_init(uint32_t xclk_freq_hz, pixformat_t pixel_format, framesize_t frame_size, uint8_t fb_count)
{
    camera_config_t camera_config = {
        .pin_pwdn = PWDN_GPIO_NUM,
        .pin_reset = RESET_GPIO_NUM,
        .pin_xclk = XCLK_GPIO_NUM,
        .pin_sscb_sda = SIOD_GPIO_NUM,
        .pin_sscb_scl = SIOC_GPIO_NUM,

        .pin_d7 = Y9_GPIO_NUM,
        .pin_d6 = Y8_GPIO_NUM,
        .pin_d5 = Y7_GPIO_NUM,
        .pin_d4 = Y6_GPIO_NUM,
        .pin_d3 = Y5_GPIO_NUM,
        .pin_d2 = Y4_GPIO_NUM,
        .pin_d1 = Y3_GPIO_NUM,
        .pin_d0 = Y2_GPIO_NUM,
        .pin_vsync = VSYNC_GPIO_NUM,
        .pin_href = HREF_GPIO_NUM,
        .pin_pclk = PCLK_GPIO_NUM,

        //EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
        .xclk_freq_hz = xclk_freq_hz,
        .ledc_timer = LEDC_TIMER_0,
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = pixel_format,       //YUV422,GRAYSCALE,RGB565,JPEG
        .frame_size = frame_size,           //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

        .jpeg_quality = 2,                  //0-63 lower number means higher quality
        .fb_count = fb_count,               //if more than one, i2s runs in continuous mode. Use only with JPEG
        .grab_mode = CAMERA_GRAB_WHEN_EMPTY
    };

    if (esp_camera_init(&camera_config) != ESP_OK) {
        return NULL;
    }

    return &g_camera;
}

static camera_fb_t* camera_get_fb(void)
{
    return esp_camera_fb_get();
}

static void camera_release_fb(camera_fb_t *fb) 
{
    esp_camera_fb_return(fb);
}
```

xinhaojie, 2022-04-28:
@cocowts I (6140) s3 ll_cam: DMA Channel=4
I (6140) cam_hal: cam init ok
I (6140) sccb: pin_sda 7 pin_scl 5
I (6160) camera: Detected camera at address=0x30
I (6160) camera: Detected OV2640 camera
I (6160) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (6240) cam_hal: buffer_size: 16384, half_buffer_size: 1024, node_buffer_size: 1024, node_cnt: 16, total_cnt: 15
I (6240) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (6240) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (6250) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (6260) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (6260) cam_hal: cam config ok
I (6270) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8

我的spiram已经打开了，初始化摄像头也成功了。不过我买的摄像头是正点原子那个带插针的。他板子上有一个12MHz的晶振，所以xclk没办法设置。我试了RGB565的可以正常读取数据。就是这个JPEG的不行。我又买了两个不带晶振的，到时候试一下吧。我的JPEG分辨率设置的96×96 压缩率63也是一样的现象。

cocowts, 2022-04-28:
@xinhaojie 
RGB565 有显示图像吗？ESP32可能只是从数据线上取数据，得到的不一定是正确的。
JPEG 使用的库有校验格式，NO-SOI 这个是说没有检测到头标记。
可以关注一下这个地方：请确认 OV2640 支持的像素格式，96×96 是否被支持

closed 2022-07-04 by github-actions[bot]