espressif/esp32-camera #383: Ov2640 photos with green/blue dark tint

achao1104, 2022-04-19:
Hello,

I am using idf4.3.2, esp32s2-wrover and ov2640, and when I take a pic of sun, the photo looks green in background
[image]
.

WangYuxin-esp, 2022-04-20:
Please confirm whether the abnormal photos are the first two photos after the camera is initialized.  
According to my test, this problem often occurs in the first few photos after the camera is initialized.

achao1104, 2022-04-21:
Yes, it is first, and I have changed to use second or third, it is not green but looks like purple now.

WangYuxin-esp, 2022-04-24:
The camera sensor is fully responsible for capturing and processing images, and esp32 device only passively receives data for now.  In other words, esp32 is only responsible for initializing the sensor and then receiving data from the sensor.  Therefore, I suggest checking the relevant configuration of the sensor:    
1)Check whether the power supply voltage of the sensor is consistent with the requirements of the technical manual.  
2)Check whether the lens used by the camera matches the sensor.  
3)The AWB(Auto White Balance) is enabled by default in the driver. Maybe you can try using the ISP function of the sensor to get normal photos.

me-no-dev, 2022-04-24:
I would suggest to just skip the first 10-ish frames so that AWB and all get's calibrated. As @WangYuxin-esp said. This is coming from the sensor itself. More than AWB, you also have AGC, Lens Correction and more enabled in the sensor.

achao1104, 2022-05-24:
> The camera sensor is fully responsible for capturing and processing images, and esp32 device only passively receives data for now. In other words, esp32 is only responsible for initializing the [...]

thanks, what can I do for "Maybe you can try using the ISP function of the sensor to get normal photos."

achao1104, 2022-05-24:
> I would suggest to just skip the first 10-ish frames so that AWB and all get's calibrated. As @WangYuxin-esp said. This is coming from the sensor itself. More than AWB, you also have AGC, Lens Correction and more enabled in the sensor.

I have skipped 11 frames and between each frames, it waits for 100ms, but the pic is still purple.
[image]

Wolfleader101, 2022-06-16:
I have a similar issue,

When I follow this tutorial my images turn out with a green tint:
https://randomnerdtutorials.com/esp32-cam-take-photo-save-microsd-card/

When I use the esp32 cam web server example, my images look perfect

closed 2022-08-23 by github-actions[bot]

MartinK7, 2022-12-21:
I think this is caused by bad auto color correction in the camera chip. You can post process these images manully.

[image]
[image]

ymich9963, 2024-02-29:
If anyone is still struggling with this, if you disable all of the auto functions except lenc, and set the sensor settings after initialisation to white balance mode 3 or 4, it should get rid of the green tint. The sensor initialises with AGC and AWB enabled so you need to disable them and fix the white balance before taking a picture.