espressif/esp32-camera #386: ESP32S3 ov2640 JPEG 失败

xinhaojie, 2022-04-28:
.xclk_freq_hz = 12000000,//EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG,//YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_240X240,//QQVGA-QXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 63, //0-63 lower number means higher quality
    .fb_count = 640, //if more than one, i2s runs in continuous mode. Use only with JPEG
    .grab_mode = CAMERA_GRAB_LATEST//CAMERA_GRAB_LATEST. Sets when buffers should be filled

上述官方配置中，验证了565格式可以获取，但是JEPG就是怎么改配置都不行，质量已经设置为63.buffer也放开到了640个。各种分辨率都试了，一直是打印NO-SOI，怎么解决呢。。。。

WangYuxin-esp, 2022-04-29:
请问当前使用的 sensor 是什么型号的？官方标配的 OV2640 使用 JPEG 是正常的。

xinhaojie, 2022-05-03:
@WangYuxin-esp 我只是自己买的ov2640的摄像头，我买了两种，一种是正点原子那种插针式的，另一种是那种24pin排线那种，我加了个转接板。JPEG模式下一直都是打印NO_SOI，无论怎么调整参数都是这个打印。RGB565输出视频颜色昏暗，很不清晰。另外官方有ov2640的摄像头买吗。
[image]，

WangYuxin-esp, 2022-05-04:
> @WangYuxin-esp 我只是自己买的ov2640的摄像头，我买了两种，一种是正点原子那种插针式的，另一种是那种24pin排线那种，我加了个转接板。JPEG模式下一直都是打印NO_SOI，无论怎么调整参数都是这个打印。RGB565输出视频颜色昏暗，很不清晰。另外官方有ov2640的摄像头买吗。 [image]，

官方推荐的 camera 开发板有 esp-eye、esp32s3-eye、esp32s2-kaluga，这些开发板自带 0V2640 摄像头，TB上有的买，可以了解一下。  
摄像头输出图像的问题，应该是 sensor 的配置不正常导致的。推荐使用官方的板子进行验证。

xinhaojie, 2022-05-05:
@WangYuxin-esp 我找到原因了，我用了20cm杜邦线进行了连接，今天我怀疑了线，然后用10cm的连接了之后完美解决

WangYuxin-esp, 2022-05-05:
太好了，看来使用的线也是需要考虑的因素。👍

xinhaojie, 2022-05-06:
@WangYuxin-esp 这板子基本的功能我快调完了，好几次遇到线引起的问题。之前调LED的时候也是，线太长导致信号有问题。

closed 2022-07-13 by github-actions[bot]

jianjunchang, 2025-01-01:
> > @WangYuxin-esp 我只是自己买的ov2640的摄像头，我买了两种，一种是正点原子那种插针式的，另一种是那种24pin排线那种，我加了个转接板。JPEG模式下一直都是打印NO_SOI，无论怎么调整参数都是这个打印。RGB565输出视频颜色昏暗，很不清晰。另外官方有ov2640的摄像头买吗。 [image]，
> 
> 官方推荐的 camera 开发板有 esp-eye、esp32s3-eye、esp32s2-kaluga，这些开发板自带 0V2640 摄像头，TB上有的买，可以了解一下。 摄像头输出图像的问题，应该是 sensor 的配置不正常导致的。推荐使用官方的板子进行验证。

您好，看到截的图，和我的一样，在亮度高的区域，颜色不对，这个您有解决吗？

WangYuxin-esp, 2025-01-02:
JPEG 模式下，配置 XCLK = 10M、VGA 试试。  
RGB565 模式下，如果不是线接错了，那么就是开发板的供电电压与 sensor 模组期望的电压不匹配。

jianjunchang, 2025-01-03:
> JPEG 模式下，配置 XCLK = 10M、VGA 试试。 RGB565 模式下，如果不是线接错了，那么就是开发板的供电电压与 sensor 模组期望的电压不匹配。

嗯嗯，这个问题已经困扰我很久了，因为要做识别，所以原始图像采集质量希望能高点，但是一直都是高亮的部分出现花屏，网上找了很久解决方案未果，我用jpeg模式之前一直不出图像（直接放弃了），回去试下您说的两项配置，rgb565的效果就像您这个截图，部分出现花屏，您的意思是图像的花屏和电压也会有关系是吗（这倒是一直没想到的因素）？接线问题应该不存在，因为我用的板载摄像头，排线都接好了没动过，现在我这里只有灰度图是正常的，最近实在没办法了准备采用灰度图识别了（肯定会损失精度），回去试下独立供电模块，希望能解决，感谢！