espressif/esp32-camera #387: Broken build for non-supported platforms

vikramdattu, 2022-04-28:
Hi,

My project [tflite-micro-esp-examples](https://github.com/espressif/tflite-micro-esp-examples) uses esp32-camera as a submodule.

esp32-camera support following archs: `esp32`, `esp32-s3`, `esp32-s2`. For other platforms, when I try to build the project for esp32-c3, I get CMake error, that esp32-camera is not supported for target esp32c3!

Suggestion: Have some generic includes and hollow APIs for non-supported platforms and throw an error while camera_init! This will make sure proper build for other platforms and generic utils and structures up for use on those platforms as well.

Thanks!

cc @me-no-dev

me-no-dev, 2022-04-28:
Which example in your project are you having issues with? [Here](https://github.com/espressif/esp32-arduino-lib-builder) is the project that builds the IDF libs for Arduino. It includes the camera driver and compiled for all chips (C3 included) without errors on unsupported chips. The camera example in arduino checks if camera is supported and if so allows the includes

vikramdattu, 2022-04-28:
Hi @me-no-dev, 

I use esp32-camera as a submodule and include all these components.

When I try to build any example, I get below error message:
```
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32c3
Component "esp32-camera" does not support target esp32c3
CMake Error at /Users/vikramdattu/work/esp-idf/tools/cmake/build.cmake:455 (message):
Call Stack (most recent call first):
  /Users/vikramdattu/work/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:5 (project)
```

I believe the issue happens because, component not registered for not supported platforms? https://github.com/espressif/esp32-camera/blob/86a4951f507e86034725b04f2428c98c49496dfe/CMakeLists.txt#L1

me-no-dev, 2022-04-28:
This actually comes from the package manager of ESP-IDF I think. @kumekay do you have any ideas here?

kumekay, 2022-05-02:
@me-no-dev Yes, the line comes from the component manager, since the manifest esp32-camera lists only 3 supported targets:  https://github.com/espressif/esp32-camera/blob/master/idf_component.yml#L2

It was filled to follow the list in the readme: https://github.com/espressif/esp32-camera#supported-soc 
Does this component actually support esp32c3?

vikramdattu, 2022-05-02:
@kumekay given that this component is not supported for `esp32c3`, I want to be able to compile projects for this platform but with limited functionality or throw error on `camera_init`. Current issue however blocks compilation itself!

kumekay, 2022-05-02:
@vikramdattu  It was made intentionally, however, I would agree that we should raise an exception only for the components downloaded by the component manager, but not for ones in EXTRA_COMPONENT_DIRS or `components` directory. Still, I think it is worth printing a warning about incompatibility in this case.

me-no-dev, 2022-05-03:
@kumekay there are other projects (like Arduino) that are build for different chips (including C3) and do include esp32-camera as a component. While possible, it would be sad to have to delete the folder just to be able to compile for C3 and others like it. Because of some historical reasons, currently `idf_component.yml` get's deleted before Arduino compilation and that is why we did not catch it first there.

kumekay, 2022-05-03:
@me-no-dev Since the issue is not related to esp-camera I suggest closing this and continue discussion in PACMAN-332

me-no-dev, 2022-05-04:
@igrr that is more or less what happens without pacman. Plus some things in the driver are usable with any platform (color conversions, etc).

So what is the correct action to take here @kumekay? Can we wildcard all platforms?

kumekay, 2022-05-09:
@me-no-dev @igrr I still would suggest differentiating the way how a component is added to the project - if a component is downloaded by the component manager,  raising an exception in CMake - looks legit to me. 
However for components in the project's `components` directory or added by EXTRA_COMPONENT_DIRS I would print a warning. In this case - no need to wildcard esp32-camera for targets that it doesn't support7

kumekay, 2022-05-10:
@me-no-dev Do you have a way to set an environment during the build? We can add an environment variable (or a CMake option, but it will be harder) to suppress the target check

me-no-dev, 2022-05-10:
Requiring env variable from users would be much the same as asking them to pull the repo manually. CMakeLists option can be added to the component itself, so I'd say it's the better option :)

vikramdattu, 2022-05-27:
Hi, any update on this? It would be really nice if there are no target rules for the component.
This will enable me to keep using a component as a submodule (helps to keep resolve issue away for older IDF versions).

me-no-dev, 2022-05-27:
I also lean more towards no target rules. The component itself already knows what to compile when and will not cause issues on other targets.

vikramdattu, 2022-05-27:
Maybe useful for someone fixing it. Changes I had to do to CMakeList to get just headers included when non-supported platform needed to compile.

[esp32_camera_include_just_headers_for_c3.txt](https://github.com/espressif/esp32-camera/files/8786352/esp32_camera_include_just_headers_for_c3.txt)

igrr, 2022-05-27:
Okay, so since we agree on that, can we remove the "supported targets" list from idf_component.yml?

closed 2022-05-27 by me-no-dev

vikramdattu, 2022-05-28:
@kumekay can you please also make changes in CMakeList, so that, there are no build errors and includes are still useable?
https://github.com/espressif/esp32-camera/issues/387#issuecomment-1139539326

kumekay, 2022-05-28:
@vikramdattu, I'm sorry I missed your patch in the comment. 
Since the original PR is already merged and you have the patch, please open the new PR?

vikramdattu, 2022-05-28:
@kumekay Sure, will do that!