espressif/esp32-camera #392: Do not wait for the frame to finish to send it

dzervas, 2022-05-18:
I'm developing a streaming solution where latency is crucial. What I'd like to do is to start reading the buffer before the frame is complete inside the buffer. I use MTU of 1400 having just the first 1400 bytes is good enough for me.

Should I just get access to `cam_obj->frame_buffer_queue` and `xQueueReceive` it?

me-no-dev, 2022-05-19:
yes. More or less that is what you need to do, but you need to handle the data faster than it arrives, else you will overflow and lose data. You might have better chance with ESP32-S3

dzervas, 2022-05-19:
what's the difference with the S3? is it faster?

me-no-dev, 2022-05-19:
yes in many ways. PSRAM works better, it has dedicated camera peripheral that does not require data gymnastics in the task and so on.

me-no-dev, 2022-05-19:
> Would it be ok to make a PR to move ll_cam.h to include and make cam_obj non-static so I can access it without doing monkey stuff with my local copy of the component?

What you are requesting is actually way outside of the scope of the driver. Because of that, you must edit the camera component yourself. I would actually not recommend what you want to do to anyone, because the danger of overflow or some other error is way too high. I do not know what you are trying to do, but do a few milliseconds mean that much?

dzervas, 2022-05-19:
Yea, I'm trying to use the ESP32-CAM as an FPV system (camera & vTX). A few ms mean a lot. I wanna achieve ~30FPS at SVGA with max 50ms latency glass-to-glass.

dzervas, 2022-05-19:
I'm using `80211_tx` and a wifi card in monitor mode - I've already got this part working, I'm just not sure for the latency yet. I've got to test it on a Pi with an RT kernel and directFB rendering

dzervas, 2022-05-20:
Ok, so I changed my approach quite a bit. I don't use PSRAM anymore (it's slow) and I'm allocating 1 fb. The whole fb is never in ram as I constantly consume it by sending it to a queue (and then send it over wifi from a different task). The problem that I have is that for some reason the `cam_obj->dma_buffer` has some static data (`0xBC 0x32 0xFD 0x3F` according to wireshark) at the first 4 bytes and the rest are zeros. It's like the dma_buffer is never written - but then where does the DMA data go?

I saw that `I2S0.in_link.addr` is set to `cam_obj->dma[0]` which in turn points to `cam_obj->dma_buffer` (so `cam_obj->dma[x]` is mapped inside `cam_obj->dma_buffer`). My code rn:

```c
	while (true) {
		if (xQueueReceive(cam_obj->event_queue, (void *) &cam_event, portMAX_DELAY) != pdTRUE) {
			printf("Failed to receive event from queue\n");
			continue;
		}

		switch (cam_obj->state) {
			case CAM_STATE_IDLE:
				if (cam_start_frame(&frame_pos)) {
					cam_obj->frames[frame_pos].fb.len = 0;
					cam_obj->state = CAM_STATE_READ_BUF;
				}
				cnt = 0;
				break;

			case CAM_STATE_READ_BUF:
				switch (cam_event) {
					case CAM_IN_SUC_EOF_EVENT:
						cnt++;

						break;

					case CAM_VSYNC_EVENT:
						ll_cam_stop(cam_obj);
						if (cnt) cnt++;
						xQueueSend(wifi_queue, (void *) &cam_obj->dma_buffer, 0);
						break;

					default:
						break;
				}
				break;

			default:
				break;
		}
		if (cam_obj->state == CAM_STATE_READ_BUF && cam_event == CAM_VSYNC_EVENT) {
			cnt = 0;

			// Probably not needed without PSRAM
			// cam_give(fb);
			if(!cam_start_frame(&frame_pos)){
				cam_obj->state = CAM_STATE_IDLE;
			} else {
				cam_obj->frames[frame_pos].fb.len = 0;
			}
		}
    }
```

dzervas, 2022-05-20:
Ok, I'm getting video but I think that the bottom frame is missing, or the receiver is not fast enough. Imma build the raspi receiver and test it again

```c
static IRAM_ATTR void queue_subframe(int cnt) {
	if (!cam_obj->dma[cnt].length || cam_obj->dma[cnt].owner) {
		return;
	}

	// Why the fuck do I only take the 3rd byte of each DMA buffer node????
	// Cause `sample2` is for YUV/RGB and `unused1`/`unused2` are obviously unused - still, waste of space
	// const dma_elem_t* dma_el = (const dma_elem_t *) cam_obj->dma[cnt].buf;
	const dma_elem_t* dma_el = (const dma_elem_t *) &cam_obj->dma_buffer[(cnt % cam_obj->dma_half_buffer_cnt) * cam_obj->dma_half_buffer_size];
	uint8_t *subframe_buffer_ptr = subframe_buffer;
	for (size_t i = 0; i < CAMERA_FRAME_NODE / 8; i++) {
		subframe_buffer_ptr[0] = dma_el[0].sample1;
		subframe_buffer_ptr[1] = dma_el[1].sample1;

		// If this is not the first frame check for JPEG end - the rest are garbage
		if (cnt && *((uint16_t *) subframe_buffer_ptr) == 0xD9FF) {
			cam_obj->state = CAM_STATE_IDLE;
			break;
		}

		dma_el += 2;
		subframe_buffer_ptr += 2;
	}

	// Check if the first 3 bytes that we receive are the JPEG start marker
	if (!cnt && (*((uint32_t *) subframe_buffer) & 0xFFFFFF) != 0xFFD8FF) {
		cam_obj->state = CAM_STATE_IDLE;
		return;
	}

	if (xQueueSend(wifi_queue, subframe_buffer, 0) != pdTRUE)
		ESP_LOGW(TAG, "wifi_queue is full at cnt: %d", cnt);
}

	while (true) {
		if (xQueueReceive(cam_obj->event_queue, (void *) &cam_event, portMAX_DELAY) != pdTRUE) {
			ESP_LOGW(TAG, "Timed out waiting to receive camera event. Did you forget to call `camera_init()`?");
			continue;
		}

		switch (cam_obj->state) {
			case CAM_STATE_IDLE: {
				ll_cam_start(cam_obj, 0);
				cam_obj->state = CAM_STATE_READ_BUF;
				cnt = 0;
				break;
			}

			case CAM_STATE_READ_BUF: {
				#ifdef CONFIG_PERISTERI_MODE_STREAM
					queue_subframe(cnt);
				#elif defined(CONFIG_PERISTERI_MODE_CAMERA)
					// TODO: Make a cam task
					// sdcard_picture(fb->buf, fb->len);
					// sdcard_record(fb->buf, fb->len);
				#endif

				cnt++;

				switch (cam_event) {
					case CAM_IN_SUC_EOF_EVENT:
						break;

					case CAM_VSYNC_EVENT: {
						ll_cam_stop(cam_obj);
						cam_obj->state = CAM_STATE_IDLE;

						break;
					}

					default:
						break;
				}
				break;
			}

			default:
				break;
		}
	}
```

closed 2022-08-17 by github-actions[bot]