espressif/esp32-camera #409: crashes on esp_camera_init() on idf-v5 cloned on 19-06-2022

moefear85, 2022-06-22:
Hi,

Is this project supported on esp-idf-v5? It's crashing on call to esp_camera_init() with the following code:

## Code ##
```
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void camera_setup()
{
    camera_config_t camera_config =
    {
        .pin_pwdn  = PWDN_GPIO_NUM,
        .pin_reset = RESET_GPIO_NUM,
        .pin_xclk = XCLK_GPIO_NUM,
        .pin_sscb_sda = SIOD_GPIO_NUM,
        .pin_sscb_scl = SIOC_GPIO_NUM,

        .pin_d7 = Y9_GPIO_NUM,
        .pin_d6 = Y8_GPIO_NUM,
        .pin_d5 = Y7_GPIO_NUM,
        .pin_d4 = Y6_GPIO_NUM,
        .pin_d3 = Y5_GPIO_NUM,
        .pin_d2 = Y4_GPIO_NUM,
        .pin_d1 = Y3_GPIO_NUM,
        .pin_d0 = Y2_GPIO_NUM,
        .pin_vsync = VSYNC_GPIO_NUM,
        .pin_href = HREF_GPIO_NUM,
        .pin_pclk = PCLK_GPIO_NUM,

        .xclk_freq_hz = 20000000,//EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
        .ledc_timer = LEDC_TIMER_0,
        .ledc_channel = LEDC_CHANNEL_0,

        .pixel_format = PIXFORMAT_JPEG, //YUV422,GRAYSCALE,RGB565,JPEG
        .frame_size = FRAMESIZE_VGA,//QQVGA-QXGA Do not use sizes above QVGA when not JPEG

        .jpeg_quality = 10, //0-63 lower number means higher quality
        .fb_count = 2, //if more than one, i2s runs in continuous mode. Use only with JPEG
        .fb_location=CAMERA_FB_IN_DRAM,
        .grab_mode = CAMERA_GRAB_LATEST,//CAMERA_GRAB_LATEST | CAMERA_GRAB_WHEN_EMPTY. Sets when buffers should be filled        
    };

    ESP_ERROR_CHECK( esp_camera_init(&camera_config) );
 }
```

## Other Files ##
[debug.log](https://github.com/espressif/esp32-camera/files/8958995/debug.log)
[sdkconfig.txt](https://github.com/espressif/esp32-camera/files/8959007/sdkconfig.txt)

igrr, 2022-06-22:
Could you please add, which esp32-camera commit ID and IDF commit ID are you using? (You can check by running `git describe --tags` in each repository.)

Based on the backtrace, this looks like an issue which was recently fixed in https://github.com/espressif/esp32-camera/pull/395/files#diff-0156d7917176219dd52ecdc7a13d3e7f572769721b7166b4d278400237722250R44.

moefear85, 2022-06-22:
ok yes I just tested... latest esp32-camera is definately working with latest esp-idf-v5

thx

closed 2022-06-22 by moefear85