espressif/esp32-camera #416: Wroom32

tcpipchip, 2022-07-07:
Will this example works on Wroom 32 ? 
(without psram)
?

closed 2022-09-14 by github-actions[bot]

jayavanth, 2023-01-20:
Had the same issue. Even if I had disabled PSRAM in the config I was getting this error. I added this line to `camera_config` and I was able to get past this issue

```
.fb_location = CAMERA_FB_IN_DRAM,
```

However, I'm facing a new issue

```
E (1067) sccb: W [3808]=01 0 fail

E (1068) ov5640: Setting framesize to: 320x240 failed
E (1073) camera: Failed to set frame size
E (1078) example:take_picture: Camera Init Failed

```

I'm not sure if this problem is related to the PSRAM

jayavanth, 2023-01-21:
Solved it! After doing the above, just had to lower the frequency `.xclk_freq_hz = 15000000`. This is for Waveshare OV5640 Camera Board (B), so it might be different for your board

tcpipchip, 2023-01-21:
Are you using esp32 without psram ?

jayavanth, 2023-01-21:
Using ESP32-S3 without PSRAM (ESP32-S3-DevKitC-1-N8)

tcpipchip, 2023-01-21:
Will work on esp32 wroom kit ?

jayavanth, 2023-01-23:
@tcpipchip I don't have a ESP32 WROOM kit, but see if you can give it a try. It might work