espressif/esp32-camera #418: SCCB_Read() return PID 0x00

kxbin, 2022-07-13:
When connecting the ov2640 camera, the address is successfully detected:
```
Detected camera at address=0x30
```
However, the correct PID is not obtained, and it prompts:
```
Mismach PID=0x0
```

I looked up the code and found that it is because the data returned by `i2c_master_read_byte()` is 0, or it is not returned at all (the default value of data is 0).

Can you help me, thanks a lot!
```
uint16_t PID = SCCB_Read(slv_addr, 0x0A);
```
```
uint8_t SCCB_Read(uint8_t slv_addr, uint8_t reg)
{
    uint8_t data=0;
    esp_err_t ret = ESP_FAIL;
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ( slv_addr << 1 ) | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, reg, ACK_CHECK_EN);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(SCCB_I2C_PORT, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    if(ret != ESP_OK) return -1;
    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ( slv_addr << 1 ) | READ_BIT, ACK_CHECK_EN);
    i2c_master_read_byte(cmd, &data, NACK_VAL);
    i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(SCCB_I2C_PORT, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    if(ret != ESP_OK) {
        ESP_LOGE(TAG, "SCCB_Read Failed addr:0x%02x, reg:0x%02x, data:0x%02x, ret:%d", slv_addr, reg, data, ret);
    }
    return data;
}
```

closed 2022-07-20 by kxbin

svendvd, 2022-11-10:
@kxbin How did you fixed this?

kxbin, 2022-11-11:
@svendvd   I replaced a camera and it worked fine

svendvd, 2022-11-11:
I feared that. Thanks for answering 🙂