espressif/esp32-camera #422: Wrong byte order for RGB565

r-downing, 2022-07-18:
If I take a framebuffer from the camera in jpeg format, and pass it through jpg2rgb565(...), then back through fmt2jpg(...) the colors are all screwed up - the rgb channels seem to be swapped and shifted.

_For context I was capturing high-res images, then passing them to the jpg2rgb... to scale them down and do some image processing._

But if I take the output buffer of jpg2rgb565(...) and swap every alternating byte before passing it back to fmt2jpg(...) then the image appears correct.

```
uint8_t rgbBuf[...];

jpg2rgb565(fb->buf, fb->len, rgbBuf, JPG_SCALE_8X);

// swap the byte order and the image looks ok
size_t i;
for(i = 0; i < sizeof(rgbBuf); i+=2){
    const uint8_t tmp = rgbBuf[i];
    rgbBuf[i] = rgbBuf[i+1];
    rgbBuf[i+1] = tmp;
}

fmt2jpg(rgbBuf, ...)

```

So I believe either the output of jpg2rgb565() or the expected input format of fmt2jpg(...) is wrong, but I'm not sure which. If you guys know, I can make a fix

WangYuxin-esp, 2022-07-21:
Yes, my test here is the same as yours. The problem may be caused by the wrong byte order of the `jpg2rgb565()` function, because when the camera outputs data in standard RGB format, the `fmt2jpg()` function is normal.

r-downing, 2022-07-21:
Ok, thanks for confirming! I'll make a PR tomorrow

closed 2022-12-05 by github-actions[bot]