espressif/esp32-camera #424: Does the camera start working every time `esp_camera_fb_get()` is called

kxbin, 2022-07-20:
Hello everyone, I would like to ask：

Does the camera starts to capture `1 frame` every time `esp_camera_fb_get()` is called?

Or start automatically capturing frames continuously in the background once initialized?

Because I found that when I set `fb_count>=3`, but `esp_camera_fb_get()` still waits more than `40ms`, he is blocked, so very confused.

WangYuxin-esp, 2022-07-20:
The camera starts automatically capturing frames continuously in the background once initialized. It takes some time for the data from sensor to esp32. You can use an oscilloscope or logic analyzer to see how much time it takes for data transmission.

kxbin, 2022-07-21:
> The camera starts automatically capturing frames continuously in the background once initialized. It takes some time for the data from sensor to esp32. You can use an oscilloscope or logic analyzer to see how much time it takes for data transmission.

@WangYuxin-esp Thank you very much for your reply, this way I understand how the camera works

By the way, is it possible to use a software fifo if we need to optimize the waiting time of `esp_camera_fb_get()`?

For example I learned that there is lwrb library that can work for us:
https://github.com/MaJerle/lwrb

See #427 for details

closed 2022-07-21 by kxbin

WangYuxin-esp, 2022-07-21:
@kxbin In my opinion, the software implementation has little impact on the time to obtain a frame of data. The best way is to optimize the configuration of camera sensors. Of course, you can try to study the implementation of this driver and test the performance after using this ringbuffer mechanism. By the way, esp-idf seems to have implemented a ringbuffer mechanism, which can be used as a reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#ring-buffers