espressif/esp32-camera #434: Need to cooperate with other users of an I2C bus

jepler, 2022-08-03:
Frequently it's necessary to place other I2C devices on the same pins as a supported camera. Additionally, it may be necessary to access the other devices before esp_camera_init or after esp_camera_deinit. It seems likely that the deinitialization of the I2C bus in esp_camera_deinit is the cause of at least one open issue, #426.

To support this use case, it should be possible for the application to select the I2C peripheral by number and appropriately configure it, and then pass this into esp_camera_init instead of the pins.

I have implemented a prototype of this for the Circuitpython fork of esp32-camera and intend to soon prepare a pull request into this repo. When `sccb_sda` and `sccb_scl` are NO_PIN, then the `sccb_i2c_port` is used instead, and esp32-camera never initializes nor resets the I2C peripheral.
```c
    self->camera_config.pin_sccb_sda = NO_PIN;
    self->camera_config.pin_sccb_scl = NO_PIN;
    self->camera_config.sccb_i2c_port = i2c->i2c_num;
```

jepler, 2022-08-03:
@ladyada putting this on your radar

jepler, 2022-08-03:
Our ongoing work to integrate esp32-camera into CircuitPython is at https://github.com/adafruit/circuitpython/pull/6684 and our un-tidy branch to implement this feature as well as other needs is (compare view) https://github.com/espressif/esp32-camera/compare/master...adafruit:esp32-camera:pass-bus-to-sccb?expand=1

igrr, 2022-08-04:
Hi @jepler, thanks for raising this issue. I have noticed that there is an open PR related to the similar requirement: https://github.com/espressif/esp32-camera/pull/413. Could you please take a look and tell if merging that PR will satisfy your needs?

igrr, 2022-08-23:
@jepler https://github.com/espressif/esp32-camera/pull/413 was updated based on your feedback and merged. When you have time, please take a look if the latest version of that change addresses your requirement.

closed 2022-12-31 by github-actions[bot]