espressif/esp32-camera #435: [ESP32-S3-DevKitC-1, OV2640] Got a lot of VSYNC-OVF, EOF-OVF when dynamic frequency scaling was enabled

nguyen-phuoc-dai, 2022-08-04:
Hi Dear,

I use ESP32-S3-DevKitC-1 with OV2640 camera and IDF v4.4.1 (latest commit). I tried the examples project, everything goes well with default settings. 

But when I go to "config" then enable power manager and dynamic frequency scaling as below screenshot

[image]

From log messages on console, I got a lot of EV-VSYNC-OVF, EV-EOF-OVF alerts. 

Is there any something wrong with power management? What settings need to be enable to fix these alerts?

`
I (1029) spiram: SPI SRAM memory test OK
I (1037) cpu_start: Pro cpu start user code
I (1037) cpu_start: cpu freq: 240000000
I (1037) cpu_start: Application information:
I (1040) cpu_start: Project name:     camera_example
I (1045) cpu_start: App version:      v2.0.1-4-ga6f13d9-dirty
I (1052) cpu_start: Compile time:     Jul 20 2022 16:49:28
I (1058) cpu_start: ELF file SHA256:  a35082379844bef8...
I (1064) cpu_start: ESP-IDF:          v4.4.1-472-gc9140caf8c-dirty
I (1071) heap_init: Initializing. RAM available for dynamic allocation:
I (1078) heap_init: At 3FC98A20 len 000475E0 (285 KiB): D/IRAM
I (1085) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (1091) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1098) heap_init: At 600FE020 len 00001FE0 (7 KiB): RTCRAM
I (1104) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (1113) spi_flash: detected chip: generic
I (1117) spi_flash: flash io: dio
I (1132) sleep: Configure to isolate all GPIO pins in sleep state
I (1132) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 40, Light sleep: DISABLED
I (1138) sleep: Disable automatic switching of GPIO sleep configuration
I (1145) coexist: coexist rom version e7ae62f
I (1150) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1166) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1175) s3 ll_cam: DMA Channel=4
I (1179) cam_hal: cam init ok
I (1182) sccb: pin_sda 41 pin_scl 17
I (1187) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1227) camera: Detected camera at address=0x30
I (1231) camera: Detected OV2640 camera
I (1231) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1333) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (1333) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (1341) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (1352) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (1359) cam_hal: cam config ok
I (1362) ov2640: Set PLL: clk_2x: 1, clk_div: 3, pclk_auto: 1, pclk_div: 8
I (1456) example:take_picture: Taking picture...
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
W (5456) cam_hal: Failed to get the frame on time!
W (5456) example:take_picture: Failed to take picture!
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
I (10456) example:take_picture: Taking picture...
I (10456) example:take_picture: Picture taken! Its size was: 153600 bytes
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
I (15457) example:take_picture: Taking picture...
I (15457) example:take_picture: Picture taken! Its size was: 153600 bytes
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
I (20458) example:take_picture: Taking picture...
I (20458) example:take_picture: Picture taken! Its size was: 153600 bytes
cam_hal: EV-EOF-OVF
`

Thanks

igrr, 2022-08-04:
@nguyen-phuoc-dai as far as I can tell, the esp32-camera driver is not power-management-aware. Since some of the work is done by the hardware in the background, the power management subsystem has to be informed that it shouldn't put the device into lower power mode during the data acquisition. This can be done by making this driver take a power management lock when starting the camera and releasing it when stopping the camera.
The change is not very complex, but i wonder if having power management enabled will then bring any benefits. Since the camera is getting frames all the time, the lock will be held and the system will not go into the low power state. It's similar to not enabling the power management at all.
Could you describe your use case? Do you intend to capture frames continuously or just once?

nguyen-phuoc-dai, 2022-08-09:
Dear @igrr 

Sorry for late response, I was on vacation on last week

In normal condition, camera does not capture any images and device should go to light sleep mode to save battery. When there is IO trigger, we will capture images continually and stop to capture when the IO trigger disappear. We DO want go to light sleep mode when the camera does not capture, that was reason why we have power management enabled.

Yes, power management lock is exactly solution that we are going with the current product - it uses the esp32 chipset - and it works like a charm

Now we are trying new esp32s3 chipset for our new product . The same code base for power management lock was used, but it looks like the lock has not effect on esp32s3. May be due to low driver for esp32s3 is different from esp32 (DVP vs I2S) ? or there is something wrong with DVP peripheral when power management was enabled?

Below was the 'examples' project with power lock was added and logging message on the console  

```
----
----

void app_main()
{
    esp_pm_lock_handle_t block_light_sleep_lock = NULL;
	
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

#if CONFIG_PM_ENABLE
    ret = esp_pm_lock_create(ESP_PM_NO_LIGHT_SLEEP, 0, TAG, &block_light_sleep_lock);
    if (ret != ESP_OK) {
    	ESP_LOGE(TAG, "Failed to create power lock (%s)", esp_err_to_name(ret));
    	esp_restart();
    }
	
    // Configure dynamic frequency scaling:
    // maximum and minimum frequencies are set in sdkconfig,
    // automatic light sleep is enabled if tickless idle support is enabled.
#if CONFIG_IDF_TARGET_ESP32
    esp_pm_config_esp32_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32S2
    esp_pm_config_esp32s2_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32C3
    esp_pm_config_esp32c3_t pm_config = {
#elif CONFIG_IDF_TARGET_ESP32S3
    esp_pm_config_esp32s3_t pm_config = {
#endif
            .max_freq_mhz = CONFIG_ESP32S3_DEFAULT_CPU_FREQ_MHZ,
            .min_freq_mhz = 40,
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
            .light_sleep_enable = true
#endif
    };
    // This statement was called or DFS was checked on menuconfig, OVF messages were fired on console
    ESP_ERROR_CHECK( esp_pm_configure(&pm_config) );
#endif // CONFIG_PM_ENABLE

    if(ESP_OK != init_camera()) {
        return;
    }
		
    while (1)
    {
	if (block_light_sleep_lock != NULL) {
		esp_pm_lock_acquire(block_light_sleep_lock);
	}
		
        ESP_LOGI(TAG, "Taking picture...");
		
        //uint32_t start_capture = esp_log_timestamp();
        camera_fb_t *pic = esp_camera_fb_get();

        if (pic != NULL) {
            // use pic->buf to access the image
            ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", pic->len);
            esp_camera_fb_return(pic);
        } else {
            ESP_LOGW(TAG, "Failed to take picture!");
        }
		
	if (block_light_sleep_lock != NULL) {
		esp_pm_lock_release(block_light_sleep_lock);
	}
		
        vTaskDelay(5000 / portTICK_RATE_MS);
    }
}
```

This is log messages on console

```
I (1168) cam_hal: cam init ok
I (1172) sccb: pin_sda 41 pin_scl 17
I (1177) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1217) camera: Detected camera at address=0x30
I (1221) camera: Detected OV2640 camera
I (1221) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1321) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (1321) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (1329) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (1340) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (1347) cam_hal: cam config ok
I (1351) ov2640: Set PLL: clk_2x: 1, clk_div: 3, pclk_auto: 1, pclk_div: 8
I (1444) example:take_picture: Taking picture...
I (2299) example:take_picture: Picture taken! Its size was: 153600 bytes
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
I (7299) example:take_picture: Taking picture...
I (7299) example:take_picture: Picture taken! Its size was: 153600 bytes
cam_hal: EV-EOF-OVF
I (12300) example:take_picture: Taking picture...
I (12300) example:take_picture: Picture taken! Its size was: 153600 bytes
I (17301) example:take_picture: Taking picture...
I (17301) example:take_picture: Picture taken! Its size was: 153600 bytes
cam_hal: EV-VSYNC-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-VSYNC-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
cam_hal: EV-EOF-OVF
I (22302) example:take_picture: Taking picture...
I (22302) example:take_picture: Picture taken! Its size was: 153600 bytes

```
Please take a look on it

Thank you very much

nguyen-phuoc-dai, 2022-08-15:
Dear all,

Does anyone run to similar issue?

LeTuanAnhEP0818E, 2022-10-20:
You should set XCLK 10MHz for OV2640.

nguyen-phuoc-dai, 2022-10-21:
Hi @LeTuanAnhEP0818E 

Thanks you for your input, i will try and let you know

closed 2022-12-28 by github-actions[bot]

wayinone, 2024-07-22:
Thanks , changing to 
```
config.xclk_freq_hz = 10000000; 
```
and I don't see the error.