espressif/esp32-camera #436: setting frame size to 96*96 failed

Farzinkh, 2022-08-05:
I am using ESP32_CAM and there is no problem,except when I set frame size to 96*96 error below shows up and other frame sizes are just fine.

`W (28923) cam_hal: NO-EOI

W (28955) cam_hal: FB-OVF

W (28955) cam_hal: NO-EOI

W (28995) cam_hal: FB-OVF

W (28995) cam_hal: NO-EOI

W (29035) cam_hal: FB-OVF

W (29035) cam_hal: NO-EOI

...

W (32923) cam_hal: Failed to get the frame on time!
`
how can i fix this?
regards,

WangYuxin-esp, 2022-08-09:
Maybe the current camera driver does not allow this size to be configured.
Could you tell me what model sensor you are using?

Farzinkh, 2022-08-11:
> Maybe the current camera driver does not allow this size to be configured. Could you tell me what model sensor you are using?

I am using default camera sensor which is OV2640. 
I have to say this problem only exist if  pixformat is set to PIXFORMAT_JPEG , I have tried PIXFORMAT_GRAYSCALE without any problem.

devellison, 2022-08-14:
I ran into similar. I think the root of the problem is that the PIXFORMAT_JPEG doesn't allocate enough space in some circumstances.

Changing this line from `/ 5` to `/ 4` worked for me.  
https://github.com/espressif/esp32-camera/blob/master/driver/cam_hal.c#L376

Another option is possibly to increase your quality value into the 14+ range.

I'm haven't played with JPEGs enough to know a better estimate of space right off, but it might be something to look into.

closed 2022-10-20 by github-actions[bot]

AgainPsychoX, 2024-02-19:
I can't get it to work reliably, sometimes it works even for a while, sometimes it stops working randomly... I tried using both low JPEG quality and higher `cam_obj->recv_size`, but it always fails with `cam_hal: Failed to get the frame on time!` sooner or later.

AgainPsychoX, 2024-02-23:
I'm currently testing with 96x96 with forced minimal size of `recv_size` of 4096 bytes, normal XCLK of 20MHz, still the same ESP32 / OV2640 and I found weird thing. It works more reliability with high JPEG quality (<12) than lower quality (>16). What's more, I get errors and weird frame capture time issues when I set  lower quality (screen from logs when I was using Q=21).

[image]

The lower quality (so, higher Q number), the more issues there are for some odd reason, only for 96x96.