espressif/esp32-camera #439: custom board doesn't work with some OV2640 modules

crides, 2022-08-13:
So I designed a custom board for use with the OV2640 cameras, and also have a Ai thinker ESP32 CAM module. I also have 2 kinds of cameras (they should be both OV2640 but are somehow different?) shown below: 

[image]

Out of the 2, both works with the ESP32 CAM module, but only the one below (with the narrower FPC cable) works on my board. The other camera simply gives errors (either a full panic, or just a `capture failed` sometimes; can't tell what triggers which). I have a oscilloscope, and both do seem to give the same signals on every pin, on both boards. 

Using a ESP32 (like, original, not S2/S3/C/H) module, if that makes a difference. The firmware for testing on both boards are the same other than the pin assignments. If there are other details needed, please let me know. Thanks!

dongmenhai, 2022-08-15:
@crides I think you should provide some necessary logs or pictures to better analyze the problem. First of all, for the sensor with the problem, has the sensor OV2640 been detected as shown in the figure below?
[image]
Secondly, can you provide the display effect of the output picture or the log printed when the picture cannot be printed?

crides, 2022-08-15:
I'm using this from the Arduino esp32 package, and unfortunately, I'm not familiar with how the setup actually works. I have `  esp_log_level_set("*", ESP_LOG_VERBOSE);` placed in my `setup()`, however no logs are shown from the library (I got the logs from the demo code by replacing the log functions).

So far, when it doesn't work, the browser just puts a "no image" icon at the place of the image, and the HTTP request just returns with a `Content-Length` of 0. The actual output in the console varies a lot, from `[E] Capture failed` (through the example code) to panics like:

```
Camera Ready! Use 'http://192.168.143.73' to connect
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (httpd)
Core  1 register dump:
PC      : 0x4008ef93  PS      : 0x00060036  A0      : 0x8008db64  A1      : 0x3ffd8b60
A2      : 0x3ffbf0b8  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060023
A6      : 0x00060023  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x3ffbf0a8  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x3ffd8c40
A14     : 0x007bf0b8  A15     : 0x003fffff  SAR     : 0x0000000f  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x40089b44  LEND    : 0x40089b5a  LCOUNT  : 0xffffffff

Backtrace:0x4008ef90:0x3ffd8b600x4008db61:0x3ffd8ba0 0x4008c1bc:0x3ffd8bd0 0x4008c16c:0xa5a5a5a5  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

or

```
Camera Ready! Use 'http://192.168.143.73' to connect

Backtrace:0xfffffffe:0x8008dc58 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
```

dongmenhai, 2022-08-19:
@crides  Sorry, I'm not familiar with Arduino configuration, I think Arduino is also a repackage of esp32-camera and esp-idf, can you use [pic_server](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/pic_server) this example directly to test and verify your ov2640 custom board? This will have more logs to analyze the problem

crides, 2022-08-19:
Hi, I forgot to update here, but I think I've solved the issue on my board. It turned out to be a signal problem on the `HSYNC` line, if I insert a 100ohm series resistor it's fine. 

and just to confirm that I didn't just "fixed it enough": the `HSYNC` signals on both cameras are like short high pulses, but every 12 or 13 pulses it's clean, all the other ones have very short low pulses (like 5 or 6 of them) in the high pulse. Is this normal? I'll post a oscilloscope graph in a day or so

crides, 2022-08-19:
Here are the oscilloscope graphs:

Overall periodic pulses (it's actually every 16 pulses it's clean):

[image]

Detail on the first and second pulses:

[image]

closed 2022-10-27 by github-actions[bot]