espressif/esp32-camera #440: Adding new sensor

PCossu93, 2022-08-19:
I am trying to use the ov7740 sensor but after a few attempts the camera does not initialize. Could you help me how to add support for ov7740?

WangYuxin-esp, 2022-08-22:
Sure. Could you put the code in your github repo? And we can work together to add support for this sensor.

PCossu93, 2022-08-23:
@WangYuxin-esp I'm just working with the "take_picture.c" example. :)
I have datasheet and application notes but i still haven't figured out how to add support for this sensor...

WangYuxin-esp, 2022-08-24:
@PCossu93 Do you have a list of registers to initialize the sensor? I can add an example, but I don't have this sensor, you need to debug it yourself.

PCossu93, 2022-08-24:
@WangYuxin-esp the link where to find the datasheet and application notes with the registers.
http://www.dragonwake.com/download/camera/vc0706_spi/OV7740_CSP.pdf
that's all I have at the moment...

WangYuxin-esp, 2022-08-24:
It seems that it is not easy to add this driver, because if we do not have this sensor for testing, we cannot know the reliability of the driver.

PCossu93, 2022-08-24:
@WangYuxin-esp I have the ov7740 sensor and the ESP32-CAM board and I can run tests on the driver and post them here :)

WangYuxin-esp, 2022-08-25:
Has the circuit package of the sensor been adjusted according to the circuit package of the official ESP32-CAM board？

PCossu93, 2022-08-25:
@WangYuxin-esp yes, I run a test with "take_picture.c" example and this is the result 
[image]

PCossu93, 2022-08-25:
@WangYuxin-esp sorry cant load the foto, here's the results

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 17:39:56
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=112f4h ( 70388) map
I (137) esp_image: segment 1: paddr=0002131c vaddr=3ffb0000 size=03590h ( 13712) load
I (142) esp_image: segment 2: paddr=000248b4 vaddr=40080000 size=0b764h ( 46948) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=285c4h (165316) map
I (222) esp_image: segment 4: paddr=000585ec vaddr=4008b764 size=0597ch ( 22908) load
I (232) esp_image: segment 5: paddr=0005df70 vaddr=50000000 size=00010h (    16) load
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
I (253) psram: This chip is ESP32-D0WD
I (255) spiram: Found 64MBit SPI RAM device
I (255) spiram: SPI RAM mode: flash 40m sram 40m
I (258) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (266) cpu_start: Pro cpu up.
I (269) cpu_start: Starting app cpu, entry point is 0x40081348
0x40081348: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (262) cpu_start: App cpu up.
I (1163) spiram: SPI SRAM memory test OK
I (1171) cpu_start: Pro cpu start user code
I (1171) cpu_start: cpu freq: 240000000
I (1171) cpu_start: Application information:
I (1174) cpu_start: Project name:     OV7740
I (1179) cpu_start: App version:      1
I (1183) cpu_start: Compile time:     Aug 25 2022 17:39:27
I (1190) cpu_start: ELF file SHA256:  b01b56f7befd59a6...
I (1196) cpu_start: ESP-IDF:          v4.4.2
I (1201) heap_init: Initializing. RAM available for dynamic allocation:
I (1208) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1214) heap_init: At 3FFB3F38 len 0002C0C8 (176 KiB): DRAM
I (1220) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1227) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1233) heap_init: At 400910E0 len 0000EF20 (59 KiB): IRAM
I (1240) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1249) spi_flash: detected chip: generic
I (1252) spi_flash: flash io: dio
I (1258) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1267) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1277) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (1287) cam_hal: cam init ok
I (1287) sccb: pin_sda 26 pin_scl 27
I (1297) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1337) camera: Detected camera at address=0x21
I (1337) camera: Detected OV7725 camera
I (1337) camera: Camera PID=0x77 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1407) esp32 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 3, dma_half_buffer_min:  3840, dma_half_buffer: 15360,lines_per_half_buffer: 12, dma_buffer_size: 30720, image_size: 153600
I (1407) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (1417) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (1427) cam_hal: cam config ok
I (1507) example:take_picture: Taking picture...
W (5507) cam_hal: Failed to get the frame on time!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d9383  PS      : 0x00060a30  A0      : 0x800f83ee  A1      : 0x3ffb6fb0
0x400d9383: app_main at C:/Espressif/frameworks/workspace/OV7740/main/main.c:124 (discriminator 13)

A2      : 0x3f4073a4  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x007b1218  A7      : 0x003fffff  A8      : 0x800d9383  A9      : 0x3ffb6f90
A10     : 0x00001583  A11     : 0x3f4073a4  A12     : 0x3f4073e8  A13     : 0x000005e3
A14     : 0x3f4073a4  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x400884a9  LEND    : 0x400884b9  LCOUNT  : 0xfffffffd
0x400884a9: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x400884b9: strlen at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace:0x400d9380:0x3ffb6fb00x400f83eb:0x3ffb6fd0 0x4008d29d:0x3ffb6ff0
0x400d9380: app_main at C:/Espressif/frameworks/workspace/OV7740/main/main.c:124 (discriminator 13)

0x400f83eb: main_task at C:/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/port_common.c:141 (discriminator 2)

0x4008d29d: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

WangYuxin-esp, 2022-08-26:
The current example shows that `"Detected camera at address=0x21"` represents that the communication of I2C is normal. I tried to write a driver here. Please refer to the `examples\basic` for testing. [ov7740_support](https://github.com/WangYuxin-esp/esp32-camera/tree/feature/add_ov7740_sensor_support).

PCossu93, 2022-08-26:
@WangYuxin-esp here the first tests result
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7004
ho 0 tail 12 room 4
load:0x40078000,len:15580
load:0x40080400,len:4492
0x40080400: _init at ??:?

entry 0x400806b4
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 20:11:49
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00100000
I (97) boot: End of partition table
I (101) boot_comm: chip revision: 3, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0f4ach ( 62636) map
I (134) esp_image: segment 1: paddr=0001f4d4 vaddr=3ffb0000 size=00b44h (  2884) load
I (135) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=20378h (131960) map
I (176) esp_image: segment 3: paddr=000403a0 vaddr=3ffb0b44 size=019bch (  6588) load
I (178) esp_image: segment 4: paddr=00041d64 vaddr=40080000 size=117b4h ( 71604) load
I (205) esp_image: segment 5: paddr=00053520 vaddr=50000000 size=00010h (    16) load
I (214) boot: Loaded app from partition at offset 0x10000
I (214) boot: Disabling RNG early entropy source...
I (227) psram: This chip is ESP32-D0WD
I (228) spiram: Found 64MBit SPI RAM device
I (228) spiram: SPI RAM mode: flash 80m sram 80m
I (231) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (238) cpu_start: Pro cpu up.
I (242) cpu_start: Starting app cpu, entry point is 0x40081368
0x40081368: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (744) spiram: SPI SRAM memory test OK
I (752) cpu_start: Pro cpu start user code
I (752) cpu_start: cpu freq: 240000000
I (752) cpu_start: Application information:
I (755) cpu_start: Project name:     blink
I (760) cpu_start: App version:      1
I (764) cpu_start: Compile time:     Aug 26 2022 20:11:24
I (770) cpu_start: ELF file SHA256:  23a287bb0c264afa...
I (776) cpu_start: ESP-IDF:          v4.4.2
I (781) heap_init: Initializing. RAM available for dynamic allocation:
I (788) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (794) heap_init: At 3FFB2EA8 len 0002D158 (180 KiB): DRAM
I (800) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (807) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (813) heap_init: At 400917B4 len 0000E84C (58 KiB): IRAM
I (819) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (828) spi_flash: detected chip: generic
I (832) spi_flash: flash io: qio
I (836) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (846) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (856) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (866) cam_hal: cam init ok
I (866) sccb: pin_sda 26 pin_scl 27
I (876) sccb: sccb_i2c_port=1
[... 64 lines cut ...]
I (788) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (794) heap_init: At 3FFB2EA8 len 0002D158 (180 KiB): DRAM
I (800) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (807) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (813) heap_init: At 400917B4 len 0000E84C (58 KiB): IRAM
I (819) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (828) spi_flash: detected chip: generic
I (832) spi_flash: flash io: qio
I (836) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (846) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (856) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (866) cam_hal: cam init ok
I (866) sccb: pin_sda 26 pin_scl 27
I (876) sccb: sccb_i2c_port=1

I (876) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (916) camera: Detected camera at address=0x21
I (916) camera: Detected OV7740 camera
I (916) camera: Camera PID=0x7742 VER=0x00 MIDL=0x00 MIDH=0x00
E (926) sccb: SCCB_Write Failed addr:0x21, reg:0x47, data:0x02, ret:-1
E (926) camera: Camera probe failed with error 0xff(ERROR)
E (926) example:take_picture: init camrea sensor fail

can you fix that?

WangYuxin-esp, 2022-08-29:
The log `"Mismatch PID=0x7742"` looks a little weird.
From the technical manual of the sensor, the device ID of the sensor should be 7740, but the value read here is 7742. Please confirm the specific model of the sensor again. Please refer to the code [here](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_ov7740_sensor_support/sensors/ov7740.c#L161)

WangYuxin-esp, 2022-09-06:
@PCossu93 , Have you reconfirmed the model of this sensor?

PCossu93, 2022-09-11:
@WangYuxin-esp hi, I apologize for the delay ... the pid written in camera registers is 7742

PCossu93, 2022-09-11:
[image]

WangYuxin-esp, 2022-09-13:
From the technical manual of the sensor, its ID num must be 0x7740. The associated registers are read-only:
[image]
The problem now is that the register list that initializes the sensor cannot be written to the sensor. refer to https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_ov7740_sensor_support/sensors/private_include/ov7740_settings.h#L144

closed 2022-11-21 by github-actions[bot]

PCossu93, 2023-03-23:
@WangYuxin-esp Hi my friend, after some debugging, i finally managed to write and read camera registers. I noticed that my ov7740 is a modified version with D0 and D1 not connected to the pins of the connector (I attach the photo and the datasheet of the ov7740). I want to use the camera in QVGA and 8-bit RAW but I can't understand which registers I have to modify to make it work in this mode. I hope for your help.
[image]
[OV7740_CSP.pdf](https://github.com/espressif/esp32-camera/files/11053065/OV7740_CSP.pdf)

WangYuxin-esp, 2023-03-24:
This is really good news. If you are already able to read and write the register of the sensor normally. It is recommended that you refer to the example here and try to batch write the configuration configuration to the register of the sensor. The [reset()](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_ov7740_sensor_support/sensors/ov7740.c#L117) will be automatically called by the system. The process of initializing the sensor is as follows:  
1. The ID of the sensor must be recognized in the [detect()](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_ov7740_sensor_support/driver/esp_camera.c#L226)
2. The [reset()](https://github.com/WangYuxin-esp/esp32-camera/blob/feature/add_ov7740_sensor_support/driver/esp_camera.c#L249) of the specified sensor will be automatically called, and the configuration of the sensor will be written to the register of the sensor.".  
What data format is currently output is not important, as long as the sensor is detected, and entering the reset() of the sensor is a very important step.

PCossu93, 2023-04-14:
here the results: 
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7004
ho 0 tail 12 room 4
load:0x40078000,len:15580
load:0x40080400,len:4492
0x40080400: _init at ??:?

entry 0x400806b4
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 19:40:50
I (29) boot: chip revision: 3
I (34) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32: SPI Speed      : 80MHz
I (42) boot.esp32: SPI Mode       : QIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0fd10h ( 64784) map
I (120) esp_image: segment 1: paddr=0001fd38 vaddr=3ffb0000 size=002e0h (   736) load
I (121) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=24e50h (151120) map
I (168) esp_image: segment 3: paddr=00044e78 vaddr=3ffb02e0 size=0229ch (  8860) load
I (171) esp_image: segment 4: paddr=0004711c vaddr=40080000 size=0fc7ch ( 64636) load
I (194) esp_image: segment 5: paddr=00056da0 vaddr=50000000 size=00010h (    16) load
I (202) boot: Loaded app from partition at offset 0x10000
I (202) boot: Disabling RNG early entropy source...
I (215) psram: This chip is ESP32-D0WD
I (215) spiram: Found 64MBit SPI RAM device
I (215) spiram: SPI RAM mode: flash 80m sram 80m
I (218) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (225) cpu_start: Pro cpu up.
I (229) cpu_start: Starting app cpu, entry point is 0x400813c0
0x400813c0: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (731) spiram: SPI SRAM memory test OK
I (739) cpu_start: Pro cpu start user code
I (739) cpu_start: cpu freq: 240000000
I (739) cpu_start: Application information:
I (742) cpu_start: Project name:     ov7740
I (746) cpu_start: App version:      1
I (751) cpu_start: Compile time:     Apr  9 2023 19:40:36
I (757) cpu_start: ELF file SHA256:  266ace8ea85ae598...
I (763) cpu_start: ESP-IDF:          v4.4.2
I (768) heap_init: Initializing. RAM available for dynamic allocation:
I (775) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (781) heap_init: At 3FFB3320 len 0002CCE0 (179 KiB): DRAM
I (787) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (794) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (800) heap_init: At 4008FC7C len 00010384 (64 KiB): IRAM
I (806) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (814) spi_flash: detected chip: gd
I (818) spi_flash: flash io: qio
I (823) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (833) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (843) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (853) cam_hal: cam init ok
I (853) sccb: pin_sda 26 pin_scl 27
I (863) sccb: sccb_i2c_port=1

I (863) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (903) camera: Detected camera at address=0x21
I (903) camera: Detected OV7740 camera
I (903) camera: Camera PID=0x7742 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1003) esp32 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 3, dma_half_buffer_min:  3840, dma_half_buffer: 15360,lines_per_half_buffer: 12, dma_buffer_size: 30720, image_size: 153600
I (1003) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (1013) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
I (1023) cam_hal: cam config ok
I (1023) ov7740: YUV422 format
I (1033) example:take_picture: Taking picture...
E (1143) cam_hal: FB-SIZE: 0 != 153600
E (1213) cam_hal: FB-SIZE: 0 != 153600
E (1273) cam_hal: FB-SIZE: 0 != 153600
E (1343) cam_hal: FB-SIZE: 0 != 153600
E (1413) cam_hal: FB-SIZE: 0 != 153600
E (1473) cam_hal: FB-SIZE: 0 != 153600
E (1543) cam_hal: FB-SIZE: 0 != 153600

PCossu93, 2023-04-14:
I think there is something wrong with the values ​​for the registers. Some values ​​are for registers that the datasheet indicates as OTP region or read only. Could you help me verify that this is the problem?

PCossu93, 2023-04-18:
The manufacturer and the seller confirmed me that this PID has the same datasheet as ov7740. I tried many values ​​for the registers looking online but only one was initializing the camera without errors and this is the result. Can you help me?

[image]

PCossu93, 2023-04-18:
Here my esp-idf project code:

#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include <string.h>
#include <esp_event.h>
#include <driver/rtc_io.h>
#include <driver/gpio.h>
#include <stddef.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// LCD Library
#define LGFX_USE_V1
#include "LovyanGFX.hpp"
#include "lgfx_user/LGFX_ESP32_sample.hpp"

// support IDF 5.x
#ifndef portTICK_RATE_MS
#define portTICK_RATE_MS portTICK_PERIOD_MS
#endif

#include "esp_camera.h"

// ESP32Cam (AiThinker) PIN Map

#define CAM_PIN_PWDN 32
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK 0
#define CAM_PIN_SIOD 26
#define CAM_PIN_SIOC 27
#define CAM_PIN_D7 35
#define CAM_PIN_D6 34
#define CAM_PIN_D5 39
#define CAM_PIN_D4 36
#define CAM_PIN_D3 21
#define CAM_PIN_D2 19
#define CAM_PIN_D1 18
#define CAM_PIN_D0 5
#define CAM_PIN_VSYNC 25
#define CAM_PIN_HREF 23
#define CAM_PIN_PCLK 22

static const char *TAG = "example:take_picture";

static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    .xclk_freq_hz = 24000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_YUV422, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_QVGA,    //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 10, //0-63 lower number means higher quality
    .fb_count = 1,       //if more than one, i2s runs in continuous mode. Use only with JPEG
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
	.sccb_i2c_port = 1,

};

// 1.28inch round lcd display GC9A01
LGFX lcd;

extern "C" {

	void app_main(void) {
		esp_camera_init(&camera_config);
		camera_fb_t * pic = NULL;
		vTaskDelay(100/portTICK_RATE_MS);
		lcd.init();
		lcd.setColorDepth(16);
		lcd.setRotation(2);
		lcd.setBrightness(100);
		vTaskDelay(100/portTICK_RATE_MS);
		for(;1;) {
				pic = esp_camera_fb_get();
				lcd.pushImage(0,0,320,240,(const uint16_t*)pic->buf);
				esp_camera_fb_return(pic);
		}
	}
}

PCossu93, 2023-04-18:
and here the register i used for initialization:

        {0x0e, 0x00},

	{0x12, 0x80},
	/* flag for soft reset delay */
	{0xFF, 0x10},
	{0x13, 0x00},

	/**************************************************************/
	/*  30fps  11 01 ;clock_divider ;sysclk=24MHz at XCLK=24MHz   */
	/*  20fps  11 02 ;clock_divider ;sysclk=16MHz at XCLK=24MHz   */
	/*  15fps  11 03 ;clock_divider ;sysclk=12MHz at XCLK=24MHz   */
	/*  10fps  11 05 ;sysclk=8MHz at XCLK=24MHz                   */
	/*  7p5fps 11 07 ;sysclk=6MHz at XCLK=24MHz                   */
	/**************************************************************/
	{0x11, 0x03},
	/**************************************************************/

	{0x12, 0x00},
	{0xd5, 0x10},
	{0x0d, 0x34},
	{0x16, 0x01},
	{0x17, 0x25},
	{0x18, 0xa0},
	{0x19, 0x03},
	{0x1a, 0xf0},
	{0x1b, 0x89},
	{0x22, 0x03},
	{0x29, 0x18},
	{0x2b, 0xf8},
	{0x2c, 0x01},
	{0x31, 0xa0},
	{0x32, 0xf0},
	{0x33, 0xc4},
	{0x3a, 0xb4},
	{0x36, 0x3f},

	{0x04, 0x60},
	{0x27, 0x80},
	{0x3d, 0x0f},
	{0x3e, 0x80},
	{0x3f, 0x40},
	{0x40, 0x7f},
	{0x41, 0x6a},
	{0x42, 0x29},
	{0x44, 0xe5},
	{0x45, 0x41},
	{0x47, 0x02},
	{0x49, 0x64},
	{0x4a, 0xa1},
	{0x4b, 0x70},
	{0x4c, 0x1a},
	{0x4d, 0x50},
	{0x4e, 0x13},
	{0x64, 0x00},
	{0x67, 0x88},
	{0x68, 0x1a},

	{0x14, 0x38},
	{0x24, 0x3c},
	{0x25, 0x30},
	{0x26, 0x72},
	{0x50, 0x97},
	{0x51, 0x7e},
	{0x52, 0x00},
	{0x53, 0x00},
	{0x20, 0x00},
	{0x21, 0x23},
	/*********************************/
	/* To enable Static Test Pattern */
	/*********************************/
	/* {0x38, 0x07}, */
	/* {0x84, 0x02}, */

	/*********************************/
	/* Normal Mode / No test pattern */
	{0x38, 0x14},
	/*********************************/
	{0xe9, 0x00},
	{0x56, 0x55},
	{0x57, 0xff},
	{0x58, 0xff},
	{0x59, 0xff},
	{0x5f, 0x04},
	{0xec, 0x00},
	{0x13, 0xff},

	{0x80, 0x7f},
	{0x81, 0x3f},
	{0x82, 0x32},
	{0x83, 0x01},
	{0x38, 0x11},
	{0x85, 0x00},
	{0x86, 0x03},
	{0x87, 0x01},
	{0x88, 0x05},
	{0x89, 0x30},
	{0x8d, 0x30},
	{0x8f, 0x85},
	{0x93, 0x30},
	{0x95, 0x85},
	{0x99, 0x30},
	{0x9b, 0x85},

	{0x9c, 0x08},
	{0x9d, 0x12},
	{0x9e, 0x23},
	{0x9f, 0x45},
	{0xa0, 0x55},
	{0xa1, 0x64},
	{0xa2, 0x72},
	{0xa3, 0x7f},
	{0xa4, 0x8b},
	{0xa5, 0x95},
	{0xa6, 0xa7},
	{0xa7, 0xb5},
	{0xa8, 0xcb},
	{0xa9, 0xdd},
	{0xaa, 0xec},
	{0xab, 0x1a},

	{0xce, 0x78},
	{0xcf, 0x6e},
	{0xd0, 0x0a},
	{0xd1, 0x0c},
	{0xd2, 0x84},
	{0xd3, 0x90},
	{0xd4, 0x1e},

	{0x5a, 0x24},
	{0x5b, 0x1f},
	{0x5c, 0x88},
	{0x5d, 0x60},

	{0xac, 0x6e},
	{0xbe, 0xff},
	{0xbf, 0x00},

	/* 320x240 */

	{0x31, 0x50},
	{0x32, 0x78},
	{0x82, 0x3f},

	/* YUV */
	{0x12, 0x00},
	{0x36, 0x3f},
	{0x53, 0x00},

	{0x33, 0xc4},
	{0x1b, 0x89},
	{0x22, 0x03},

	/* VSYNC, inverse */
	{0x28, 0x2},
	{0xFF, 0xFF}

me-no-dev, 2023-04-18:
your image is probably fine, but you need to convert YUYV to RGB565. [this function](https://github.com/espressif/esp32-camera/blob/master/conversions/include/img_converters.h#L122) can turn your buffer into RGB888, which you can then turn into RGB565 easily and display it.