espressif/esp32-camera #443: ESP32 freezes when serial port is opened.

nguterresn, 2022-08-29:
As the title states, when opening the serial port (via Monitor for example), the ESP32 freezes.

It really doesn't matter if the `Serial.begin` is called or not, it will freeze anyways.
I tested this with different cables, different ports, different computers - nothing works and seems to be a problem of the ESP32 only, since ESP8266 work correctly.

My board is the ESP32-CAM-MB and it works seamlessly until the serial port is opened. I tested running a simple blink code and the blink would stop whenever the serial port was opened.

Does anyone has fix for this?

me-no-dev, 2022-08-29:
you probably have wired the USB-UART wrongly or the terminal software you are using is setting wrong values for DTR and RTS. With wrong DTR/RTS levels, the board will go into reset/download or similar mode that you do not want.

nguterresn, 2022-08-29:
From the serial monitor:

```js
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
```

DTR and RTS disabled.

[image]

ESP32 is connected using a usb cable.