espressif/esp32-camera #446: CONFIG_SPIRAM_SUPPORT not present in idf.py 5.1 (master branch)

wuyuanyi135, 2022-08-31:
In the format conversion c files there are some branches checking whether `CONFIG_SPIRAM_SUPPORT` is set for using the external PSRAM.

https://github.com/espressif/esp32-camera/blob/56119890f5ca01bade9980d2abfe568e1d5f34a8/conversions/jpge.cpp#L33

 I am currently using the master branch of idf.py (v5.1-dev-436-g5c1044d84d), seems the `CONFIG_SPIRAM_SUPPORT` no longer exists, but only `CONFIG_SPIRAM=y` present. I tried to add `CONFIG_SPIRAM_SUPPORT=y` in my `sdkconfig` but it got removed after build, hence no way to enable the PSRAM support. 

I wonder how should I configure to use the external RAM in this case?

igrr, 2022-08-31:
For the ESP32 target the build system automatically adds CONFIG_SPIRAM_SUPPORT define to sdkconfig, if CONFIG_SPIRAM is defined(https://github.com/espressif/esp-idf/blob/5c1044d84d625219eafa18c24758d9f0e4006b2c/components/esp_hw_support/sdkconfig.rename.esp32#L20).

We don't do this for other build targets, though. I'll check if this is something we can change on the IDF side.

igrr, 2022-08-31:
Another option might be to not do the compile time check, but instead use
```c++
heap_caps_malloc_prefer(nSize, 2, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT, MALLOC_CAP_DEFAULT);
```
This should automatically provide the external memory if it is available and internal memory otherwise. This also accounts for the case when CONFIG_SPIRAM_SUPPORT=y && CONFIG_SPIRAM_IGNORE_NOTFOUND=y and PSRAM was not actually initialized.

wuyuanyi135, 2022-08-31:
Indeed. Actually these flags are all in the format conversion and the frame buffer allocation is not rely on these flags. So the current configuration actually worked for me.

closed 2022-10-10 by wuyuanyi135