espressif/esp32-camera #447: OV2640 JPEG Blue and Red swapped

y3nd, 2022-09-02:
Hello,

We have an issue with some OV2640 sensors where the JPEG image produced by the camera in SVGA mode has always the blue and red channels swapped : red objects are completly blue and blue objects are completly red... The JPEG file does not seem corrupted.
The green channel is the same.

This problem does not occur in CIF or UXGA modes

WangYuxin-esp, 2022-09-19:
Hi, this error is most likely a problem with the sensor itself. Because the JPEG data is completely generated by the sensor itself, the ESP32 just receives the data. I suggest you replace the sensor and test it again using the example [here](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/pic_server).

y3nd, 2022-11-25:
This was due to an invalid window setting

closed 2022-11-25 by y3nd

Systanet, 2024-02-16:
I've got exactly the same blue/red swap issue whether using [espressive/esp32 OV2640 library](https://github.com/espressif/esp32-camera/blob/master/sensors/ov2640.c) and the set_window function from UXGA to 800 x 600.   Please could you share a little more detail on your 'invalid window setting'  fix?