espressif/esp32-camera #449: Unclear examples

KenwoodFox, 2022-09-09:
This also relates a little to #448 

Examples area little unclear, heres what im working with currently. https://github.com/SNHU-Robotics/SumoBot-Joe/tree/feat/prototytpe

Seems to compile fine! but.. where is the jpeg stream and what address and do i need to attach it to the server itself?

I'd love to include some more rich examples but i'd like to get one think working first!

WangYuxin-esp, 2022-09-14:
Here are some small examples, please refer to [camera examples](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera) or [esp-who](https://github.com/espressif/esp-who).

WangYuxin-esp, 2022-09-14:
Maybe we should add a link to a related example in the readme document.

KenwoodFox, 2022-09-14:
I think that would help a lot!

closed 2022-11-22 by github-actions[bot]