espressif/esp32-camera #450: Camera probe failed with error 0x105

lucianobonzatto, 2022-09-10:
hello, i'm tring to get image form ESP32-CAM using platformio. 
this is my fededback message: 

E (2207) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

and my platformio.ini:

[env:esp32dev]
platform = espressif32
board = esp-wrover-kit
framework = arduino
monitor_speed = 115200

It's necessary set DBOARD_HAS_PSRAM?

WangYuxin-esp, 2022-09-14:
Before officially initializing the camera, ESP32 will send a command to the sensor through the I2C bus. Normally, the sensor should send an ACK, and then ESP32 can initialize the sensor through the I2C bus. The current log shows that the sensor has not sent an ACK. Therefore, the configuration of the ESP32 pin is abnormal, or the sensor is working in an abnormal state.

yashshukla99, 2022-09-14:
I'm also facing same issue, even afte camera initialization done.

yashshukla99, 2022-09-14:
I'm also facing same issue, even after camera initialization done.

[image]

WangYuxin-esp, 2022-09-14:
@yashshukla99 Is the official development board currently used? Which sensor model is used? And it is recommended that you use the example [here](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/basic) for testing.

yashshukla99, 2022-09-14:
Thanks @WangYuxin-esp , I will test above code and I am using [this](https://miro.medium.com/max/2400/1*lv0RGuCyNtIg-lFgVaWsRQ.png) one with OV2640 sensor.

Jason2866, 2022-11-11:
@yashshukla99 I have several of this AI-Thinker (or faked clone?). The are borderline.
 - without a very stable good power supply the always fail
 - dont use a breadboard
 - disconnect anything from GPIO 0 after flashing (clk for cam!)
if you are lucky you get a picture. Many of this boards do not and just fail with 0x105

aroerina, 2022-12-08:
I often face this problem, but it is usually due to a poor connection of the FPC connector, which can be solved by inserting the connector all the way to the back.

OmPatel47097, 2023-02-01:
Tried all of the solutions, I think I got scamed or my electrical skills are too bad  : /

Jason2866, 2023-02-01:
@OmPatel47097  Probably you have got just bad luck with the one you have.
Since i use ESP32-S3 Cam Modells never had any problem.

OmPatel47097, 2023-02-08:
> @OmPatel47097 Probably you have got just bad luck with the one you have. Since i use ESP32-S3 Cam Modells never had any problem.

Probably I might need to buy a new one

maboeckx, 2023-02-14:
you shoud trie to use the CAMERA_MODEL_AI_THINKER board
worked for me when i had this problem

sergioaiello, 2023-02-15:
I faced the same issue. With the same log.
I solved changing from I2C port#1 (default) to port#0.

durop, 2023-02-17:
@sergioaiello what do you mean by changing I2c port 1 to 2? 
Do you have some sketch?

sergioaiello, 2023-02-17:
no, it is very easy.
Go to "menuconfig" ("F1" if you use VisualCode editor) and search for "SCCB" (OV camera's interface)
You should get the menu "I2C peripheral to use for SCCB" with two choices: "I2C0" and "I2C1"
Apparently the default is the latter. Change to the former and compile after a "fullclean". Give it a try and let me know.

durop, 2023-02-18:
@sergioaiello, thanks for the response. Im using vscode with PlatafformIO extension. 
 I dont get any result when searching "SCCB".

sergioaiello, 2023-02-18:
the trigger is the file Kconfig.projbuild
Is that file relevant to platformio extension?
I copy-paste the excerpt related to I2C port selection:
-----------------------------------
    choice SCCB_HARDWARE_I2C_PORT
        bool "I2C peripheral to use for SCCB"
        default SCCB_HARDWARE_I2C_PORT1

        config SCCB_HARDWARE_I2C_PORT0
            bool "I2C0"
        config SCCB_HARDWARE_I2C_PORT1
            bool "I2C1"

    endchoice
 ------------------------------
You can check the file
https://github.com/espressif/esp32-camera/blob/master/Kconfig 

Also you can get the entire github project that refers to the Kconfig.projbuild file.
talk to you then

Deewansonic, 2023-03-05:
#define CAMERA_MODEL_AI_THINKER // Has PSRAM
using/ uncommenting the above worked for me.

psi-4ward, 2023-03-06:
> #define CAMERA_MODEL_AI_THINKER // Has PSRAM using/ uncommenting the above worked for me.

Can you share your sketch (and probably platform.ini)?

Deewansonic, 2023-03-06:
> > #define CAMERA_MODEL_AI_THINKER // Has PSRAM using/ uncommenting the above worked for me.
> 
> Can you share your sketch (and probably platform.ini)?

#include "esp_camera.h" 
 #include <WiFi.h> 
  
 // 
 // WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality 
 //            Ensure ESP32 Wrover Module or other board with PSRAM is selected 
 //            Partial images will be transmitted if image exceeds buffer size 
 // 
 //            You must select partition scheme from the board menu that has at least 3MB APP space. 
 //            Face Recognition is DISABLED for ESP32 and ESP32-S2, because it takes up from 15  
 //            seconds to process single frame. Face Detection is ENABLED if PSRAM is enabled as well 
  
 // =================== 
 // Select camera model 
 // =================== 
 //#define CAMERA_MODEL_WROVER_KIT // Has PSRAM 
// #define CAMERA_MODEL_ESP_EYE // Has PSRAM 
 //#define CAMERA_MODEL_ESP32S3_EYE // Has PSRAM 
 //#define CAMERA_MODEL_M5STACK_PSRAM // Has PSRAM 
 //#define CAMERA_MODEL_M5STACK_V2_PSRAM // M5Camera version B Has PSRAM 
 //#define CAMERA_MODEL_M5STACK_WIDE // Has PSRAM 
 //#define CAMERA_MODEL_M5STACK_ESP32CAM // No PSRAM 
 //#define CAMERA_MODEL_M5STACK_UNITCAM // No PSRAM 
 #define CAMERA_MODEL_AI_THINKER // Has PSRAM 
 //#define CAMERA_MODEL_TTGO_T_JOURNAL // No PSRAM 
 // ** Espressif Internal Boards ** 
 //#define CAMERA_MODEL_ESP32_CAM_BOARD 
 //#define CAMERA_MODEL_ESP32S2_CAM_BOARD 
 //#define CAMERA_MODEL_ESP32S3_CAM_LCD 
  
 #include "camera_pins.h" 
  
 // =========================== 
 // Enter your WiFi credentials 
 // =========================== 
 const char* ssid = "**********"; 
 const char* password = "**********"; 
  
 void startCameraServer(); 
 void setupLedFlash(int pin); 
  
 void setup() { 
   Serial.begin(115200); 
   Serial.setDebugOutput(true); 
   Serial.println(); 
  
   camera_config_t config; 
   config.ledc_channel = LEDC_CHANNEL_0; 
   config.ledc_timer = LEDC_TIMER_0; 
   config.pin_d0 = Y2_GPIO_NUM; 
   config.pin_d1 = Y3_GPIO_NUM; 
   config.pin_d2 = Y4_GPIO_NUM; 
   config.pin_d3 = Y5_GPIO_NUM; 
   config.pin_d4 = Y6_GPIO_NUM; 
   config.pin_d5 = Y7_GPIO_NUM; 
   config.pin_d6 = Y8_GPIO_NUM; 
   config.pin_d7 = Y9_GPIO_NUM; 
   config.pin_xclk = XCLK_GPIO_NUM; 
   config.pin_pclk = PCLK_GPIO_NUM; 
   config.pin_vsync = VSYNC_GPIO_NUM; 
   config.pin_href = HREF_GPIO_NUM; 
   config.pin_sscb_sda = SIOD_GPIO_NUM; 
   config.pin_sscb_scl = SIOC_GPIO_NUM; 
   config.pin_pwdn = PWDN_GPIO_NUM; 
   config.pin_reset = RESET_GPIO_NUM; 
   config.xclk_freq_hz = 20000000; 
   config.frame_size = FRAMESIZE_UXGA; 
   config.pixel_format = PIXFORMAT_JPEG; // for streaming 
   //config.pixel_format = PIXFORMAT_RGB565; // for face detection/recognition 
   config.grab_mode = CAMERA_GRAB_WHEN_EMPTY; 
   config.fb_location = CAMERA_FB_IN_PSRAM; 
   config.jpeg_quality = 12; 
   config.fb_count = 1; 
    
   // if PSRAM IC present, init with UXGA resolution and higher JPEG quality 
   //                      for larger pre-allocated frame buffer. 
   if(config.pixel_format == PIXFORMAT_JPEG){ 
     if(psramFound()){ 
       config.jpeg_quality = 10; 
       config.fb_count = 2; 
       config.grab_mode = CAMERA_GRAB_LATEST; 
     } else { 
       // Limit the frame size when PSRAM is not available 
       config.frame_size = FRAMESIZE_SVGA; 
       config.fb_location = CAMERA_FB_IN_DRAM; 
     } 
   } else { 
     // Best option for face detection/recognition 
     config.frame_size = FRAMESIZE_240X240; 
 #if CONFIG_IDF_TARGET_ESP32S3 
     config.fb_count = 2; 
 #endif 
   } 
  
 #if defined(CAMERA_MODEL_ESP_EYE) 
   pinMode(13, INPUT_PULLUP); 
   pinMode(14, INPUT_PULLUP); 
 #endif 
  
   // camera init 
   esp_err_t err = esp_camera_init(&config); 
   if (err != ESP_OK) { 
     Serial.printf("Camera init failed with error 0x%x", err); 
     return; 
   } 
  
   sensor_t * s = esp_camera_sensor_get(); 
   // initial sensors are flipped vertically and colors are a bit saturated 
   if (s->id.PID == OV3660_PID) { 
     s->set_vflip(s, 1); // flip it back 
     s->set_brightness(s, 1); // up the brightness just a bit 
     s->set_saturation(s, -2); // lower the saturation 
   } 
   // drop down frame size for higher initial frame rate 
   if(config.pixel_format == PIXFORMAT_JPEG){ 
     s->set_framesize(s, FRAMESIZE_QVGA); 
   } 
  
 #if defined(CAMERA_MODEL_M5STACK_WIDE) || defined(CAMERA_MODEL_M5STACK_ESP32CAM) 
   s->set_vflip(s, 1); 
   s->set_hmirror(s, 1); 
 #endif 
  
 #if defined(CAMERA_MODEL_ESP32S3_EYE) 
   s->set_vflip(s, 1); 
 #endif 
  
 // Setup LED FLash if LED pin is defined in camera_pins.h 
 #if defined(LED_GPIO_NUM) 
   setupLedFlash(LED_GPIO_NUM); 
 #endif 
  
   WiFi.begin(ssid, password); 
   WiFi.setSleep(false); 
  
   while (WiFi.status() != WL_CONNECTED) { 
     delay(500); 
     Serial.print("."); 
   } 
   Serial.println(""); 
   Serial.println("WiFi connected"); 
  
   startCameraServer(); 
  
   Serial.print("Camera Ready! Use 'http://"); 
   Serial.print(WiFi.localIP()); 
   Serial.println("' to connect"); 
 } 
  
 void loop() { 
   // Do nothing. Everything is done in another task by the web server 
   delay(10000); 
 }

Deewansonic, 2023-03-06:
> #include "esp_camera.h"
>  #include <WiFi.h>
>
>  //
>  // WARNING!!! PSRAM IC required for UXGA resolution and high JPEG quality
>  //            Ensure ESP32 Wrover Module or other board with PSRAM [...]

So far, the modification has been made already.  Replace with your code.

psi-4ward, 2023-03-07:
Have some success today:

* Used the latest GIT Sources from `arduino-esp32`
* Created the project by importing it from `arduino-esp32/libraries/ESP32/examples/Camera/CameraWebServer`
* Used `#define CAMERA_MODEL_AI_THINKER` AND  `#define CAMERA_MODEL_ESP32_CAM_BOARD`
* Powered the Devices by a stable 5V and left VCC from the Programmer unconnected
* Removed and replugged the camera module

Three devices working, one gives me:
```
E (541) sccb: SCCB_Write Failed addr:0x30, reg:0x37, data:0xc3, ret:263
E (542) camera: Camera probe failed with error 0xffffffff(ESP_FAIL)
Camera init failed with error 0xffffffff
```
So this board has a damaged PSRAM?

Deewansonic, 2023-03-07:
You only could uncomment one board particularly. Use #define CAMERA_MODEL_AI_THINKER only please. Don't use two boards. It should work.

psi-4ward, 2023-03-07:
> You only could uncomment one board particularly. Use #define CAMERA_MODEL_AI_THINKER only please. Don't use two boards. It should work. 

Does not make any difference. Still get the `SCCB_Write Failed`

Rombersoft, 2023-04-02:
I've resolved the same problem. I've selected "Ai Thinker ESP-32 cam" board. Flash mode "DIO". Partition Scheme "Huge APP (3Mb...)". And it's all. My Cam works fine. This is my board.
[image]

closed 2023-06-09 by github-actions[bot]

xstable, 2023-09-04:
> So far, the modification has been made already. Replace with your code.

You made my day!!! Awesome. With your modification it work for me. 
Thanks  a lot!

aaalvinnn, 2023-10-07:
I face the same problem and have managed to fix it:
my device is esp32cam from AI-thinker, and it turns out to be that i forgot to set the Pin of it which leads to this problem.
[image]

maralian, 2023-11-05:
Hi, the problem is in the programming example camera web server

I modified the programming and managed to fix it

use this in the folder (camerawebserver.ino) if you are using arduino.ide

#include "esp_camera.h"
#include <WiFi.h>
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "fb_gfx.h"
#include "soc/soc.h" 
#include "soc/rtc_cntl_reg.h"  
#include "esp_http_server.h"
const char* ssid = "CASA_DA_MARA";
const char* password = "Maralian1234";
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22
#define PART_BOUNDARY "123456789000000000000987654321"
static const char* _STREAM_CONTENT_TYPE = "multipart/x-mixed-replace;boundary=" PART_BOUNDARY;
static const char* _STREAM_BOUNDARY = "\r\n--" PART_BOUNDARY "\r\n";
static const char* _STREAM_PART = "Content-Type: image/jpeg\r\nContent-Length: %u\r\n\r\n";
static esp_err_t stream_handler(httpd_req_t *req){
  camera_fb_t * fb = NULL;
  esp_err_t res = ESP_OK;
  size_t _jpg_buf_len = 0;
  uint8_t * _jpg_buf = NULL;
  char * part_buf[64];
  res = httpd_resp_set_type(req, _STREAM_CONTENT_TYPE);
  if(res != ESP_OK){
    return res;
  }
  while(true){
    fb = esp_camera_fb_get();
    if (!fb) {
      Serial.println("Camera capture failed");
      res = ESP_FAIL;
    } else {
      if(fb->width > 400){
        if(fb->format != PIXFORMAT_JPEG){
          bool jpeg_converted = frame2jpg(fb, 80, &_jpg_buf, &_jpg_buf_len);
          esp_camera_fb_return(fb);
          fb = NULL;
          if(!jpeg_converted){
            Serial.println("JPEG compression failed");
            res = ESP_FAIL;
          }
        } else {
          _jpg_buf_len = fb->len;
          _jpg_buf = fb->buf;
        }
      }
    }
    if(res == ESP_OK){
      size_t hlen = snprintf((char *)part_buf, 64, _STREAM_PART, _jpg_buf_len);
      res = httpd_resp_send_chunk(req, (const char *)part_buf, hlen);
    }
    if(res == ESP_OK){
      res = httpd_resp_send_chunk(req, (const char *)_jpg_buf, _jpg_buf_len);
    }
    if(res == ESP_OK){
      res = httpd_resp_send_chunk(req, _STREAM_BOUNDARY, strlen(_STREAM_BOUNDARY));
    }
    if(fb){
      esp_camera_fb_return(fb);
      fb = NULL;
      _jpg_buf = NULL;
    } else if(_jpg_buf){
      free(_jpg_buf);
      _jpg_buf = NULL;
    }
    if(res != ESP_OK){
      break;
    }
  }
  return res;
}
void startCameraServer();

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //desativa o detector de brownout
 
  Serial.begin(115200);
  Serial.setDebugOutput(false);
  
  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.grab_mode = CAMERA_GRAB_WHEN_EMPTY;
  config.fb_location = CAMERA_FB_IN_PSRAM;
  config.frame_size = FRAMESIZE_UXGA;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  if(config.pixel_format == PIXFORMAT_JPEG){
    if(psramFound()){
      config.jpeg_quality = 10;
      config.fb_count = 2;
      config.grab_mode = CAMERA_GRAB_LATEST;
    } else {
      config.frame_size = FRAMESIZE_SVGA;
      config.fb_location = CAMERA_FB_IN_DRAM;
    }
  } else {
    // Best option for face detection/recognition
    config.frame_size = FRAMESIZE_240X240;
#if CONFIG_IDF_TARGET_ESP32S3
    config.fb_count = 2;
#endif
  }
  
  // Inicia a câmera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }
  // Conecta o WiFi
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  Serial.print("Camera Stream Ready! Go to: http://");
  Serial.print(WiFi.localIP());
  
  // Start streaming web server
  startCameraServer();
}
void loop() {
  delay(1);  //nenhuma ação relevante é realizada no loop
}

xstable, 2023-11-06:
@maralian please always put sourcecode in three backticks : 
\`\`\` <linebreak>
your-code-goes-here <linebreak>
\`\`\`

sunjialinsun, 2023-12-24:
> hello, i'm tring to get image form ESP32-CAM using platformio. this is my fededback message:
>
> E (2207) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND) Camera init failed with error [...]

I get the same issue , and figure out is is due to select the worng camera model. you just need update the .ino file with correct cam model.

DNS, 2024-05-04:
> Camera probe failed with error 0x105

In my case, it's caused by poor quality programmer/uploader (ESP32-CAM-MB).
The female pin was bent. Sometimes it cause error 0x105, and sometimes success but show very bad quality videos.

When I changed to the other programmer, the problem was solved.

Hairyloon, 2024-05-19:
Is it too late to join in this discussion?
I have a similar problem.  Curiously, I get a different error if I flash it from a different PC; my guess is that the other one has an older set of libraries, or some such thing, rather than being the difference between machines.

On this one, I get:
```
E (543) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105
```

On the other one I get:
```
[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004
```

I'm also getting these warnings:
```
W (179) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
W (301) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
```

dixieflatline42, 2024-05-22:
> Is it too late to join in this discussion? I have a similar problem. Curiously, I get a different error if I flash it from a different PC; my guess is that the other one has an older set of [...]

Man, I was having the same problem. Use this reference to flash your code: 

https://github.com/espressif/esp-iot-solution/tree/master/examples/camera/basic

It's work to me.

Hairyloon, 2024-05-24:
Thanks. That doesn't seem to work on mine: maybe my camera is dud.  There is another one here somewhere. I'll try swapping them about a bit.

Bongumsa, 2024-06-03:
The problem is mostly caused by incorrectly defining camera pins. The esp32 wrover kit has a different camera pin configuration that the espcam by AI thinker.

Hairyloon, 2024-06-03:
> The problem is mostly caused by incorrectly defining camera pins. The esp32 wrover kit has a different camera pin configuration that the espcam by AI thinker.

I would say "Often" rather than "Mostly", and I have tried both.
Would it be a lot of work to write a sketch that runs through the options and reports back on which one is right?
If it is the most common cause of this problem, then maybe it would be worth doing?

Or has somebody done it already?

Bummbumm6, 2024-06-05:
After checking, I barely didn't provide enough power for the camera to run properly, and it drove me crazy hahaha

zeta-omega-chi, 2024-07-29:
> * Used `#define CAMERA_MODEL_AI_THINKER` AND  `#define CAMERA_MODEL_ESP32_CAM_BOARD`

This works for esp32-cam with ov2460 from AIThinker

Piyush-GG, 2024-08-14:
I have this issue too and I need a solution as soon as possible, I don't really know how to fix this.

my board is the xiao esp32 s3 sense and it keeps on giving the cam init error:
Camera init failed with error 0x105

thanks a lot!

WangYuxin-esp, 2024-08-15:
This may be a hardware issue, check the I2C (SCCB) pins on the baseboard.

Piyush-GG, 2024-08-15:
I ran a 12c check and it said nothing was connected to it

WangYuxin-esp, 2024-08-15:
Is the power pin of the sensor normal? Two devices with normal power supply and correctly connected I2C pins should be able to communicate with each other.

Piyush-GG, 2024-08-16:
how do i check the connections?

saisankeerth40, 2024-08-20:
i got this error please i have to submit my project on 24 august

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
E (358) esp_cor⸮⸮VW}⸮⸮⸮͡⸮ No core dump partition found!
E (358) esp_core_dump_flash: No core dump partition found!

E (512) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
Camera init failed with error 0x105

i use arduino ide

Radmund, 2024-08-22:
> > * Used `#define CAMERA_MODEL_AI_THINKER` AND  `#define CAMERA_MODEL_ESP32_CAM_BOARD`
> 
> This works for esp32-cam with ov2460 from AIThinker

I also did this hack:

#define CAMERA_MODEL_AI_THINKER` AND  `#define CAMERA_MODEL_ESP32_CAM_BOARD`

And it worked for me too. I was searching for days and sent even back some ESP32-CAM back, but now I got one from a good provider and still had the problem. after this hack it worked to my surprise. 
I must say I use the programmerboard ESP32-CAM-MB  and the ESP32-CAM together and connect it to an USB port and now I see an IP address and have acces to the camserver. I must still test it without the programmerboard.Thanks

hollerller, 2024-09-07:
Hi, I'm using platformio with espressif and it's giving me the same error 0x105(ESP_ERR_NOT_FOUND).

```
␛[0;32mI (871) sccb: sccb_i2c_port=0␛[0m
␛[0;32mI (871) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;31mE (911) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)␛[0m
```

When I upload the Arduino code, it works, so I think it's not a problem with the board itself.

Is there anything else I can try? I also changed the i2c port to 0 as recommended, and tried several examples, but nothinkg works.

gpanton, 2024-10-21:
This is late to the conversation, however I was experiencing many issues including memory. Strangle I was testing with two AI-thinkers, one working the other not.
Compared the boot sequence and nothing apparent.
Giving up I replaced the failing espcam. Wow, the same issue. I then out of frustration decided to replace the camera.
Bugger me, the problem went away.
Maybe I just did not seat the camera correctly, maybe a faulty camera, who cares it's resolved.

Logicbloke, 2024-11-02:
1. Check the pinout for your **exact** model.
2. Make sure it matches something on the camera_pins.h header file, this is included in the example.
3. Create your own if none present in the header file are matching.
4. Profit!

abdullahtaha262, 2024-12-13:
I was able to fix it before I completely lose hope!

if I search the cam from the model, I get the "Ai thinker"
However, when I uncomment it on the example it wasn't working.

the solution was to uncomment a different one for my case it was
'''
#define CAMERA_MODEL_WROVER_KIT
'''

my board is ESP32 wrover dev

Xertyfied, 2024-12-18:
For me the problem was this uncommented line : '''define CAMERA_MODEL_ESP_EYE  // Has PSRAM''' , i commented it and it worked fine, ai thinker with mb, #define CAMERA_MODEL_ESP32S2_CAM_BOARD and #define CAMERA_MODEL_AI_THINKER

MohammadR-Kh, 2025-04-03:
I had the same error while trying to upload the code to cam, also I was trying to program it with another esp32, so here's how I did it.
first connect the TX of cam to TX of esp32, RX to RX, (only if you are doing it with another ESP32 like me), power it with 5V and GND, then connect the EN of esp32 to GND, while holding the button (IO0 of cam to GND) press upload, the as you see "Connecting..." give the reset button of the cam a single press while holding the button (IO0 to GND), you should see the uploading starts, after that release the button and press the reset button of cam, it worked for me while trying to program the ESP32-CAM using another ESP32

WangYuxin-esp, 2025-05-21:
See the documentation [here](https://github.com/espressif/esp-who/blob/master/docs/zh_CN/get-started/ESP32-S3-EYE_Getting_Started_Guide.md) and the [examples](https://github.com/espressif/esp-iot-solution/tree/master/examples/camera) here.

Sujjiiiya, 2025-05-22:
@WangYuxin-esp 
Thank you so much for the quick response and helpful links!

I’ve reviewed the ESP32-S3-EYE Getting Started Guide and the camera examples in `esp-iot-solution`. They look great, but I’m using a different board — specifically an **ESP32-S3 WROOM (N16R8)** development board with a built-in OV2640 FPC connector.

So far, I’ve been using the following pin configuration based on other ESP32-S3 WROOM examples:

```cpp
#define SIOD_GPIO_NUM 40
#define SIOC_GPIO_NUM 39

Unfortunately, the camera still fails to initialize (ESP_ERR_NOT_FOUND). If anyone is familiar with the exact GPIO mappings or internal routing for the OV2640 on this board, I’d really appreciate any insight or confirmation.

Sujjiiiya, 2025-05-22:
Hi again!

Just to be 100% sure, I’m attaching a picture of my board with the OV2640 camera ribbon inserted. The ribbon does not go more in than this TT

Could someone please take a look and let me know if the ribbon is inserted properly? If you notice anything wrong with the orientation or connection, I’d really appreciate your advice.

Thanks in advance!

[image]

WangYuxin-esp, 2025-05-22:
The connection in the picture is normal.
First, please apply for the circuit design of the development board from the factory where you purchased it. This will allow you to check the pins and levels of the camera interface.
Second, the conditions for the normal operation of the camera sensor are:
1) Normal power supply: including DOVDD, AVDD, DVDD
2) XCLK: XCLK is used to provide clock for the sensor. You can use an oscilloscope to measure whether the XCLK pin has clock output.
3) The levels of the PWDN and RST pins meet the requirements of the sensor.
The best way is to buy a development board with more information, which will reduce the difficulty of development.

Sujjiiiya, 2025-05-23:
Hello again , it looks like the board name I am using is correct and now I am getting this in my serial monitor , and the baud rate is 115200.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8

and thats all comes in the serial monitor and noting like wifi connecting or a link further

Sujjiiiya, 2025-05-23:
Final Update — RESOLVED

Hi everyone! Just wanted to share that my issue is finally resolved

After trying many things (pin remapping, Wi-Fi switching, PSRAM tweaks, minimal test sketches, etc.), the real fix turned out to be:

Changing the "Upload Mode" setting in Arduino IDE to USB OTG CDC / TinyUSB

Previously, I was using UART / Hardware CDC and even though uploads were successful, the code wasn't running afterward — only the ESP-ROM bootloader logs appeared. No camera output, no Wi-Fi connection, and no debug prints. After switching to USB OTG CDC, everything started working instantly!

Big thanks to everyone who gave suggestions — your help really kept me going :)