espressif/esp32-camera #454: [PlatformIO][espidf] Linking issues. Wrong PlatformIO target schema?

hpsaturn, 2022-09-25:
## Overview

Over PlatformIO and following the instructions with a `git submodule` way. I had the next link issues:

<details>

```cpp
Successfully created esp32 image.
Indexing .pio/build/esp32/libb05/libesp32-camera.a
Linking .pio/build/esp32/firmware.elf
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(ov2640.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/ov2640.c:502: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(nt99141.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/nt99141.c:932: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(ov3660.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/ov3660.c:965: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(ov5640.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/ov5640.c:1042: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(esp_camera.o):(.literal.camera_probe+0x20): undefined reference to `camera_enable_out_clock'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(esp_camera.o):(.literal.camera_probe+0x24): undefined reference to `camera_disable_out_clock'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(esp_camera.o): in function `camera_probe':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/driver/esp_camera.c:155: undefined reference to `camera_enable_out_clock'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/driver/esp_camera.c:232: undefined reference to `camera_disable_out_clock'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(esp_camera.o): in function `esp_camera_deinit':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/driver/esp_camera.c:237: undefined reference to `camera_disable_out_clock'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(bf3005.o):(.literal.set_xclk+0x4): undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(bf3005.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/bf3005.c:460: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(ov7725.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/ov7725.c:485: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(sc030iot.o): in function `set_xclk':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/sensors/sc030iot.c:283: undefined reference to `xclk_timer_conf'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_start_frame+0x4): undefined reference to `ll_cam_start'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_start_frame+0x8): undefined reference to `ll_cam_do_vsync'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_dma_config+0x38): undefined reference to `ll_cam_dma_sizes'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_dma_config+0x3c): undefined reference to `ll_cam_get_dma_align'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_init+0x14): undefined reference to `ll_cam_set_pin'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_init+0x18): undefined reference to `ll_cam_config'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_stop+0x0): undefined reference to `ll_cam_vsync_intr_enable'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_stop+0x4): undefined reference to `ll_cam_stop'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_deinit+0x0): undefined reference to `ll_cam_deinit'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_config+0x24): undefined reference to `ll_cam_set_sample_mode'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_config+0x28): undefined reference to `ll_cam_init_isr'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o):(.literal.cam_task+0x10): undefined reference to `ll_cam_memcpy'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o): in function `cam_start_frame':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/driver/cam_hal.c:98: undefined reference to `ll_cam_start'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/driver/cam_hal.c:100: undefined reference to `ll_cam_do_vsync'
/home/avp/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libb05/libesp32-camera.a(cam_hal.o): in function `cam_dma_config':
/home/avp/workspace/pio/ttgo/espnow-esp32-camera/lib/esp32-camera/driver/cam_hal.c:259: undefined reference to `ll_cam_dma_sizes'
[... cut ...]
    "type": "git",
    "url": "https://github.com/espressif/esp32-camera"
  },
  "frameworks": "espidf",
  "platforms": "*",
  "build": {
    "flags": [
      "-Idriver/include",
      "-Iconversions/include",
      "-Idriver/private_include",
      "-Iconversions/private_include",
      "-Isensors/private_include",
      "-Itarget/private_include",
      "-fno-rtti"
    ],
    "includeDir": ".",
    "srcDir": ".",
    "srcFilter": ["-<*>", "+<driver>", "+<conversions>", "+<sensors>", "+<target>"]
  }
}
```

Jason2866, 2022-09-29:
A partial fix, not statisfying since there are already nice S3 Cam boards available.
Maybe @valeros can help?

valeros, 2022-09-30:
Hi @hpsaturn ! It seems like you're trying to integrate an IDF component as a generic PlatformIO library. This way the component is not processed by the underlying IDF build system and hence is not added to the build process properly. More detailed info on how to add IDF components to PlatformIO is described here https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components

hpsaturn, 2022-09-30:
Thanks for your comments. I wrote this issue in order to improve the library documentation in this repo, on the section that mentioned Platformio with ESPIDF platform using the component like a submodule. I know that PlatformIO has a different flow for dependencies and build process.

closed 2022-12-06 by hpsaturn