espressif/esp32-camera #455: Getting Error while initializing OV2640 camera in a TTGO T-Journal camera

hpsaturn, 2022-09-25:
## Overview

I want use this driver without Arduino Framework. I have the configuration ok for PlatformIO with platform ESPIDF, but I have the next init error:

<details>

```cpp
␛[0;32mI (29) boot: ESP-IDF 4.4.1 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 15:38:28␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (39) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (43) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (48) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (52) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (58) boot: Partition Table:␛[0m
␛[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (91) boot: End of partition table␛[0m
␛[0;32mI (95) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0cfb4h ( 53172) map␛[0m
␛[0;32mI (130) esp_image: segment 1: paddr=0001cfdc vaddr=3ffb0000 size=0303ch ( 12348) load␛[0m
␛[0;32mI (135) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=22878h (141432) map␛[0m
␛[0;32mI (187) esp_image: segment 3: paddr=000428a0 vaddr=3ffb303c size=005c4h (  1476) load␛[0m
␛[0;32mI (188) esp_image: segment 4: paddr=00042e6c vaddr=40080000 size=0d9f4h ( 55796) load␛[0m
␛[0;32mI (216) esp_image: segment 5: paddr=00050868 vaddr=50000000 size=00010h (    16) load␛[0m
␛[0;32mI (223) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (223) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (236) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (236) cpu_start: Starting app cpu, entry point is 0x40081f60␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (250) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (250) cpu_start: cpu freq: 160000000␛[0m
␛[0;32mI (250) cpu_start: Application information:␛[0m
␛[0;32mI (255) cpu_start: Project name:     espnow-esp32-camera␛[0m
␛[0;32mI (261) cpu_start: App version:      b97a989-dirty␛[0m
␛[0;32mI (266) cpu_start: Compile time:     Sep 25 2022 15:38:08␛[0m
␛[0;32mI (272) cpu_start: ELF file SHA256:  b8aa920c63b20045...␛[0m
␛[0;32mI (278) cpu_start: ESP-IDF:          4.4.1␛[0m
␛[0;32mI (283) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (290) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (296) heap_init: At 3FFB3F68 len 0002C098 (176 KiB): DRAM␛[0m
␛[0;32mI (303) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (309) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (315) heap_init: At 4008D9F4 len 0001260C (73 KiB): IRAM␛[0m
␛[0;32mI (323) spi_flash: detected chip: winbond␛[0m
␛[0;32mI (326) spi_flash: flash io: dio␛[0m
␛[0;32mI (331) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (341) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations␛[0m
␛[0;32mI (351) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 ␛[0m
␛[0;32mI (361) cam_hal: cam init ok␛[0m
␛[0;32mI (361) sccb: pin_sda 25 pin_scl 23␛[0m
␛[0;32mI (371) sccb: sccb_i2c_port=1
␛[0m
␛[0;32mI (371) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (401) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (431) camera: Detected camera at address=0x30␛[0m
␛[0;32mI (431) camera: Detected OV2640 camera␛[0m
␛[0;32mI (431) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2␛[0m
␛[0;32mI (511) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 2␛[0m
␛[0;32mI (511) cam_hal: Allocating 11520 Byte frame buffer in PSRAM␛[0m
␛[0;31mE (521) cam_hal: cam_dma_config(306): frame buffer malloc failed␛[0m
␛[0;31mE (531) cam_hal: cam_config(390): cam_dma_config failed␛[0m
␛[0;31mE (531) camera: Camera config failed with error 0xffffffff␛[0m
␛[0;31mE (541) example:take_picture: Camera Init Failed␛[0m
```

</details>

## setup:

```cpp
#ifdef BOARD_TTGO_TJOURNAL

#define CAM_PIN_PWDN 32
#define CAM_PIN_RESET 15 //software reset will be performed
#define CAM_PIN_XCLK 27
#define CAM_PIN_SIOD 25
#define CAM_PIN_SIOC 23

#define CAM_PIN_D7 18
#define CAM_PIN_D6 39
#define CAM_PIN_D5 5
#define CAM_PIN_D4 34
#define CAM_PIN_D3 35
#define CAM_PIN_D2 17
#define CAM_PIN_D1 4
#define CAM_PIN_D0 2
#define CAM_PIN_VSYNC 22
#define CAM_PIN_HREF 26
#define CAM_PIN_PCLK 21

#endif
``` 
complete [source code](https://github.com/hpsaturn/espnow-camera/tree/espidf)

- TTGO T-Journal camera (The first version)
- esp32-camera driver installed via git submodule way
- PlatformIO pakages:

```
Platform espressif32 @ 5.1.1 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20004.220825 (required: platformio/framework-arduinoespressif32 @ ~3.20004.0)
├── framework-espidf @ 3.40401.0 (required: platformio/framework-espidf @ ~3.40401.0)
├── tool-cmake @ 3.16.4 (required: platformio/tool-cmake @ ~3.16.0)
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── toolchain-esp32ulp @ 1.22851.191205 (required: platformio/toolchain-esp32ulp @ ~1.22851.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3)
``` 

## Notes

Using the same driver (esp32-camera master branch) but with Arduino Framework and PlatformIO, the camera works fine. [source code](https://github.com/hpsaturn/espnow-camera/tree/arduinoframework)

me-no-dev, 2022-09-25:
seems that you have not enabled PSRAM

hpsaturn, 2022-09-25:
How I do it? Because when I enable the PSRAM items on the menuconfig the device enter in pacnic loop:

```cpp
␛[0;32mI (241) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (241) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (253) psram: This chip is ESP32-D0WD␛[0m
␛[0;31mE (254) psram: PSRAM ID read error: 0xffffffff␛[0m
␛[0;31mE (254) cpu_start: Failed to init external RAM!␛[0m

abort() was called at PC 0x400822f0 on core 0

Backtrace:0x40082cc1:0x3ffe3ba00x4008a679:0x3ffe3bc0 0x4009030a:0x3ffe3be0 0x400822f0:0x3ffe3c50 0x400794dc:0x3ffe3c80  |<-CORRUPTED
```

On the other hand the TTGO T-Journal don' t have PSRAM. For this reason I had disable it.

This is my configuration that produce the panic:

[image]
[image]

me-no-dev, 2022-09-26:
You did not post your init camera code, but there is [this](https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h#L145) setting to tell the code that it should not use PSRAM. Do you have enough memory?

felmue, 2022-09-26:
Hi guys

from what I can tell the TTGO T-Journal camera board does **not** have PSRAM. From what I recall PSRAM requires GPIO16 and GPIO17, but GPIO17 is connected to the D2 of the camera.

Thanks
Felix

hpsaturn, 2022-09-26:
Yes, I believe that this camera don't have PSRAM. For this reason I published the issue with PSRAM setting disabled. On the other hand, when I using Arduino Framework with this library, the pin 17 is connected to D0 and the camera works fine, with this code:

https://github.com/hpsaturn/espnow-camera/blob/arduinoframework/src/main.cpp

But I want work with esp-idf. it is why ask you.

I changed the PINs but I have the same loop crash. My current camera pins is:

```cpp
// ESP32 TTGO (T-Journal) PIN Map
#ifdef BOARD_TTGO_TJOURNAL

#define CAM_PIN_PWDN 32
#define CAM_PIN_RESET 15 //software reset will be performed
#define CAM_PIN_XCLK 27
#define CAM_PIN_SIOD 25
#define CAM_PIN_SIOC 23

#define CAM_PIN_D7 19
#define CAM_PIN_D6 36
#define CAM_PIN_D5 18
#define CAM_PIN_D4 39
#define CAM_PIN_D3 5
#define CAM_PIN_D2 34
#define CAM_PIN_D1 35
#define CAM_PIN_D0 17
#define CAM_PIN_VSYNC 22
#define CAM_PIN_HREF 26
#define CAM_PIN_PCLK 21

#endif
```

hpsaturn, 2022-09-26:
> 

My initialization camera code for espidf platform is the next:

https://github.com/hpsaturn/espnow-camera/tree/espidf/src

I have enough memory? well I think that yes, because with ArduinoFramework this camera works. I only need know how to init this camera with espidf platform. In the first post of this thread I posted the two references to the two source codes.

me-no-dev, 2022-09-26:
this last code is working fine. The example in Arduino is actually a copy of example that was available for IDF in esp-who (now gone), but how Arduino handles external memory is slightly different. Anyway, it seems that you have it resolved now.

hpsaturn, 2022-09-26:
Ok! thanks guys! I think that it is resolved for now. I don't found any schematic to say that maybe this camera has PSRAM. For the moment it is enough for make some tests.

closed 2022-09-26 by hpsaturn