espressif/esp32-camera #459: multiple definition of `xclk_timer_conf'

Kampi, 2022-10-04:
I got the error

```
c:/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\debug\src\Application\Camera\esp-camera-ll\xclk.o: in function `xclk_timer_conf':
C:\Projects\streamer/src/Application/Camera/esp-camera-ll/xclk.c:19: multiple definition of `xclk_timer_conf'; .pio\build\debug\src\Application\Camera\esp-camera-ll\esp32s3\ll_cam.o:C:\Projects\streamer/src/Application/Camera/esp-camera-ll/esp32s3/ll_cam.c:450: first 
defined here
```

when compiling the code for the ESP32-S3 with the code from the `targets` directory. The function `xclk_timer_conf` is defined in `xclk.c` and `ll_cam.c`. My guess is that only the code from `ll_cam.c` is relevant when using the ESP32-S3?

me-no-dev, 2022-10-05:
Yes and is as such defined in CMakeLists.txt: https://github.com/espressif/esp32-camera/blob/master/CMakeLists.txt#L55 and https://github.com/espressif/esp32-camera/blob/master/CMakeLists.txt#L62 for ESP32 and ESP32-S2. Maybe PIO requires something specific configuration for your project?

Kampi, 2022-10-05:
You are right. The problem only occurs when adding the camera by using `lib_deps`. Adding the camera as a component avoids this issue.

closed 2022-10-05 by Kampi