espressif/esp32-camera #461: Error compile when using esp-idf v4.3.4

duytuongtran27, 2022-10-08:
Here is my terminal output error

E:/Desktop/Luan van/esp-idf/components/esp_psram/esp32/esp_psram_impl_quad.c: In function 'psram_2t_mode_enable':
E:/Desktop/Luan van/esp-idf/components/esp_psram/esp32/esp_psram_impl_quad.c:593:5: error: implicit declaration of function 'GPIO_OUTPUT_SET'; did you mean 'GPIO_OUT_REG'? [-Werror=implicit-function-declaration]
  593 |     GPIO_OUTPUT_SET(D0WD_PSRAM_CS_IO, 1);
      |     ^~~~~~~~~~~~~~~
      |     GPIO_OUT_REG
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the e:\desktop\luan van\kid_room_monitor\build\log\idf_py_stderr_output_15676 and e:\desktop\luan van\kid_room_monitor\build\log\idf_py_stdout_output_15676

WangYuxin-esp, 2022-10-09:
It is recommended to update your IDF version. This problem is only about the version of the development environment.

SoniaZotz, 2022-10-18:
esp-idf v4.3.4 is the latest version .....which version should we use

duytuongtran27, 2022-10-22:
i try using v4.4.2 and i can work well

SoniaZotz, 2022-10-25:
using esp-idf-v4.4.2 -->>> Unknown CMake command "register_component".

Building in: E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\esp32-camera-master\build
Configuring in: E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\esp32-camera-master\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=E:\Espressif\frameworks\esp-idf-v4.4.2\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32 E:\Espressif\frameworks\esp-idf-v4.4.2\workspace\esp32-camera-master
CMake Warning (dev) at E:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/utilities.cmake:375 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-mlongcalls;-Wno-frame-address;].
Call Stack (most recent call first):
  E:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/toolchain-esp32.cmake:10 (remove_duplicated_flags)
  E:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:3 (project)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at E:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/utilities.cmake:375 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-mlongcalls;-Wno-frame-address;].
Call Stack (most recent call first):
  E:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/toolchain-esp32.cmake:12 (remove_duplicated_flags)
  E:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:3 (project)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at E:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/utilities.cmake:375 (list):
  Policy CMP0007 is not set: list command no longer ignores empty elements.
  Run "cmake --help-policy CMP0007" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.  List has value =
  [-mlongcalls;].
Call Stack (most recent call first):
  E:/Espressif/frameworks/esp-idf-v4.4.2/tools/cmake/toolchain-esp32.cmake:14 (remove_duplicated_flags)
  E:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:3 (project)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:89 (register_component):
-- Configuring incomplete, errors occurred!
See also "E:/Espressif/frameworks/esp-idf-v4.4.2/workspace/esp32-camera-master/build/CMakeFiles/CMakeOutput.log".
  Unknown CMake command "register_component".

me-no-dev, 2022-10-25:
from the log it seems that you are trying to build from the component root folder instead of a project that includes this component. This repository is not a buildable project, but a component that you can include in your projects

SoniaZotz, 2022-10-25:
Thank you for clarifying this .....can you give me  an build example project that I would use this component
Just starting out and this is a bit confusing for me 
your help is appreciated

closed 2023-01-04 by github-actions[bot]